raspberrypi/userland #499: Can't set OMX.broadcom.image_fx eDomain to OMX_PortDomainVideo

jvcleave, 2018-11-22:
According to [docs](http://www.jvcref.com/files/PI/documentation/ilcomponents/image_fx.html)

```
 Port 190 also supports being connected to an output port of video domain, by setting the port definition type with a structure of OMX_PortDomainVideo.
```

test case based off `opt/vc/src/hello_pi/hello_video.c`

https://gist.github.com/jvcleave/b4546bebedcfd8671e3ff68f7696c2c8

```
2760607.343: OMX.broadcom.image_fx:init
2760607.495: top:create_component image_fx
2760607.638: image_fxRIL:GetParameter(1000002)
2760607.655: image_fxRIL:GetParameter(1000004)
2760607.668: image_fxRIL:GetParameter(1000003)
2760607.685: image_fxRIL:GetParameter(1000005)
2760607.714: image_fxRIL:GetParameter(7f000008)
2760607.733: image_fxRIL:GetParameter(2000001)
2760607.752: image_fxRIL:GetParameter(7f000007)
2760607.767: image_fxRIL:GetParameter(7f000006)
2760607.780: image_fxRIL:GetParameter(2000001)
2760607.797: image_fxRIL:GetParameter(7f000007)
2760607.812: image_fxRIL:GetParameter(7f000006)
2760607.838: image_fxRIL:SetParameter(7f0000a9)
2760607.854: image_fx:215:SetCallbacks
2760607.873: image_fx:215:SetParameter(7f000001)
2760607.893: image_fx:215:GetParameter(7f000002)
2760607.914: image_fx:215:SetParameter(7f000004)
2760608.200: image_fx:215:SetCallbacks
2760608.388: image_fx:215:GetParameter(1000002)
2760608.401: image_fxRIL:GetParameter(1000002)
2760608.566: image_fx:215:GetParameter(1000003)
2760608.581: image_fxRIL:GetParameter(1000003)
2760608.747: image_fx:215:GetParameter(2000001)
2760608.761: image_fxRIL:GetParameter(2000001)
2760608.945: image_fx:215:SendCommand(2,190,3f72fe54) Disable 190
2760608.965: image_fx:215:sending cmd 2 to 190
2760608.989: image_fx:215:starting command 2
2760609.002: image_fxRIL:GetParameter(2000001)
2760609.030: image_fx:215:processingCommand=0
2760609.355: image_fx:215:GetParameter(2000001)
2760609.369: image_fxRIL:GetParameter(2000001)
2760609.584: image_fx:215:SendCommand(2,191,3f72fe54) Disable 191
2760609.606: image_fx:215:sending cmd 2 to 191
2760609.632: image_fx:215:starting command 2
2760609.646: image_fxRIL:GetParameter(2000001)
2760609.672: image_fx:215:processingCommand=0
2760609.840: image_fx:215:GetParameter(1000004)
2760609.856: image_fxRIL:GetParameter(1000004)
2760610.154: image_fx:215:GetParameter(1000005)
2760610.171: image_fxRIL:GetParameter(1000005)
2760610.516: image_fx:215:SendCommand(0,2,3f72fe54) State to idle
2760610.540: image_fx:215:starting command 0
2760610.560: image_fxRIL: setting port 190 to state 1 (was 0)
2760610.585: image_fxRIL: setting port 190 state - now state 0, error 0
2760610.609: image_fxRIL: setting port 191 to state 1 (was 0)
2760610.633: image_fxRIL: setting port 191 state - now state 0, error 0
2760610.659: image_fxRIL: setting port 190 to state 1 (was 0)
2760610.682: image_fxRIL: setting port 190 state - now state 0, error 0
2760610.705: image_fxRIL: setting port 191 to state 1 (was 0)
2760610.727: image_fxRIL: setting port 191 state - now state 0, error 0
2760610.749: image_fx:215:GetState(1)
2760610.775: image_fxRIL: setting port 190 to state 1 (was 0)
2760610.798: image_fxRIL: setting port 190 state - now state 0, error 0
2760610.820: image_fxRIL: setting port 191 to state 1 (was 0)
2760610.845: image_fxRIL: setting port 191 state - now state 0, error 0
2760610.863: image_fxRIL:handling STATE_CHANGE event in process_thread.
2760610.931: image_fx:215:SendCommand(0,2,3f72fe54) State to idle
2760625.083: image_fxRIL: setting port 190 to state 1 (was 1)
2760625.107: image_fxRIL: setting port 190 state - now state 1, error 0
2760625.121: image_fxRIL:GetParameter(2000001)
2760625.145: image_fxRIL: setting port 191 to state 1 (was 1)
2760625.165: image_fxRIL: setting port 191 state - now state 1, error 0
2760625.178: image_fxRIL:GetParameter(2000001)
2760625.196: image_fx:215:completed state transition to OMX_StateIdle
2760625.220: image_fx:215:processingCommand=0
2761611.340: image_fx:215:GetState(2)
2761611.616: image_fx:215:GetParameter(2000001)
2761611.632: image_fxRIL:GetParameter(2000001)
2761613.101: image_fx:215:SetParameter(2000001)
2761613.116: image_fxRIL:SetParameter(2000001)
2761613.336: image_fx:215:GetParameter(2000001)
2761613.351: image_fxRIL:GetParameter(2000001)
2761615.212: image_fx:215:SendCommand(2,-1,0) Disable -1
2761615.235: image_fx:215:starting command 2
2761615.253: image_fx:215:processingCommand=0
2761615.279: image_fx:215:GetState(2)
2761615.295: image_fx:215:GetState(2)
2761615.321: image_fx:215:SendCommand(0,1,0) State to loaded
2761615.342: image_fx:215:GetState(2)
2761615.363: image_fx:215:GetState(2)
2761615.383: image_fx:215:starting command 0
2761615.407: image_fxRIL: setting port 190 to state 0 (was 1)
2761615.430: image_fxRIL: setting port 190 state - now state 1, error 0
2761615.445: image_fxRIL:GetParameter(2000001)
2761615.468: image_fxRIL: setting port 191 to state 0 (was 1)
2761615.493: image_fxRIL: setting port 191 state - now state 1, error 0
2761615.509: image_fxRIL:GetParameter(2000001)
2761615.534: image_fxRIL: setting port 190 to state 0 (was 1)
2761615.556: image_fxRIL: setting port 190 state - now state 1, error 0
2761615.569: image_fxRIL:GetParameter(2000001)
2761615.592: image_fxRIL: setting port 191 to state 0 (was 1)
2761615.615: image_fxRIL: setting port 191 state - now state 1, error 0
2761615.631: image_fxRIL:GetParameter(2000001)
2761615.660: image_fxRIL: setting port 190 to state 0 (was 1)
2761615.681: image_fxRIL: setting port 190 state - now state 1, error 0
2761615.693: image_fxRIL:GetParameter(2000001)
2761615.717: image_fxRIL: setting port 191 to state 0 (was 1)
2761615.741: image_fxRIL: setting port 191 state - now state 1, error 0
2761615.759: image_fxRIL:GetParameter(2000001)
2761615.774: image_fxRIL:handling STATE_CHANGE event in process_thread.
2761635.085: image_fxRIL: setting port 190 to state 0 (was 0)
2761635.105: image_fxRIL: setting port 190 state - now state 0, error 0
2761635.118: image_fxRIL:GetParameter(2000001)
2761635.140: image_fxRIL: setting port 191 to state 0 (was 0)
2761635.162: image_fxRIL: setting port 191 state - now state 0, error 0
2761635.177: image_fxRIL:GetParameter(2000001)
2761635.196: image_fx:215:completed state transition to OMX_StateLoaded
2761635.214: image_fx:215:processingCommand=0
2761635.231: image_fx:215:GetState(1)
2761635.247: image_fx:215:GetParameter(7f000003)
2761635.268: image_fx:215:ComponentTunnelRequest(190,clear)
2761635.283: image_fxRIL:GetParameter(7f000006)
2761635.305: image_fx:215:GetParameter(7f000003)
2761635.326: image_fx:215:ComponentTunnelRequest(191,clear)
2761635.340: image_fxRIL:GetParameter(7f000006)
2761635.356: image_fx:215:GetParameter(7f000003)
2761635.372: image_fx:215:GetParameter(7f000003)
2761635.391: image_fx:215:GetParameter(7f000003)
2761635.419: image_fx:215:ComponentDeInit
2761635.439: image_fxRIL:id->processQuit in process_thread.
```

jvcleave, 2018-11-22:
`OMX_ErrorBadParameter` seems to be the issue:

added error checking - new output:

```
PORT #: 190 DISABLED
PORT #: 191 DISABLED
ATTEMPT #: 1 
imageFX is OMX_StateIdle: 1 
ERROR LINE: 238 
OMX_ErrorNone 
 

 INITIAL DEFINITION: 

nPortIndex: 190 
eDir: in 
nBufferCountActual: 1 
nBufferCountMin: 1 
nBufferSize: 0 
nBufferAlignment: 16 
bEnabled: 0 
bPopulated: 0 
bBuffersContiguous: 0 
bBuffersContiguous: 0 
eDomain: OMX_PortDomainImage 
image nFrameWidth: 160 
image nFrameHeight: 64 
image nStride: 160 
image nSliceHeight: 16 
image coding type: 0 
image eColorFormat: 20 
ERROR LINE: 247 
OMX_ErrorBadParameter 
 ERROR LINE: 251 
OMX_ErrorNone 
 

 RESULT DEFINITION: 

nPortIndex: 190 
eDir: in 
nBufferCountActual: 1 
nBufferCountMin: 1 
nBufferSize: 0 
nBufferAlignment: 16 
bEnabled: 0 
bPopulated: 0 
bBuffersContiguous: 0 
bBuffersContiguous: 0 
eDomain: OMX_PortDomainImage 
image nFrameWidth: 160 
image nFrameHeight: 64 
image nStride: 160 
image nSliceHeight: 16 
image coding type: 0 
image eColorFormat: 20 
```

And new log:
```
1859911.373: OMX.broadcom.image_fx:init
1859911.522: top:create_component image_fx
1859911.657: image_fxRIL:GetParameter(1000002)
1859911.672: image_fxRIL:GetParameter(1000004)
1859911.684: image_fxRIL:GetParameter(1000003)
1859911.701: image_fxRIL:GetParameter(1000005)
1859911.728: image_fxRIL:GetParameter(7f000008)
1859911.745: image_fxRIL:GetParameter(2000001)
1859911.760: image_fxRIL:GetParameter(7f000007)
[... 89 lines cut ...]
1860919.589: image_fxRIL: setting port 191 state - now state 1, error 0
1860919.606: image_fxRIL:GetParameter(2000001)
1860919.621: image_fxRIL:handling STATE_CHANGE event in process_thread.
1860936.421: image_fxRIL: setting port 190 to state 0 (was 0)
1860936.442: image_fxRIL: setting port 190 state - now state 0, error 0
1860936.455: image_fxRIL:GetParameter(2000001)
1860936.479: image_fxRIL: setting port 191 to state 0 (was 0)
1860936.504: image_fxRIL: setting port 191 state - now state 0, error 0
1860936.518: image_fxRIL:GetParameter(2000001)
1860936.540: image_fx:229:completed state transition to OMX_StateLoaded
1860936.558: image_fx:229:processingCommand=0
1860936.578: image_fx:229:GetState(1)
1860936.594: image_fx:229:GetParameter(7f000003)
1860936.615: image_fx:229:ComponentTunnelRequest(190,clear)
1860936.630: image_fxRIL:GetParameter(7f000006)
1860936.653: image_fx:229:GetParameter(7f000003)
1860936.677: image_fx:229:ComponentTunnelRequest(191,clear)
1860936.692: image_fxRIL:GetParameter(7f000006)
1860936.708: image_fx:229:GetParameter(7f000003)
1860936.724: image_fx:229:GetParameter(7f000003)
1860936.744: image_fx:229:GetParameter(7f000003)
1860936.774: image_fx:229:ComponentDeInit
1860936.794: image_fxRIL:id->processQuit in process_thread.
1860936.846: image_fx:229: total=0 not_loaded=0
```

6by9, 2018-11-22:
It'll accept a **valid** video domain port definition.

Call ```PrintDef``` on the port definition that you are calling SetParameter with and you'll find that it prints weird values for ```eCompressionFormat``` and ```eColorFormat```.

```OMX_VIDEO_PORTDEFINITIONTYPE``` has ```OMX_U32 xFramerate;``` and ```OMX_BOOL bFlagErrorConcealment;``` after ```nSliceHeight``` and before ```bFlagErrorConcealment```.
```OMX_IMAGE_PORTDEFINITIONTYPE```has no fields in there, therefore ```eCompressionFormat``` is 8 bytes earlier in the union.

Extend your ```def.eDomain = OMX_PortDomainVideo;``` to also set
```
def.format.video.eCompressionFormat = def.format.image.eCompressionFormat;
def.format.video.eColorFormat = def.format.image.eColorFormat;
```
or explicitly set those values and it should accept it.

jvcleave, 2018-11-23:
Thanks! no Bad Parameter but eDomain remains as `OMX_PortDomainImage`. Is this expected?

6by9, 2018-11-23:
> but eDomain remains as OMX_PortDomainImage. Is this expected?

Yes. What should happen is that it has taken on all the properties that you set in the video domain port definition, but the port will have remained as an image domain port. That bit won't be obvious as you haven't changed any parameters from the port def that you read back.

It's mainly so that you can call GetParameter(PortDefinition) on a video domain output port, amend only the nPortIndex to image_fx input, and then setParameter(PortDefinition) on the image_fx. The port itself doesn't change domain. Exactly the same path used when setting up a tunnel to image_fx, and as image_fx is used for deinterlacing in things like omxplayer I know that works.

jvcleave, 2018-11-23:
Thanks! - good to know.

I do know that it works as expected with decoder as an input. However I was trying to use a similar setup with the camera as an input and having issues

6by9, 2018-11-23:
As above, GetParameter on the camera output port, change the nPortIndex, setParameter on the image_fx input port, should work.

Minor question of why you're trying to tunnel to image_fx. The camera component should also provide access to the same set of effect plugins as image_fx, unless you're trying to apply two simultaneously.

jvcleave, 2018-11-23:
yeah - I was trying mostly trying to apply 2 to see if it worked.

jvcleave, 2018-11-23:
Ok - got the camera to work! 

had to do a little tweak with nSliceHeight that I don't think I typically have to do with decoder

```
 OMX_PARAM_PORTDEFINITIONTYPE cameraOutputPort;
 OMX_INIT_STRUCTURE(cameraOutputPort);
 cameraOutputPort.nPortIndex = CAMERA_OUTPUT_PORT;
 error =  OMX_GetParameter(camera, OMX_IndexParamPortDefinition, &cameraOutputPort);
 OMX_TRACE(error);
    

OMX_PARAM_PORTDEFINITIONTYPE imageFXPortDefinition;
OMX_INIT_STRUCTURE(imageFXPortDefinition);
imageFXPortDefinition.nPortIndex = IMAGE_FX_INPUT_PORT;

error =  OMX_GetParameter(imageFX, OMX_IndexParamPortDefinition, &imageFXPortDefinition);
OMX_TRACE(error);

imageFXPortDefinition.eDomain = OMX_PortDomainVideo;
imageFXPortDefinition.format.video = cameraOutputPort.format.video;
/*
 Had to set the next line otherwise the nSliceHeight 
 for IMAGE_FX_OUTPUT_PORT would end up being 16 (good for glitch art)
*/
imageFXPortDefinition.format.video.nSliceHeight = settings->sensorHeight;

error =  OMX_SetParameter(imageFX, OMX_IndexParamPortDefinition, &imageFXPortDefinition);

imageFXPortDefinition.nPortIndex = IMAGE_FX_OUTPUT_PORT;
error =  OMX_SetParameter(imageFX, OMX_IndexParamPortDefinition, &imageFXPortDefinition);

```

closed 2018-11-23 by jvcleave

6by9, 2018-11-23:
nSliceHeight can be set on each link.
All components support it being ALIGN_UP(nFrameHeight, 16), and some components support it being 16 for a low memory use case.
If ```nSliceHeight < nFrameHeight``` then you need to accumulate buffers until you get the ```OMX_BUFFERFLAG_ENDOFFRAME``` flag on a buffer.

Camera supports both (IIRC actually any multiple of 16), video_decode only supports full frames. image_fx supports both.

Reading the code, image_fx will copy through the input port definition to the output port when the input port is enabled (assuming the output isn't already enabled at that point), and trigger a ```OMX_EventPortSettingsChanged``` event. However it claims to keep ```nSliceHeight```, and recomputes ```nBufferSize```.
If you want specific ```nSliceHeight``` behaviour then you need to specify it.