raspberrypi/userland #501: Kernel error after running raspivid.

glddiv, 2018-11-26:
I used  command `raspivid -t 0 -r test.raw -o test.h264` on PI Zero to run the camera. After a few seconds I got the following error:

Message from syslogd@raspberrypi at Nov 22 04:07:57 ...
 kernel:[  420.764892] Internal error: Oops: 17 [#1] ARM

Message from syslogd@raspberrypi at Nov 22 04:07:57 ...
 kernel:[  420.765229] Process vc.ril.video_en (pid: 833, stack limit = 0xd3952188)

Message from syslogd@raspberrypi at Nov 22 04:07:57 ...
 kernel:[  420.765242] Stack: (0xd3953ca8 to 0xd3954000)

Message from syslogd@raspberrypi at Nov 22 04:07:57 ...
 kernel:[  420.765260] 3ca0:                   c948bb08 d68ca400 d3953cec d3953cc0 c01e2f10 c01e2d3c

Message from syslogd@raspberrypi at Nov 22 04:07:57 ...
 kernel:[  420.765276] 3cc0: 00001000 00000000 c948bb08 00001000 c948bb08 d68ca400 0097a000 00000000

Message from syslogd@raspberrypi at Nov 22 04:07:57 ...
 kernel:[  420.765291] 3ce0: d3953d0c d3953cf0 c0204d04 c01e2ee4 00000000 0097a000 00000007 00001000

Message from syslogd@raspberrypi at Nov 22 04:07:57 ...
 kernel:[  420.765307] 3d00: d3953d2c d3953d10 c018759c c0204cd0 d7ccac44 00001000 0097a000 00000000

Message from syslogd@raspberrypi at Nov 22 04:07:57 ...
 kernel:[  420.765325] 3d20: d3953d74 d3953d30 c0190138 c0187554 00001000 00001000 d7ccac44 00000000

Message from syslogd@raspberrypi at Nov 22 04:07:57 ...
 kernel:[  420.765341] 3d40: 00000000 00000001 00001000 00000000 c948bb08 00001000 00001000 0097a000

Message from syslogd@raspberrypi at Nov 22 04:07:57 ...
 kernel:[  420.765360] 3d60: 00000000 c948bbe4 d3953dc4 d3953d78 c0203820 c019006c 00001000 00001000

Message from syslogd@raspberrypi at Nov 22 04:07:57 ...
 kernel:[  420.765377] 3d80: d7ccac44 00000000 0097b000 00000000 cd54a960 d6fb60f0 c0204d28 00000000

Message from syslogd@raspberrypi at Nov 22 04:07:57 ...
 kernel:[  420.765393] 3da0: 00001000 c948bbe4 d3953ef0 00001000 00001000 c065f388 d3953e24 d3953dc8

Message from syslogd@raspberrypi at Nov 22 04:07:57 ...
 kernel:[  420.765408] 3dc0: c00f54dc c02036f4 00001000 00001000 d7ccac44 00000000 c948bb08 d3952000

Message from syslogd@raspberrypi at Nov 22 04:07:57 ...
 kernel:[  420.765423] 3de0: 00001000 cd54a960 0097a000 00000000 d7ccac44 00000000 5bf62b9d 00000000

Message from syslogd@raspberrypi at Nov 22 04:07:57 ...
 kernel:[  420.765439] 3e00: d3953f08 c948bbe4 c948bb08 00000000 cd54a960 d3953ef0 d3953e6c d3953e28

Message from syslogd@raspberrypi at Nov 22 04:07:57 ...
 kernel:[  420.765455] 3e20: c00f7ba4 c00f53c8 c050fdc8 c01490f8 400cc404 c09b8130 d3953f0c d3953e48

Message from syslogd@raspberrypi at Nov 22 04:07:57 ...
 kernel:[  420.765474] 3e40: c0518be8 d3953f08 d3953ef0 c948bb74 c948bb08 00000000 0000b000 c948bb08

Message from syslogd@raspberrypi at Nov 22 04:07:57 ...
 kernel:[  420.765495] 3e60: d3953edc d3953e70 c01eece8 c00f7a1c 00000000 009d1000 d535a4e0 000005f8

Message from syslogd@raspberrypi at Nov 22 04:07:57 ...
 kernel:[  420.765510] 3e80: 00000000 00000000 00000000 ffffffff d3953ea8 00000000 00000001 00000002

Message from syslogd@raspberrypi at Nov 22 04:07:57 ...
 kernel:[  420.765526] 3ea0: 009d15f8 00000001 d3952000 009d15f8 d3953f5c cd54a960 d3953f80 00979000

Message from syslogd@raspberrypi at Nov 22 04:07:57 ...
 kernel:[  420.765541] 3ec0: 00000000 0000b000 00000000 00000000 d3953f4c d3953ee0 c015af94 c01eebd4

Message from syslogd@raspberrypi at Nov 22 04:07:57 ...
 kernel:[  420.765556] 3ee0: 0000b000 b3bfed2c 009d412f 0000b000 00000001 00001000 0000a000 d3953ee8

Message from syslogd@raspberrypi at Nov 22 04:07:57 ...
 kernel:[  420.765573] 3f00: 00000001 00000002 cd54a960 00000000 00979000 00000000 00000000 d3953e94

Message from syslogd@raspberrypi at Nov 22 04:07:57 ...
 kernel:[  420.765589] 3f20: 00000000 00000000 0000b000 cd54a960 009d412f d3953f80 c000ff64 d3952000

Message from syslogd@raspberrypi at Nov 22 04:07:57 ...
 kernel:[  420.765607] 3f40: d3953f7c d3953f50 c015b1b0 c015aec0 00000002 cd54a963 cd54a960 cd54a963

Message from syslogd@raspberrypi at Nov 22 04:07:57 ...
 kernel:[  420.765627] 3f60: 0000b000 009d412f c000ff64 d3952000 d3953fa4 d3953f80 c015b404 c015b10c

Message from syslogd@raspberrypi at Nov 22 04:07:57 ...
 kernel:[  420.765644] 3f80: 00979000 00000000 0000b000 009d412f 009e7b78 00000004 00000000 d3953fa8

Message from syslogd@raspberrypi at Nov 22 04:07:57 ...
 kernel:[  420.765659] 3fa0: c000fdc0 c015b3bc 0000b000 009d412f 00000005 009d412f 0000b000 00000000

Message from syslogd@raspberrypi at Nov 22 04:07:57 ...
 kernel:[  420.765674] 3fc0: 0000b000 009d412f 009e7b78 00000004 0000b000 00000004 00000001 b3bfefac

Message from syslogd@raspberrypi at Nov 22 04:07:57 ...
 kernel:[  420.765690] 3fe0: 00000000 b3bfed48 00000000 b6df3184 80000010 00000005 00000000 00000000

Message from syslogd@raspberrypi at Nov 22 04:07:57 ...
 kernel:[  420.766119] Code: e5923124 e1a030a3 e2134001 1a00000d (e5953034) 
^Cmmal: Aborting program

6by9, 2018-11-26:
Your SD card is going to fill up very fast.
raspivid defaults to 1920x1080, so adding ```-r test.raw``` will be saving just over 3MB per frame, and trying to save 30frames per second, so 90MB/s (734Mbit/s). No storage will cope with that.

Tested on a Pi3B+ without any issue observed (other than lots of dropped frames as it waited for the storage to catch up).
Please provide the full kernel log output using dmesg.

glddiv, 2018-11-27:
@6by9  thank for your reply.
I don't know how to get the kernel message log, because my kernel has panic and I can't run any command except repower it.
What's more, I remove this SD Card  and insert it to Pi 3B it can work normally, except that the picture will occasionally be stuck.

glddiv, 2018-11-27:
Fortunately the system is sometimes not completely crashed, this is dmesg information
[dmesg_log.txt](https://github.com/raspberrypi/userland/files/2618015/dmesg_log.txt)

JamesH65, 2018-11-29:
Looking at the log, there's a NULL ptr dereference in the EXT4 system. This is to do with saving the image to the SD card (or whatever is being used). I suspect there is a problem with the SD card itself, either it's full, or corrupted in some way. I don't believe this isn't a problem with raspivid.

glddiv, 2018-12-06:
I used this command raspivid -t 0 -r /dev/null -o /dev/null result is the same.

JamesH65, 2018-12-06:
So what is accessing the SD card so that the EXT subsystem is being used? Hmm. Could be anything in Linux. Going to be almost impossible to track down since it doesn't seem to happen for anyone else.

JamesH65, 2018-12-06:
Have you tried a new SD card with a new install of the OS on it? Could it be a power problem? What power supply are you using, do you get the lightening icon appearing?

glddiv, 2018-12-11:
I did more tests, it seems that my Pi Zero W has a problem in the test results, but I only have one Pi Zero W, I can't do more tests. This is the test result:
[image]

JamesH65, 2019-01-08:
OK, been testing this and I am  seeing no problems at all. Have used the steps outlined above, and device runs without failure, so far for over 10 minutes. Using 4.14.50+

So not sure what to suggest, I suppose it could be a faulty device, but it is a very odd error if so.

glddiv, 2019-03-02:
@JamesH65 Thank you for everything you have done and  sorry for replying so late, I got a new Pi Zero W, but the problem is still here. So I think the problem doesn't depend on Pi Zero W. Finally, I found the problem is on the camera. My camera is OV5647 with a clock frequency of 27MHZ. After I replaced the camera with OV5647 with a clock frequency of 24MHz, it works very well. (I'm not sure if I missed something because I only did a simple test).

closed 2019-03-04 by glddiv