raspberrypi/userland #502: "countof" macro causes compilation failures when glm is uincluded

kaomoneus, 2018-11-26:
Namely this line:
https://github.com/raspberrypi/userland/blob/d574b51a60a075baefe863670466ee24e6c4256e/interface/vcos/vcos_types.h#L205

In GLM we also have definition of countof, properly namespaced though:
https://github.com/g-truc/glm/blob/9749727c2db4742369219e1d452f43e918734b4e/glm/detail/setup.hpp#L544

**Bug**
If I include vcos_types first, and then include glm lib, then I have a compilation error,
because compiler tries to expand a 'countof' macro instead of recognizing 'countof' as a new function name itself.

**Solutions?**
Introduce smth like ENABLE_VC_BACKWARD_COMPATABILITY and activate everything related to backward compatability only if it is enabled.

6by9, 2018-11-26:
Sorry, my instinctive reaction is no.

VCOS is frequently used from plain C, therefore namespaces aren't applicable. It's not applicable to macros either.

And introducing a define such as ENABLE_VC_BACKWARD_COMPATABILITY will break any exisiting applications until they add the required define.

kaomoneus, 2018-11-26:
OK, that was just a proposal. But let's think.
What should we do in case when we need GLM and VCOS together? GLM is quite a frequently used lib. There are a pretty much of chances to use both VCOS and GLM.
Undefine "countof" before including GLM? That should work, but perhaps not the best solution.

ENABLE_VC_BACKWARD_COMPATABILITY could be added into Cmake (I could provide you a patch), enabled by default.

**Alternative**
Use DISABLE_VC_BACKWARD_COMPATABILITY. So once we define such thing in user code, we could use GLM and perhaps another things.
Or... somehow find another way to isolate those macros...

kaomoneus, 2018-11-27:
> Can't you just #undef countof after including the vcos headers?

Sure, this is what I did. Looks like this is the best possible thing to do.

JamesH65, 2018-12-03:
So, can this be closed?

kaomoneus, 2018-12-03:
I think so, in general this is not a vcos issue.

closed 2018-12-03 by pelwell