raspberrypi/userland #512: Eliminating Xorg tearing, possible?

vanfanel, 2018-12-06:
Hi there,

For Intel graphics hardware, there's an Xorg config option to prevent tearing, and it works truly great:
```

Section "Device"
  Identifier  "Intel Graphics"
  Driver      "intel"
  Option      "TearFree" "true"
EndSection

```

Is there someting like this for Raspbian/Xorg on the Pi hardware using any of the graphics stacks?

JamesH65, 2018-12-07:
Are you actually encountering tearing then?

AFAIK, there are no XORG specific tearing prevention options.

popcornmix, 2018-12-07:
The framebuffer will always tear.
Tear-free updates are possible through dispmax/egl/mmal.

JamesH65, 2019-01-08:
This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

vanfanel, 2019-01-08:
Well, I don't know the internals on the Intel Xorg driver, but that "TearFree" option REALLY makes a difference when moving windows around and scrolling webpages.
@popcornmix : I was never interested in Xorg on the Pi until now, so... what's used to draw the desktop and windows? plain framebuffer? Then how does the Intel driver get tear free screen refresh on X11?

popcornmix, 2019-01-08:
We have a single framebuffer that the arm can write to and the VPU displays without synchronisation.
It will tear. That is by design and we have no plans to change that (solutions are expensive in memory terms and the benefits are minor for normal desktop use).

There are other interfaces that can suppose double/triple buffering and vsync synchronisation (dispmanx/egl/mmal). These are advisable in cases where tearing is more critical (e.g. video playback with omxplayer).

It may be possible to run the vc4-kms-v3d arm side driver in a mode where it is double/triple buffered with vsync, but it's not something I've tried.

vanfanel, 2019-01-08:
@popcornmix : Thanks for the information. Now I understand the problem... Well, this can be closed now, thanks!

closed 2019-01-08 by vanfanel

Askmewho, 2020-04-23:
@popcornmix please, try it or leave us a minor instructions to do it if they are not that unfriendly.

hotnuma, 2021-05-17:
Two years later, I assume it's better to buy a mini PC with Intel integrated graphics.
It's more expensive but at least Intel engineers are able to fix that type of issue.

Askmewho, 2021-05-17:
I dont have tearing on any of my rk3399 boards. that being said, you can also have tearless experience with rpi4 on Ubuntu. at least on my laat test it didnt @hotnuna

JamesH65, 2021-05-17:
Intel: 110,000 employees (not sure how many engineers, but lots), revenue of US$77.87 billion (2020)
Raspberry Pi (trading): 100 employees (25 engineers), revenue of £31M (2019)

That might give some indication why Intel seem to be able to do things a little quicker, I suspect their graphics driver team is considerably bigger than the whole of the Pi development team which deals with our entire system. (Note their wireless chips, which after over 3 years still don't work properly on a laptop I have, so even they have problems)

KMS is still under development, need to get that going before really getting going on Vulkan. Nearly there, you can try out the current version easily. This is nothing to do with Broadcom BTW, all dev work on this is funded by RP.

But as above, we are not forcing you to buy Raspberry Pi, if another device suits you better, then buy that one.

Pi support is still the best of all the SBC's, and getting better all the time. 

Just an anecdote though, I have a Samsung Smart TV, it tears when playing back Netflix....

HinTak, 2021-05-17:
Hmm, FWIW, I think many of the comments are unfair, on one issue: the typical Intel device is more expensive than the whole of the most expensive pi. There is only so much software can do, when you work with the hardware which is far cheaper and optimized for entirely different aspects (power consumption, for one, physical size for another) than performance.

Askmewho, 2021-05-17:
@JamesH65  Dear, I did not even compare rpi with Intel, that was another guy. let's begin.

I can do stuff on RK3399 that would be impossible on RPI4, mostly due that fake kms situation. and that would be the case on other bifrost newer sbcs. also, all of them are far better running 64 bit distributions. it's debatable.

what I consider disrespectful it's that your brand isn't able to choose, and also that you are skipping the situation that your brand, supposedly an non profit organization to create cheap sbcs for tinkering and learning is completely fake.

you brand is being lead by a Broadcom engineer that will work with broadcom SoCs that earn money for broadcom. your brand is Broadcom. and yes, broadcom make tons of money with your brand. PINE64 is a real non profitable organization, they are not attached like your brand, they can choose not to use rockchip if they want, they can choose amlogic, allwinner, etc. yes, Rockchip/amlogic can make profits with them, but PINE64/Radxa/OrangePi/Odroid arent attached to a vendor, they are able to choose it. 

RPI will always offer Broadcom SoCs (except for the microcontroller sbc), even considering they have horrible blobs and horrible GPU designs that can barely compete with the lowest end bifrost gpus. The I/O isnt great neither! 

You are avoiding the VPU situation completely!!! I have VPU on the browser on rk3399 on legacy kernel from day 1. I do not use legacy bc of mesa of course, but I have both h264/h265 and VPU on the browser. They do not have your 25 engineers. they barely hae a few and not full time jobs!!

HinTak, 2021-05-17:
FWIW, I don't think it is appropriate to draw too much equivalent between RP foundation and the Broadcom commercial and corporate entity. A non-profit organization can certainly receive major sponsorship from one corporate entity, in terms of both actual money, as well as man power, ie a few engineers are allowed to spend months of their working time on "outside" projects. Yes, say corporate entity receives good publicity from "donating" to a non-profit organization; and there are strings attached sometimes. In this case, RP gets broadcom hardware at a cheaper price and a guarantee they will be made and available in the quality needed. Both parties benefited from the arrangement. That's not very different from Bill Gate white-washing his reputation since he left Microsoft.

JamesH65, 2021-05-18:
> @JamesH65 Dear, I did not even compare rpi with Intel, that was another guy. let's begin.
> 
> I can do stuff on RK3399 that would be impossible on RPI4, mostly due that fake kms situation. and that would be the case on other bifrost newer sbcs. also, all of them are far better running 64 bit distributions. it's debatable.
> 
> what I consider disrespectful it's that your brand isn't able to choose, and also that you are skipping the situation that your brand, supposedly an non profit organization to create cheap sbcs for tinkering and learning is completely fake advertising his non profitable goals
> 
> you brand is being lead by a Broadcom engineer that WORK at broadcom SoCs that earn money for broadcom. your brand is Broadcom. and yes, broadcom make tons of money with your brand. PINE64 is a real non profitable organization, they are not attached like your brand, they can choose not to use rockchip if they want, they can choose amlogic, allwinner, etc. yes, Rockchip/amlogic can make profits with them, but PINE64/Radxa/OrangePi/Odroid arent attached to a vendor, they are able to choose it.
> 
> RPI will always offer Broadcom SoCs (except for the microcontroller sbc), even considering they have horrible blobs and horrible GPU designs that can barely compete with the lowest end bifrost gpus. The I/O isnt great neither!
> 
> You are avoiding the VPU situation completely!!! I have VPU on the browser on rk3399 on legacy kernel from day 1. I do not use legacy bc of mesa of course, but I have both h264/h265 and VPU on the browser on lgacy. They do not have your 25 engineers. they barely have a few and not full time jobs!!
> 
> your brand have the media and the markets, bc all the rest of the brands cant compete due trade war taxes and retailer bias against chinese brands. I hope your brand sunk deeply, bc that's the best for ARM Linux. We dont need threadX. We dont need horrible broadcom videocores, we dont need painful I/O, it's time for rockchip. and if other brand does fit the task better and cheaper, then other brand it's. RPI is just a broadcom sbc.
> 
> also, you are lying, Anholt was paid by Broadcom to make the basics of the gpu driver. he himself mention he worked at broadcom. not me. "Prior to that I was at Broadcom working on Raspberry Pi's graphics driver." http://www.anholt.net/
> 
> RPI will sunk eventually due the horrible hardware that broadcom delivers anyway. Also, £31M (2019) with just 100 employees... and you are telling me that's somehow low to justify the poor support that your potato 4 have?
> 
> I will continue working with rpi if they step aside broadcom or broadcom start to make real hardware and avoid his horrible gpus and also add real I/O into the rpi SoCs. until then, rockchip. all the way. I still have some rpi4 laying around so I have to deal with this horrible hardware. but I promise I will not recommend it anymore.
> 
> it's not my fault, you all failed me and many of us when launching this rpi4 with this "upgraded" 32 gflops gpu, 5.15v only power input (not PD), horrible software support taking in count your resources, horrible tearing, no gamma control, not working FULL KMS, not available m.2 , no crypto extensions...etc.
> 
> the rockpi4C is far more featured at similar price (comparing it with the 4 gb ram potato 4 model), at least in china. It use PD, so forget about rpi power issues. it have an m.2 nvme port that reaches 1.6 GB/s due 4 pci lanes 2.0. It have 2 not shared usb 3.0 interfaces. it have a much more powerful gpu than the videocore 6, even being a lot older. it doesnt have ThreadX running all the time underneath. and this year... they will appear the next gen rockchip socs, with much more features that will leave your potato 4 far behind.

I am afraid you are woefully out of date. Eben Upton no longer works for Broadcom. He works for Raspberry Pi (Trading), a wholly owned subsidiary of the Charity, Raspberry Pi Foundation. As for the rest of your post, well, I'll let others decide. Reads like nonsense to me.

But please move to Rock Pi if you wish. No-one here will stop you!