raspberrypi/userland #522: Raspicam: any way to send the video/still preview to an spi monitor?

tesseract241, 2019-01-12:
I have been successfully using an SPI monitor on my raspberry for both CLI and X sessions.  
The raspicam utility though only outputs through the HDMI, but as far as I can see one can create a vc component for an spi monitor just as well as an HDMI one.  
Is there an option hidden somewhere to tell raspicam to use the 2nd framebuffer device? Or is it hard-coded?

JamesH65, 2019-01-12:
The system can ouput the preview to HDMI, composite, DSI and DPI. I don't *think* SPI is directly  possible as that is not connected to the video system. @popcornmix any comments?

You might be able to use the snapshot system but that wil be slower.

6by9, 2019-01-14:
No, there is no easy way to tell the GPU to render to an SPI display.

HDMI, DSI, DPI, and analogue video are all under the control of the GPU by default, and therefore it can render to those displays directly.
SPI is under control of the Linux kernel, and there is no standard path for raspicam to render to that frame buffer.

There is rpi-fbcp (https://github.com/tasanakorn/rpi-fbcp) that uses the DispmanX snapshot API to take a snapshot of the GPU output and replicate it into an alternate framebuffer device. 
(Eric has also had thoughts that the vc4-drm-kms driver may be able to DMA directly from the pixelvalve to SPI, but I don't believe he has ever actually implemented it)

JamesH65, 2019-01-14:
Closing this issue as questions answered/issue resolved.

closed 2019-01-14 by JamesH65