raspberrypi/userland #526: hello_pi runtime errors

jwinarske, 2019-01-15:
In master, I see two runtime errors with hello_pi projects "hello_font", and "hello_tiger".

./hello_font: symbol lookup error: /opt/vc/lib/libbrcmEGL.so: undefined symbol: glPointSizePointerOES
./hello_tiger: symbol lookup error: /opt/vc/lib/libbrcmEGL.so: undefined symbol: glPointSizePointerOES

popcornmix, 2019-01-15:
On either raspian or raspian lite run ``rebuild.sh`` in ``/opt/vc/src/hello_pi``
https://www.raspberrypi.org/documentation/usage/demos/

jwinarske, 2019-01-15:
Using a Raspbian Stretch image (not-Lite), with the legacy driver enabled, it works.

If I have either non-legacy display driver enabled I get "* failed to add service - already in use?".

popcornmix, 2019-01-15:
Yes, these apps only work with the legacy/firmware 3d driver.

closed 2019-01-16 by jwinarske