raspberrypi/userland #532: buildme appears to fail during install in new Kali 2019.1 release

C0ntr07, 2019-02-22:
(This could be a user issue ... :-) )

After running ./buildme this output is received:

'<
/tmp/userland/build/arm-linux/release /tmp/userland
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
CMake Error at CMakeLists.txt:3 (project):
  The CMAKE_C_COMPILER:

    arm-linux-gnueabihf-gcc

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:3 (project):
  The CMAKE_CXX_COMPILER:

    arm-linux-gnueabihf-g++

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

-- Configuring incomplete, errors occurred!
See also "/tmp/userland/build/arm-linux/release/CMakeFiles/CMakeOutput.log".
See also "/tmp/userland/build/arm-linux/release/CMakeFiles/CMakeError.log".
make: *** No targets specified and no makefile found.  Stop.
/tmp/userland>'

Linux kali 4.14.93-Re4son-v8+ #1 SMP PREEMPT Thu Jan 24 03:28:29 UTC 2019 aarch64 GNU/Linux
cmake version 3.13.4

6by9, 2019-02-23:
Kali isn't a supported distribution, but it sounds like you haven't installed the compiler.
"sudo apt install build-essentials" or whatever the Kali equivalent to apt is

Ah, no, you appear to be on a 64bit ARM build, and buildme doesn't have parsing for aarch64. Grab https://github.com/raspberrypi/userland/pull/507 and it may work better.

6by9, 2019-03-07:
buildme has been updated for aarch64. If you are building on a 64-bit userspace then you need to use ```./buildme --aarch64```.