raspberrypi/userland #533: Crash in vcos

TomMettam, 2019-02-25:
I'm experiencing a crash in vcos after starting and stopping a video a few times.  Here's the backtrace:

```
#0  event_flags_timer_expired (cxt=0x6e97bd18)
    at /usr/src/debug/userland/20181120-r0/git/interface/vcos/generic/vcos_generic_event_flags.c:283
        waitreq = 0x6e97bd18
        flags = 0x729252b8
        plist = 0x20119
        prev = 0x20101
        thread = 0x0
#1  0x7181bd3e in _task_timer_expiration_routine (cxt=0x6e0ffb20)
    at /usr/src/debug/userland/20181120-r0/git/interface/vcos/pthreads/vcos_pthreads.c:161
        thread = 0x6e0ffb20
#2  0x7181be36 in _timer_thread (arg=0x6e0ffb40)
    at /usr/src/debug/userland/20181120-r0/git/interface/vcos/pthreads/vcos_pthreads.c:735
        now = {tv_sec = 1551105466, tv_nsec = 230810879}
        timer = 0x6e0ffb40
#3  0x768b3aac in start_thread (arg=0xd3ad406f) at pthread_create.c:486
        ret = <optimized out>
        start = <optimized out>
        pd = 0xd3ad406f
        unwind_buf = {cancel_jmp_buf = {{jmp_buf = {-870730574, -743620497,
                1765790784, 1855437960, 0, 1765789272, 1765790784, 1765789568,
                1765790896, 1855437960, 1000000, 0 <repeats 53 times>},
              mask_was_saved = 0}}, priv = {pad = {0x0, 0x0, 0x0, 0x0},
            data = {prev = 0x0, cleanup = 0x0, canceltype = 0}}}
        not_first_call = <optimized out>
        robust = <optimized out>
#4  0x76850d7c in ?? () at ../sysdeps/unix/sysv/linux/arm/clone.S:73
   from /lib/libc.so.6
--Type <RET> for more, q to quit, c to continue without paging--
No locals.
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

Of course, I am fully open to the likely possibility that this is due to a problem with my code, but I was hoping someone might be able to point me towards what kind of behaviour might cause this problem?

6by9, 2019-02-25:
You're going to have to provide at least some of the calling code.

It looks like you've got a timer callback as part of vcos_generic_event_flags_get happening with invalid context. Are you reusing an event flag without cleaning up first?

TomMettam, 2019-02-25:
I don't really know what the calling code is - it's not happening synchronously with any part of my application. I'm using ilclient to play and stop videos. I'm trying to do further debug now to narrow down exactly when this happens and what triggers it, but without knowing what component it applies to, it's tricky.

6by9, 2019-02-25:
ilclient in itself can't play videos.
It can create the components that mean you can decode and/or render videos, but there has to be external code calling into ilclient.

At a minimum we need a backtrace from all threads (```thread apply all bt``` in gdb), but ideally we need to see how you are calling into ilclient.

You also haven't said what error you are getting when it crashes - segmentation fault, or something else?

TomMettam, 2019-02-25:
I really do appreciate your help, and I will try my best to gather the information required.

The project is actually fully open source, you can view it here: https://github.com/augmentality/node-omx

I'm just not exactly sure which thread is actually "calling" this, since it's on a thread.

```
(gdb) thread apply all bt

Thread 33 (Thread 0x4fa7f440 (LWP 10397)):
#0  __libc_do_syscall () at libc-do-syscall.S:48
#1  0x768bad26 in __lll_lock_wait (futex=futex@entry=0x6d4c25ac, private=0) at lowlevellock.c:46
#2  0x768b5848 in __GI___pthread_mutex_lock (mutex=mutex@entry=0x6d4c25ac) at pthread_mutex_lock.c:78
#3  0x7181c822 in vcos_mutex_lock (latch=0x6d4c25ac) at /usr/src/debug/userland/20181120-r0/git/interface/vcos/pthreads/vcos_platform.h:508
#4  event_flags_timer_expired (cxt=0x6f2a7d18) at /usr/src/debug/userland/20181120-r0/git/interface/vcos/generic/vcos_generic_event_flags.c:257
#5  0x7181bd3e in _task_timer_expiration_routine (cxt=0x12ee6b0) at /usr/src/debug/userland/20181120-r0/git/interface/vcos/pthreads/vcos_pthreads.c:161
#6  0x7181be36 in _timer_thread (arg=0x12ee6d0) at /usr/src/debug/userland/20181120-r0/git/interface/vcos/pthreads/vcos_pthreads.c:735
#7  0x768b3aac in start_thread (arg=0x4387e74) at pthread_create.c:486
#8  0x76850d7c in ?? () at ../sysdeps/unix/sysv/linux/arm/clone.S:73 from /lib/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

Thread 29 (Thread 0x50cff440 (LWP 10393)):
#0  __libc_do_syscall () at libc-do-syscall.S:48
#1  0x768b874a in futex_abstimed_wait_cancelable (private=0, abstime=0x71700a14, expected=0, futex_word=0x71700a08) at ../sysdeps/unix/sysv/linux/futex-internal.h:205
#2  __pthread_cond_wait_common (abstime=0x71700a14, mutex=0x717009c4, cond=0x717009e0) at pthread_cond_wait.c:539
#3  __pthread_cond_timedwait (cond=cond@entry=0x717009e0, mutex=mutex@entry=0x717009c4, abstime=abstime@entry=0x71700a14) at pthread_cond_wait.c:667
#4  0x7181be5e in _timer_thread (arg=0x717009c0) at /usr/src/debug/userland/20181120-r0/git/interface/vcos/pthreads/vcos_pthreads.c:724
#5  0x768b3aac in start_thread (arg=0x4387e74) at pthread_create.c:486
#6  0x76850d7c in ?? () at ../sysdeps/unix/sysv/linux/arm/clone.S:73 from /lib/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

Thread 25 (Thread 0x58aff440 (LWP 10389)):
#0  __libc_do_syscall () at libc-do-syscall.S:48
#1  0x768b84ba in futex_wait_cancelable (private=0, expected=0, futex_word=0x6e1ffb2c) at ../sysdeps/unix/sysv/linux/futex-internal.h:88
#2  __pthread_cond_wait_common (abstime=0x0, mutex=<optimized out>, cond=0x6e1ffb00) at pthread_cond_wait.c:502
#3  __pthread_cond_wait (cond=cond@entry=0x6e1ffb00, mutex=mutex@entry=0x6e1ffae4) at pthread_cond_wait.c:655
#4  0x7181be0a in _timer_thread (arg=0x6e1ffae0) at /usr/src/debug/userland/20181120-r0/git/interface/vcos/pthreads/vcos_pthreads.c:722
#5  0x768b3aac in start_thread (arg=0x4387e74) at pthread_create.c:486
--Type <RET> for more, q to quit, c to continue without paging--
#6  0x76850d7c in ?? () at ../sysdeps/unix/sysv/linux/arm/clone.S:73 from /lib/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

Thread 21 (Thread 0x68df7440 (LWP 10359)):
#0  __libc_do_syscall () at libc-do-syscall.S:48
#1  0x768b874a in futex_abstimed_wait_cancelable (private=0, abstime=0x68df6cc8, expected=0, futex_word=0x6e117390) at ../sysdeps/unix/sysv/linux/futex-internal.h:205
#2  __pthread_cond_wait_common (abstime=0x68df6cc8, mutex=0x6e117398, cond=0x6e117368) at pthread_cond_wait.c:539
#3  __pthread_cond_timedwait (cond=0x6e117368, mutex=0x6e117398, abstime=0x68df6cc8) at pthread_cond_wait.c:667
#4  0x718ddacc in __gthread_cond_timedwait (__cond=0x6e117368, __mutex=0x6e117398, __abs_timeout=0x68df6cc8) at /usr/include/c++/8.2.0/arm-yoe-linux-gnueabi/bits/gthr-default.h:871
#5  0x718df7a8 in std::condition_variable::__wait_until_impl<std::chrono::duration<long long, std::ratio<1ll, 1000000000ll> > > (this=0x6e117368, __lock=..., __atime=...) at /usr/include/c++/8.2.0/condition_variable:178
#6  0x718deefc in std::condition_variable::wait_until<std::chrono::duration<long long, std::ratio<1ll, 1000000000ll> > > (this=0x6e117368, __lock=..., __atime=...) at /usr/include/c++/8.2.0/condition_variable:106
#7  0x718e1e78 in std::condition_variable::wait_until<std::chrono::_V2::system_clock, std::chrono::duration<long long int, std::ratio<1, 1000000000> >, AudioThread::waitForBuffer()::<lambda()> >(std::unique_lock<std::mutex> &, const std::chrono::time_point<std::chrono::_V2::system_clock, std::chrono::duration<long long, std::ratio<1, 1000000000> > > &, AudioThread::<lambda()>) (this=0x6e117368, __lock=..., __atime=..., __p=...)
    at /usr/include/c++/8.2.0/condition_variable:129
#8  0x718e1c98 in std::condition_variable::wait_for<long long int, std::ratio<1, 1000>, AudioThread::waitForBuffer()::<lambda()> >(std::unique_lock<std::mutex> &, const std::chrono::duration<long long, std::ratio<1, 1000> > &, AudioThread::<lambda()>) (this=0x6e117368, __lock=..., __rtime=..., __p=...) at /usr/include/c++/8.2.0/condition_variable:156
#9  0x718e1790 in AudioThread::waitForBuffer (this=0x6e117280) at /augmentality/node_modules/@augmentality/node-omx/libomx/AudioThread.cpp:262
#10 0x718e17e0 in AudioThread::addData (this=0x6e117280, ab=0x65c7dc58) at /augmentality/node_modules/@augmentality/node-omx/libomx/AudioThread.cpp:270
[... cut ...]
#2  0x718d751c in NativePlayer::~NativePlayer (this=0x6d8edba0, __in_chrg=<optimized out>) at /augmentality/node_modules/@augmentality/node-omx/libomx/NativePlayer.cpp:276
#3  0x718d33a8 in Player::stop (info=...) at /augmentality/node_modules/@augmentality/node-omx/Player.cpp:317
#4  0x718cf9a0 in Nan::imp::FunctionCallbackWrapper (info=...) at /augmentality/node_modules/nan/nan_callbacks_12_inl.h:176
#5  0x0083ce68 in v8::internal::FunctionCallbackArguments::Call(void (*)(v8::FunctionCallbackInfo<v8::Value> const&)) ()
#6  0x008a0bbc in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

```

TomMettam, 2019-02-25:
Thread 21:  AudioThread::waitForBuffer is the input thread waiting for available buffer space in the AudioThread
Thread 20:  Is the video output thread sleeping while waiting for an available buffer from openmax
Thread 19:  is the Audio output thread doing the same /\

Thread 10-12:  appear to be previous player instances which are hung waiting for ILCLIENT_BUFFER_FLAG_EOS.. 

I guess this is what is causing the segfault. I'm not sure why it's hanging

6by9, 2019-02-25:
I suspect something is calling or has called vcos_event_wait with a timeout value (there seem to be a couple of candidates for that). The timer has been setup, and this is the callback that is in progress. However when the timeout expires it hits invalid information for signalling back to the waiting thread.

I haven't checked, and I can't remember for certain, but I don't recall ilclient being used from a multithreaded app, and yet I think you are using it as such (creating one ilclient in NativePlayer and then using it from threads for each of video and audio decoding). As a quick test I'd try creating an ilclient per thread/decoding pipe. Do remember to clean up afterwards.

TomMettam, 2019-02-25:
Thanks for the tips. I don't think that's the cause, though - Calling OMX_Init on secondary threads causes things to break, and besides, OMXPlayer only calls OMX_Init once, on the main thread.

6by9, 2019-02-25:
```OMX_Init``` != ```ilclient_init```

I'd assumed from the library name that you were actually using the supplied [ilclient library](https://github.com/raspberrypi/userland/tree/master/host_applications/linux/apps/hello_pi/libs/ilclient), but I now see you're reimplementing it. Sorry, you really are on your own there.

I'm saying you want to call ILClient::init per thread, not your constructor (which calls bcm_host_init and OMX_Init).

TomMettam, 2019-02-26:
I found the issue. I was double-deleting the NativePlayer (the destructor was triggering a playbackCompleted event which was deleting the nativeplayer again, corrupting the stack). Thanks again for your help and time looking at this!

closed 2019-02-26 by TomMettam