raspberrypi/userland #545: Enum usage mixup

lurch, 2019-03-22:
An error spotted by a `case value '1' not in enumerated type` compiler warning...

In https://github.com/raspberrypi/userland/blob/master/interface/vmcs_host/vc_service_common.c the `vchi2service_status_string` function takes a `VC_SERVICE_VCHI_STATUS_T` parameter. However in the `switch` statement inside that function it then checks against enum values from the `VCHIQ_STATUS_T` type.

`VCHIQ_STATUS_T` is defined in https://github.com/raspberrypi/userland/blob/master/interface/vchiq_arm/vchiq_if.h as:
```c
typedef enum
{
   VCHIQ_ERROR   = -1,
   VCHIQ_SUCCESS = 0,
   VCHIQ_RETRY   = 1
} VCHIQ_STATUS_T;
```
whereas `VC_SERVICE_VCHI_STATUS_T` is defined in https://github.com/raspberrypi/userland/blob/master/interface/vmcs_host/vc_service_common.h as:
```c
typedef enum {
   VC_SERVICE_VCHI_SUCCESS = 0,
   VC_SERVICE_VCHI_VCHIQ_ERROR = -1,
   VC_SERVUCE_VCHI_RETRY = -2,
   VC_SERVICE_VCHI_UNKNOWN_ERROR = -3
} VC_SERVICE_VCHI_STATUS_T;
```

pelwell, 2019-03-22:
The RETRY value will never get as far as the user level, but feel free to explicitly mention it in the switch statement to shut the compiler up.

lurch, 2019-03-23:
Hmmm, AIUI it's actually the other way round, i.e. the signature for `vchi2service_status_string` indicates it takes a `VC_SERVICE_VCHI_STATUS_T` (with valid values 0, -1, -2, -3) but then inside the function it's actually [compared](https://github.com/raspberrypi/userland/blob/master/interface/vmcs_host/vc_service_common.c#L51) against `VCHIQ_RETRY` i.e. 1, which is a value that the function never expects.

I dunno if the "fix" might be to change the function signature to `const char* vchi2service_status_string(VCHIQ_STATUS_T status)` ? :man_shrugging: