raspberrypi/userland #547: raspistill makes errant call to nanosleep in timelapse mode

stewartoallen, 2019-03-24:
it seemed like raspistill was locking up in timelapse mode.  but stracing it revealed that it was, in fact, just sleeping for a long time.  there seems to be a calculation error (maybe on a missed frame) that results in a huge value for tv_sleep in nanosleep every once in a while.  

this seems to happen far more often with the noir pi camera module than the standard v2.

stewartoallen, 2019-03-24:
example of strace leading up to this:

```
nanosleep({tv_sec=0, tv_nsec=671000000}, NULL) = 0
open("/tmp/camera.jpg~", O_WRONLY|O_CREAT|O_TRUNC|O_LARGEFILE, 0666) = 5
ioctl(4, _IOC(_IOC_WRITE, 0xc4, 0x04, 0x0c), 0x7ed42ec4) = 0
ioctl(4, _IOC(_IOC_WRITE, 0xc4, 0x04, 0x0c), 0x7ed42ec4) = 0
futex(0x76f193bc, FUTEX_WAIT_BITSET_PRIVATE|FUTEX_CLOCK_REALTIME, 1, NULL, 0xffffffff) = -1 EAGAIN (Resource temporarily unavailable)
stat64("/etc/localtime", {st_mode=S_IFREG|0644, st_size=3545, ...}) = 0
ioctl(4, _IOC(_IOC_WRITE, 0xc4, 0x04, 0x0c), 0x7ed42ec4) = 0
ioctl(4, _IOC(_IOC_WRITE, 0xc4, 0x04, 0x0c), 0x7ed42ec4) = 0
futex(0x76f193bc, FUTEX_WAIT_BITSET_PRIVATE|FUTEX_CLOCK_REALTIME, 1, NULL, 0xffffffff) = -1 EAGAIN (Resource temporarily unavailable)
ioctl(4, _IOC(_IOC_WRITE, 0xc4, 0x04, 0x0c), 0x7ed42ec4) = 0
ioctl(4, _IOC(_IOC_WRITE, 0xc4, 0x04, 0x0c), 0x7ed42eac) = 0
ioctl(4, _IOC(_IOC_WRITE, 0xc4, 0x04, 0x0c), 0x7ed42fbc) = 0
futex(0x76f193bc, FUTEX_WAIT_BITSET_PRIVATE|FUTEX_CLOCK_REALTIME, 1, NULL, 0xffffffff) = 0
ioctl(4, _IOC(_IOC_WRITE, 0xc4, 0x04, 0x0c), 0x7ed430a4) = 0
ioctl(4, _IOC(_IOC_WRITE, 0xc4, 0x04, 0x0c), 0x7ed42eac) = 0
futex(0x76f193bc, FUTEX_WAIT_BITSET_PRIVATE|FUTEX_CLOCK_REALTIME, 1, NULL, 0xffffffff) = -1 EAGAIN (Resource temporarily unavailable)
futex(0x7ed431c4, FUTEX_WAIT_BITSET_PRIVATE|FUTEX_CLOCK_REALTIME, 1, NULL, 0xffffffff) = 0
write(5, "\17\347TN]R8\321U\201\332I\347\0\16\376\365^\366i6\225\237\312\302\360Q\270\372\22{\325"..., 1596) = 1596
close(5)                                = 0
rename("/tmp/camera.jpg~", "/tmp/camera.jpg") = 0
link("/tmp/camera.jpg", "/var/www/html/camera.jpg~") = 0
rename("/var/www/html/camera.jpg~", "/var/www/html/camera.jpg") = 0
ioctl(4, _IOC(_IOC_WRITE, 0xc4, 0x04, 0x0c), 0x7ed42fec) = 0
futex(0x76f193bc, FUTEX_WAIT_BITSET_PRIVATE|FUTEX_CLOCK_REALTIME, 1, NULL, 0xffffffff) = 0
futex(0x10f0c18, FUTEX_WAKE_PRIVATE, 1) = 1
clock_gettime(CLOCK_MONOTONIC_RAW, {tv_sec=8589, tv_nsec=755716462}) = 0
nanosleep({tv_sec=0, tv_nsec=666000000}, NULL) = 0
open("/tmp/camera.jpg~", O_WRONLY|O_CREAT|O_TRUNC|O_LARGEFILE, 0666) = 5
ioctl(4, _IOC(_IOC_WRITE, 0xc4, 0x04, 0x0c), 0x7ed42ec4) = 0
ioctl(4, _IOC(_IOC_WRITE, 0xc4, 0x04, 0x0c), 0x7ed42ec4) = 0
stat64("/etc/localtime", {st_mode=S_IFREG|0644, st_size=3545, ...}) = 0
ioctl(4, _IOC(_IOC_WRITE, 0xc4, 0x04, 0x0c), 0x7ed42ec4) = 0
ioctl(4, _IOC(_IOC_WRITE, 0xc4, 0x04, 0x0c), 0x7ed42ec4) = 0
ioctl(4, _IOC(_IOC_WRITE, 0xc4, 0x04, 0x0c), 0x7ed42ec4) = 0
ioctl(4, _IOC(_IOC_WRITE, 0xc4, 0x04, 0x0c), 0x7ed42eac) = 0
ioctl(4, _IOC(_IOC_WRITE, 0xc4, 0x04, 0x0c), 0x7ed42fbc) = 0
futex(0x76f193bc, FUTEX_WAIT_BITSET_PRIVATE|FUTEX_CLOCK_REALTIME, 1, NULL, 0xffffffff) = 0
ioctl(4, _IOC(_IOC_WRITE, 0xc4, 0x04, 0x0c), 0x7ed430a4) = 0
ioctl(4, _IOC(_IOC_WRITE, 0xc4, 0x04, 0x0c), 0x7ed42eac) = 0
futex(0x7ed431c4, FUTEX_WAIT_BITSET_PRIVATE|FUTEX_CLOCK_REALTIME, 1, NULL, 0xffffffff) = 0
write(5, "\203\375*{\313P\212Z2\253\271z\21\367{\322Z\337Kq\23n\v\32(\4eI'\266kB"..., 2723) = 2723
close(5)                                = 0
rename("/tmp/camera.jpg~", "/tmp/camera.jpg") = 0
link("/tmp/camera.jpg", "/var/www/html/camera.jpg~") = 0
rename("/var/www/html/camera.jpg~", "/var/www/html/camera.jpg") = 0
ioctl(4, _IOC(_IOC_WRITE, 0xc4, 0x04, 0x0c), 0x7ed42fec) = 0
futex(0x76f193bc, FUTEX_WAIT_BITSET_PRIVATE|FUTEX_CLOCK_REALTIME, 1, NULL, 0xffffffff) = 0
futex(0x10f0c18, FUTEX_WAKE_PRIVATE, 1) = 1
clock_gettime(CLOCK_MONOTONIC_RAW, {tv_sec=8590, tv_nsec=762927243}) = 0
nanosleep({tv_sec=1271310, tv_nsec=978000000}, 
```

stewartoallen, 2019-03-24:
interestingly, i had two freezes in a row on separate PIs with different camera types (standard v2 vs noir).  both stopped with the exact same sleep value:

```
futex(0x76e843bc, FUTEX_WAIT_BITSET_PRIVATE|FUTEX_CLOCK_REALTIME, 1, NULL, 0xffffffff) = 0
futex(0x7ee0a1c4, FUTEX_WAIT_BITSET_PRIVATE|FUTEX_CLOCK_REALTIME, 1, NULL, 0xffffffff) = 0
write(5, "f`K2/^q\267\7\256ON1Pn\6 \10\\0$\261=OA\214\177\236(oRL"..., 2178) = 2178
close(5)                                = 0
rename("/tmp/camera.jpg~", "/tmp/camera.jpg") = 0
link("/tmp/camera.jpg", "/var/www/html/camera.jpg~") = 0
rename("/var/www/html/camera.jpg~", "/var/www/html/camera.jpg") = 0
ioctl(4, _IOC(_IOC_WRITE, 0xc4, 0x04, 0x0c), 0x7ee09fec) = 0
futex(0x76e843bc, FUTEX_WAIT_BITSET_PRIVATE|FUTEX_CLOCK_REALTIME, 1, NULL, 0xffffffff) = 0
futex(0x9bbc18, FUTEX_WAKE_PRIVATE, 1)  = 1
clock_gettime(CLOCK_MONOTONIC_RAW, {tv_sec=17180, tv_nsec=267202092}) = 0
```

JamesH65, 2019-03-24:
I thought this had been fixed, but its possible that its not been released yet. But try an update first, then if that doesn't work, you can download the source from our github and rebuild it yourself.

6by9, 2019-03-25:
#535 fixed an overflow issue in timelapse mode. Added to rpi-update releases since 8th March, but probably not in apt.
Please confirm the version you are running (```raspistill -v``` gives you the git hash for the source tree).

stewartoallen, 2019-03-25:
I won't have access to the affected PI+cameras for a few hours. They've just been built/updated to the latest. It seem likely if the fix was recent that it's not been released yet. Thanks.

lurch, 2019-07-08:
@stewartoallen Have you been able to re-test with the latest version of `raspistill` yet?