raspberrypi/userland #548: Unable to find libvcfiled_check.a in Raspbian

joolswills, 2019-03-25:
Apologies if this is the wrong place but wasn't sure where to ask.

I am unable to locate `/opt/vc/lib/libvcfiled_check.a` on a current Raspbian Stretch Lite with libraspberry-pi* packages installed. Has it be removed or am I missing a package (I used apt-file to search but couldn't locate it)?

running Raspbian Stretch on a RPI3.

libraspberrypi-bin  (1.20190215-1).
libraspberrypi-dev (1.20190215-1).
libraspberrypi-doc (1.20190215-1).

6by9, 2019-03-25:
vcfiled was removed from the build in January as it is irrelevant on the Pi.
What is your need for vcfiled?

(Oddly the relevant commit appears not to have been merged here, only in our firmware tree - those need to be resynced).

joolswills, 2019-03-25:
https://github.com/raspberrypi/quake3 uses it.

6by9, 2019-03-25:
OK, quake3 needs a cleanup then.

vcfiled is a VCHI service that allows the VPU to access the host side (ARM) filing system. 
It may initialise the ARM side of the service, but the firmware doesn't have the code to implement that side of the RPC, so it's going to do nothing.

joolswills, 2019-03-25:
Thanks.

It builds fine without the library. The quake3 code also references the old bcm library names. I will submit a PR for that repo.

closed 2019-03-25 by joolswills