raspberrypi/userland #550: Raspistill/MMALSharp annotation text colour

techyian, 2019-04-08:
Hi,

I have a bug raised over at the MMALSharp repo https://github.com/techyian/MMALSharp/issues/83 and I'm also seeing similar behaviour in raspistill too. For reference the raspistill command I'm using is:

`raspistill -o test.jpg -ae 32,0xff0000,0x808000 -a "Annotation text"`

Looking at the official docs, I'd hope this would print out the text in red with a black background, however it seems to be a dark grey. This is what I'm seeing in MMALSharp too.

I am able to set the background colour successfully in MMALSharp so I'm quite sure the colour conversion functions I have in place are working correctly.

Please could someone take a quick look at this to confirm if it's an issue with the firmware?

Thanks.

6by9, 2019-04-09:
The foreground colour only applies the Y (luma) component. It'll adopt the U&V (chroma) from the background.

Due to the subsampling in the chroma components it's very hard to avoid aliasing artefacts without a making it a complete 3 component font.

techyian, 2019-04-11:
Thanks for confirming. Has it always behaved this way as I was pretty sure I've been able to set the text colour separately from the background in the past? I haven't tested this area since before the move to the MMAL_PARAMETER_CAMERA_ANNOTATE_V4_T struct so didn't know if something had regressed.

6by9, 2019-04-11:
> Has it always behaved this way 

Yes.
https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=97929&start=25#p701272
> Custom background colour and text size are both doing the right thing. I've even done raspicam changes to test it out - those'll get pushed to my github later, along with a test firmware.
I haven't plumbed in custom text colour as only Y appears to be used - I'll probably still do the plumbing as you could then at least alter the Y for dimmer text overlay. Doing U&V will be more awkward due to the chroma subsampling.

techyian, 2019-04-11:
Thanks very much, I'll close the ticket off.

closed 2019-04-11 by techyian