raspberrypi/userland #551: /bin/rm segmentation fault

bt35, 2019-04-18:
### Description
I was testing Hassio on my Pi Zero W. At some point well after Hassio was installed and running the WIFI stopped working (carrier drops). When I was troubleshooting that issue I found that dhcpcd was constantly trying to clean stale PIDs related to /sbin/resolvconf. Eventually dhcpcd would timeout and fail to bring up network interfaces. I thought that manually removing the stale PID files and lock files would make it so dhcpcd could start. That was when I found that **/bin/rm** would segfault when I tried to remove any file. 

### Steps to Reproduce 
I am not sure there are any steps to reproduce as I am not sure what might have caused the issue in the first place. However I was hoping that someone might be able to look at the strace, or GBD output and give me some hints. 

Here is a snippet of journalctl -u dhcpcd as it was trying to kill stale PIDs. Output below is chopped due to using the UART Serial connection and minicom.

```
Apr 17 02:08:44 raspberrypi dhcpcd[204]: clearing stale lock pid 380
Apr 17 02:08:44 raspberrypi dhcpcd[204]: Segmentation fault
Apr 17 02:08:44 raspberrypi dhcpcd[204]: /sbin/resolvconf: 733: kill: No such pr
Apr 17 02:08:44 raspberrypi dhcpcd[204]: clearing stale lock pid 380
Apr 17 02:08:44 raspberrypi dhcpcd[204]: Segmentation fault
Apr 17 02:08:44 raspberrypi dhcpcd[204]: /sbin/resolvconf: 733: kill: No such pr
Apr 17 02:08:44 raspberrypi dhcpcd[204]: clearing stale lock pid 380
Apr 17 02:08:44 raspberrypi dhcpcd[204]: Segmentation fault
Apr 17 02:08:44 raspberrypi dhcpcd[204]: /sbin/resolvconf: 733: kill: No such pr
Apr 17 02:08:44 raspberrypi dhcpcd[204]: clearing stale lock pid 380
Apr 17 02:08:44 raspberrypi dhcpcd[204]: Segmentation fault
Apr 17 02:08:44 raspberrypi dhcpcd[204]: /sbin/resolvconf: 733: kill: No such pr
Apr 17 02:08:44 raspberrypi dhcpcd[204]: clearing stale lock pid 380
```

Snippet of the logs where the carrier drops are seen. 
```
Apr 17 02:07:26 raspberrypi dhcpcd[204]: wlan0: waiting for carrier
Apr 17 02:07:26 raspberrypi dhcpcd[204]: wlan0: carrier acquired
Apr 17 02:07:26 raspberrypi dhcpcd[204]: DUID
Apr 17 02:07:26 raspberrypi dhcpcd[204]: wlan0: IAID
Apr 17 02:07:26 raspberrypi dhcpcd[204]: wlan0: adding address
Apr 17 02:07:26 raspberrypi dhcpcd[204]: wlan0: carrier lost
Apr 17 02:07:26 raspberrypi dhcpcd[204]: Segmentation fault
Apr 17 02:07:26 raspberrypi dhcpcd[204]: Segmentation fault
Apr 17 02:07:26 raspberrypi dhcpcd[204]: wlan0: deleting address
Apr 17 02:07:26 raspberrypi dhcpcd[204]: wlan0: carrier acquired
Apr 17 02:07:26 raspberrypi dhcpcd[204]: wlan0: IAID
Apr 17 02:07:26 raspberrypi dhcpcd[204]: wlan0: adding address
Apr 17 02:07:26 raspberrypi dhcpcd[204]: wlan0: carrier lost
Apr 17 02:07:26 raspberrypi dhcpcd[204]: /sbin/resolvconf: 733: k
```

###  Platform Details:
```
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

### Strace of /bin/rm as it Segfaults
```
execve("/bin/rm", ["rm", "stuff"], [/* 19 vars */]) = 0
brk(NULL)                               = 0x1e15000
uname({sysname="Linux", nodename="raspberrypi", ...}) = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
access("/etc/ld.so.preload", R_OK)      = 0
open("/etc/ld.so.preload", O_RDONLY|O_CLOEXEC) = 3
fstat64(3, {st_mode=S_IFREG|0644, st_size=42, ...}) = 0
mmap2(NULL, 42, PROT_READ|PROT_WRITE, MAP_PRIVATE, 3, 0) = 0xb6f92000
close(3)                                = 0
open("/usr/lib/arm-linux-gnueabihf/libarmmem.so", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\210\5\0\0004\0\0\0"..., 512) = 512
lseek(3, 20868, SEEK_SET)               = 20868
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1000) = 1000
lseek(3, 20540, SEEK_SET)               = 20540
read(3, "A,\0\0\0aeabi\0\1\"\0\0\0\0056\0\6\6\10\1\t\1\n\3\f\1\22\4\24"..., 45) = 45
fstat64(3, {st_mode=S_IFREG|0644, st_size=21868, ...}) = 0
mmap2(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6f90000
mmap2(NULL, 86080, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6f4d000
mprotect(0xb6f52000, 61440, PROT_NONE)  = 0
mmap2(0xb6f61000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x4000) = 0xb6f61000
close(3)                                = 0
munmap(0xb6f92000, 42)                  = 0
open("/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
fstat64(3, {st_mode=S_IFREG|0644, st_size=39495, ...}) = 0
mmap2(NULL, 39495, PROT_READ, MAP_PRIVATE, 3, 0) = 0xb6f86000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
open("/lib/arm-linux-gnueabihf/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0@h\1\0004\0\0\0"..., 512) = 512
lseek(3, 1231820, SEEK_SET)             = 1231820
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 2880) = 2880
lseek(3, 1228284, SEEK_SET)             = 1228284
read(3, "A.\0\0\0aeabi\0\1$\0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\23\1\24"..., 47) = 47
fstat64(3, {st_mode=S_IFREG|0755, st_size=1234700, ...}) = 0
mmap2(NULL, 1303872, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6e0e000
mprotect(0xb6f38000, 61440, PROT_NONE)  = 0
mmap2(0xb6f47000, 12288, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x129000) = 0xb6f47000
mmap2(0xb6f4a000, 9536, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0xb6f4a000
close(3)                                = 0
set_tls(0xb6f90ed0, 0xb6f915a8, 0xb6f94050, 0xb6f90ed0, 0xb6f94050) = 0
mprotect(0xb6f47000, 8192, PROT_READ)   = 0
mprotect(0xb6f4d000, 20480, PROT_READ|PROT_WRITE) = 0
mprotect(0xb6f4d000, 20480, PROT_READ|PROT_EXEC) = 0
cacheflush(0xb6f4d000, 0xb6f52000, 0, 0x15, 0) = 0
mprotect(0xb6f61000, 4096, PROT_READ)   = 0
mprotect(0x2b000, 4096, PROT_READ)      = 0
mprotect(0xb6f93000, 4096, PROT_READ)   = 0
munmap(0xb6f86000, 39495)               = 0
brk(NULL)                               = 0x1e15000
brk(0x1e36000)                          = 0x1e36000
open("/usr/lib/locale/locale-archive", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fstat64(3, {st_mode=S_IFREG|0644, st_size=1679808, ...}) = 0
mmap2(NULL, 1679808, PROT_READ, MAP_PRIVATE, 3, 0) = 0xb6c73000
close(3)                                = 0
ioctl(0, TCGETS, {B115200 opost isig icanon echo ...}) = 0
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=NULL} ---
+++ killed by SIGSEGV +++
```
### Core Dump and rm with symbols viewed in GDB

Below is GDB output when running the core and /bin/rm with symbols. (Compiled manually on another raspberry Pi).

I think the crash happens at fts_open. Sadly I know little to nothing about what the output below means and I know next to nothing about Arm/Assembly. I am hoping to learn something from this though so any insight would be awesome. 

```
GNU gdb (Raspbian 7.12-6) 7.12.0.20161007-git
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from rm...done.
[New LWP 31990]
Cannot access memory at address 0x7
Cannot access memory at address 0x3
Core was generated by `rm test'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x000172b4 in rpl_fts_open (argv=0x0, options=600, compar=0xbe90cea7) at lib/fts.c:439
439               if (! fts_palloc(sp, MAX(maxarglen, MAXPATHLEN)))
(gdb) list
434     #ifndef MAXPATHLEN
435     # define MAXPATHLEN 1024
436     #endif
437             {
438               size_t maxarglen = fts_maxarglen(argv);
439               if (! fts_palloc(sp, MAX(maxarglen, MAXPATHLEN)))
440                       goto mem1;
441             }
442
443             /* Allocate/initialize root's parent. */
(gdb) where
#0  0x000172b4 in rpl_fts_open (argv=0x0, options=600, compar=0xbe90cea7) at lib/fts.c:439
#1  0x00015cdc in freadahead (fp=0x1ceb860) at lib/freadahead.c:38
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb)
```
Also is this the right place to ask bout this?

lurch, 2019-07-08:
> Also is this the right place to ask bout this?

IMHO this isn't the right place to ask about this. Try the RPi forums https://www.raspberrypi.org/forums/ or maybe it's a problem specific to Hassio ? https://www.home-assistant.io/help/

> I am not sure there are any steps to reproduce

I think without that, it's going to be _very_ hard for anybody to look into this for you :confused:

popcornmix, 2019-07-08:
Yes, this repo if for the software for communicating with GPU in /opt/vc.
This isn't related to that.

if /bin/rm is failing then you have a problem with your install.
A wild guess would be libc (or other libs) are not the version /bin/rm is expecting.

But as @lurch says asking people who know about your distribution (Hassio or forum users) would be a better place.

closed 2019-07-08 by popcornmix