raspberrypi/userland #556: [RFE] Support libglvnd for ES2 and EGL

kwizart, 2019-06-25:
I wonder if it makes sense to support libglvnd for the userland part and EGL/GLES2 libraries (at least).

This way, it would be possible to switch to the appropriate libGL when using v3d or the raspberry-pi-vc userspace or even use softpipe driver (when using vncserver).

popcornmix, 2019-06-25:
There is more involved with using the firmware gl driver than just switching to different EGL/GLES2 libs (e.g. dispmanx calls to create a surface).

It's not something we are likely to do (the future is the arm side driver) but if you were interested in creating a PR we'd look into it.

kwizart, 2019-07-05:
I've made a quick test on raspbian 10 and I'm not sure to understand how is possible to switch between v3d/vc4 (libdrm/mesa) over the firmware version (in /boot/config.txt) without libglvnd support (or similar feature).

XECDesign, 2019-07-08:
More discussion here: https://github.com/RPi-Distro/repo/issues/134#issuecomment-477633553