raspberrypi/userland #558: fatrace fails to start

travellingkiwi, 2019-07-07:
Attempting to use fatrace after installation results in

```
root@piman-401:/home/hamish# fatrace -f W
Cannot initialize fanotify: Invalid argument
root@piman-401:/home/hamish#
```

This doesn't appear to be a new issue (It is the same on my Pi-3B+). There's reports of the same issue from 2015 and 2018 for debian.

https://bugs.launchpad.net/fatrace/+bug/1755787

JamesH65, 2019-07-07:
Not sure why you are reporting this here. It's a problem with fatrace itself (what ever that is), rather than Raspberry Pi specific, so should be reported to the fatrace maintainer.

pmbdk, 2020-01-08:
This issue was fixed in release 0.14 (as of 2019-09-30) of fatrace. The latest release is 0.15 which also works. Until this is updated in raspbian it is very easy to install the tool manually: 

Get r0.15 from https://launchpad.net/fatrace/trunk/0.15/+download/fatrace-0.15.tar.xz
Untar like: tar xf fatrace-0.15.tar.xz
Build: make
Use: ./fatrace

closed 2020-01-08 by 6by9