raspberrypi/userland #562: Wifi script not able to handle both DHCP option 3 and option 121.

sliddjur, 2019-07-19:
**Describe the bug**
On my Windows server 2012 DHCP server I run both `option 3 - default gateway` and `option 121 classless route`
When I have both of them running, the wifi script only got the option 121 route. The ethernet script got the default route.

**To reproduce**
Configure raspberry pi 4 to connect to wifi via the raspi-config.
If I removed the option 121 route, the raspberry pi got the default gateway as expected over wifi.

**Expected behaviour**
I expected raspbian to handle both routes given by DHCP server

**Actual behaviour**
It only gave me one of the routes

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
`raspberry pi 4`
* Which OS and version (`cat /etc/rpi-issue`)?

```
Raspberry Pi reference 2019-07-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 175dfb027ffabd4b8d5080097af0e51ed9a4a56c, stage2
```

* Which firmware version (`vcgencmd version`)?

```
pi@raspberrypi:~ $ vcgencmd version
Jul  9 2019 14:37:58 
Copyright (c) 2012 Broadcom
version d2b1b7fb01475cb3914b2086299e32d724e832f1 (clean) (release) (start)
```

* Which kernel version (`uname -a`)?

```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.19.57-v7l+ #1244 SMP Thu Jul 4 18:48:07 BST 2019 armv7l GNU/Linux
```

**Logs**
Cannot find any wireless logs in dmesg

**Additional context**
No more

lurch, 2019-08-27:
I don't think this is a userland issue.
@XECDesign Does this sound like a [Raspbian](https://github.com/RPi-Distro/repo) or [kernel](https://github.com/raspberrypi/linux) issue?

XECDesign, 2019-08-27:
I don't know enough about DHCP to know off the top of my head, but if nobody else chimes in, I can dig in later and find out.

sliddjur, 2019-08-30:
This works as expected on my workstation, which is a Ubuntu 19.04.
It looks like there is something in the IP script that doesnt handle both options properly.

blistovmhz, 2020-02-27:
Same issue here since buster update.
If Options 121 is set, Option 3 is ignored entirely. I haven't confirmed yet if this only affects wireless.

cbarreholm, 2020-10-29:
Just to confirm.
I'm running Ubiquiti Dream Machine Pro and have configured the DHCP option 121. Noticed that when installing a Raspberry Pi 4 and Zero W with same Raspberry Pi OS released 2020-08-20 I got no default route. This works fine on my Ubuntu 20.04 Laptop.

That said did get around this by extending my DHCP 121 option to include the default route. Assuming that 192.168.0.1 is my gateway I ran this script to generate the option:
`
$ ./rfc3442.route-4-dhcp.pl 0.0.0.0/0 192.168.0.1 192.168.20.0/24 192.168.10.1
option_121_route_0.0.0.0/0_via_192.168.0.1 : 0x00c0a80001
option_249_route_0.0.0.0/0_via_192.168.0.1 : 0x00c0a80001
option_121_route_192.168.20.0/24_via_192.168.10.1 : 0x18c0a814c0a80a01
option_249_route_192.168.20.0/24_via_192.168.10.1 : 0x18c0a814c0a80a01
aggregate_opt_121 : 0x00c0a8000118c0a814c0a80a01
aggregate_opt_249 : 0x00c0a8000118c0a814c0a80a01
`
When I only had option `0x18c0a814c0a80a01` it worked find with my Ubuntu laptop. But now I changed it to `0x00c0a8000118c0a814c0a80a01` which resulted in my Pi getting a default route.

HinTak, 2020-10-30:
I don't think it is a userland issue either. On many linux systems (possibly specifically raspbian), dhcpcd, or the demon (network manager? Netplan?) that's responsible for configuring the network interfaces based on media sense (i.e. plugging in a live ethernet cable), the detection of a live ethernet cable would automatically assign default route / gateway through it, since a cable connection is very much lively to be better than a wifi one, when both are available. If you want to use both, and manually configure them, you might have to do your ethernet first and/or find out  what other processes are trying to configure your network interfaces and shut them all down.