raspberrypi/userland #564: raspistill hangs after running a couple of times

rolandjitsu, 2019-07-24:
After running a few times the `raspistill -v -o test.jpg -t 1000` cmd just hangs at:
```
"raspistill" Camera App (commit 7cbfbd38d982 Tainted)

Camera Name imx219
Width 3280, Height 2464, filename test.jpg
Using camera 0, sensor mode 0

GPS output Disabled

Quality 85, Raw no
Thumbnail enabled Yes, width 64, height 48, quality 35
Time delay 1000, Timelapse 0
Link to latest frame enabled  no
Full resolution preview No
Capture method : Single capture

Preview Yes, Full screen Yes
Preview window 0,0,1024,768
Opacity 255
Sharpness 0, Contrast 0, Brightness 50
Saturation 0, ISO 0, Video Stabilisation No, Exposure compensation 0
Exposure Mode 'auto', AWB Mode 'auto', Image Effect 'none'
Flicker Avoid Mode 'off'
Metering Mode 'average', Colour Effect Enabled No with U = 128, V = 128
Rotation 0, hflip No, vflip No
ROI x 0.000000, y 0.000000, w 1.000000 h 1.000000
```

This is the log from a prev run:
```
"raspistill" Camera App (commit 7cbfbd38d982 Tainted)

Camera Name imx219
Width 3280, Height 2464, filename test.jpg
Using camera 0, sensor mode 0

GPS output Disabled

Quality 85, Raw no
Thumbnail enabled Yes, width 64, height 48, quality 35
Time delay 1000, Timelapse 0
Link to latest frame enabled  no
Full resolution preview No
Capture method : Single capture

Preview Yes, Full screen Yes
Preview window 0,0,1024,768
Opacity 255
Sharpness 0, Contrast 0, Brightness 50
Saturation 0, ISO 0, Video Stabilisation No, Exposure compensation 0
Exposure Mode 'auto', AWB Mode 'auto', Image Effect 'none'
Flicker Avoid Mode 'off'
Metering Mode 'average', Colour Effect Enabled No with U = 128, V = 128
Rotation 0, hflip No, vflip No
ROI x 0.000000, y 0.000000, w 1.000000 h 1.000000
Camera component done
Encoder component done
Starting component connection stage
Connecting camera preview port to video render.
Connecting camera stills port to encoder input port
Opening output file test.jpg
Enabling encoder output port
Starting capture -1
Finished capture -1
Closing down
Close down completed, all components disconnected, disabled and destroyed
```

Here's the drivers I have:
```
Module                  Size  Used by
wireguard             147456  0
ip6_udp_tunnel         16384  1 wireguard
udp_tunnel             16384  1 wireguard
fuse                  110592  3
ctr                    16384  4
aes_arm_bs             20480  0
crypto_simd            16384  1 aes_arm_bs
cryptd                 24576  1 crypto_simd
ccm                    20480  6
arc4                   16384  2
rt2800usb              28672  0
rt2800lib             102400  1 rt2800usb
rt2x00usb              24576  1 rt2800usb
spidev                 16384  2
rt2x00lib              57344  3 rt2800lib,rt2800usb,rt2x00usb
mac80211              659456  3 rt2800lib,rt2x00lib,rt2x00usb
cfg80211              573440  2 rt2x00lib,mac80211
cp210x                 24576  0
rfkill                 28672  3 cfg80211
crc_ccitt              16384  1 rt2800lib
usbserial              40960  1 cp210x
vc4                   155648  4
drm_kms_helper        180224  2 vc4
drm                   385024  4 vc4,drm_kms_helper
snd_soc_core          188416  1 vc4
snd_compress           20480  1 snd_soc_core
snd_pcm_dmaengine      16384  1 snd_soc_core
snd_bcm2835            32768  1
syscopyarea            16384  1 drm_kms_helper
sysfillrect            16384  1 drm_kms_helper
sysimgblt              16384  1 drm_kms_helper
fb_sys_fops            16384  1 drm_kms_helper
snd_pcm                98304  4 vc4,snd_pcm_dmaengine,snd_bcm2835,snd_soc_core
snd_timer              32768  1 snd_pcm
snd                    69632  7 snd_compress,snd_timer,snd_bcm2835,snd_soc_core,snd_pcm
spi_bcm2835            16384  0
i2c_bcm2835            16384  2
w1_gpio                16384  0
wire                   40960  1 w1_gpio
cn                     16384  1 wire
hwmon                  16384  1 wire
uio_pdrv_genirq        16384  0
fixed                  16384  0
uio                    20480  1 uio_pdrv_genirq
bcm2835_v4l2           53248  0
v4l2_common            16384  1 bcm2835_v4l2
videobuf2_vmalloc      16384  1 bcm2835_v4l2
videobuf2_memops       16384  1 videobuf2_vmalloc
videobuf2_v4l2         24576  1 bcm2835_v4l2
videobuf2_core         45056  2 bcm2835_v4l2,videobuf2_v4l2
videodev              184320  4 v4l2_common,videobuf2_core,bcm2835_v4l2,videobuf2_v4l2
media                  32768  1 videodev
i2c_dev                16384  4
ip_tables              24576  0
x_tables               32768  1 ip_tables
ipv6                  425984  39 wireguard
```

I'm doing this on a CM3 with the following kernel and os:
```
Linux testpi 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux
```

A CTRL+C doesn't seem to kill the cmd as a `ps aux | grep mmal` shows me:
```
root       113  0.0  0.0      0     0 ?        I<   08:19   0:00 [mmal-vchiq]
root       163  0.0  0.0      0     0 ?        I<   08:19   0:00 [mmal-vchiq]
```

Furthermore, a reboot does not seem to fix the issue, I have to power cycle to restore it to a working state.

rolandjitsu, 2019-07-24:
Ok, so after trying a few other things, I've discovered that I had a service running which is using I2C (wiringPiI2C) on address `0x51` and `0x50` to get some info from some peripheral and after stopping that, the raspistill cmd seems to work just fine (at least for the amount of time I tried it).

6by9, 2019-07-24:
You're using I2C on which bus? i2c0 or i2c1?
i2c0 is the same one that the camera (and DSI display) use, and you can not use it safely from Linux if you are using either of those two functions. There is no inter-processor arbitration for the peripheral.
Use of i2c1 should be perfectly safe.

Please add ```start_debug=1``` to config.txt, and then run ```sudo vcdbg log assert``` and ```sudo vcdbg log msg``` after you have had the lockup.

Is this on a custom CM baseboard, or the CMIO? How long are the camera cables? If you get corruption or other issues on the I2C bus to the camera then I can see the GPU waiting for a long time before failing the capture request.

6by9, 2019-07-24:
> Furthermore, a reboot does not seem to fix the issue, I have to power cycle to restore it to a working state.

That would imply a hardware issue. No software state is retained over a reboot, therefore have you got something funny going on in the harware with the camera shutdown line or similar?

rolandjitsu, 2019-07-24:
> You're using I2C on which bus? i2c0 or i2c1?
We're using both. 0 for camera and 1 for another peripheral.

> i2c0 is the same one that the camera (and DSI display) use, and you can not use it safely from Linux if you are using either of those two functions. There is no inter-processor arbitration for the peripheral.
I2C0 is only used for camera.

> Use of i2c1 should be perfectly safe.
Ok.

> Please add `start_debug=1` to config.txt, and then run `sudo vcdbg log assert` and `sudo vcdbg log msg` after you have had the lockup.
I will try that.

> Is this on a custom CM baseboard, or the CMIO? How long are the camera cables? If you get corruption or other issues on the I2C bus to the camera then I can see the GPU waiting for a long time before failing the capture request.
Yes, it's a custom board. Standard length (~15cm).

Actually, if we collect data over I2C1 from the other peripheral at the same time we also run the camera, then the raspistill cmd gets stuck. If I reboot and run only the camera, it works just fine.

rolandjitsu, 2019-07-24:
> > Furthermore, a reboot does not seem to fix the issue, I have to power cycle to restore it to a working state.
> 
> That would imply a hardware issue. No software state is retained over a reboot, therefore have you got something funny going on in the hardware with the camera shutdown line or similar?

Well, it only happens when we cannot power off/reboot properly after the raspistill cmd gets stuck. 

Here's our gpio setup:
```
BANK0 (GPIO 0 to 27):
GPIO 0: level=1 fsel=4 alt=0 func=SDA0
GPIO 1: level=1 fsel=4 alt=0 func=SCL0
GPIO 2: level=0 fsel=1 func=OUTPUT
GPIO 3: level=0 fsel=1 func=OUTPUT
GPIO 4: level=0 fsel=0 func=INPUT
GPIO 5: level=1 fsel=0 func=INPUT
GPIO 6: level=1 fsel=0 func=INPUT
GPIO 7: level=1 fsel=1 func=OUTPUT
GPIO 8: level=1 fsel=1 func=OUTPUT
GPIO 9: level=0 fsel=4 alt=0 func=SPI0_MISO
GPIO 10: level=0 fsel=4 alt=0 func=SPI0_MOSI
GPIO 11: level=0 fsel=4 alt=0 func=SPI0_SCLK
GPIO 12: level=1 fsel=0 func=INPUT
GPIO 13: level=0 fsel=0 func=INPUT
GPIO 14: level=1 fsel=4 alt=0 func=TXD0
GPIO 15: level=1 fsel=4 alt=0 func=RXD0
GPIO 16: level=1 fsel=0 func=INPUT
GPIO 17: level=1 fsel=1 func=OUTPUT
GPIO 18: level=0 fsel=0 func=INPUT
GPIO 19: level=0 fsel=0 func=INPUT
GPIO 20: level=0 fsel=0 func=INPUT
GPIO 21: level=0 fsel=0 func=INPUT
GPIO 22: level=1 fsel=0 func=INPUT
GPIO 23: level=0 fsel=0 func=INPUT
GPIO 24: level=0 fsel=0 func=INPUT
GPIO 25: level=0 fsel=0 func=INPUT
GPIO 26: level=0 fsel=0 func=INPUT
GPIO 27: level=0 fsel=1 func=OUTPUT
BANK1 (GPIO 28 to 45):
GPIO 28: level=0 fsel=0 func=INPUT
GPIO 29: level=0 fsel=0 func=INPUT
GPIO 30: level=0 fsel=0 func=INPUT
GPIO 31: level=0 fsel=0 func=INPUT
GPIO 32: level=0 fsel=0 func=INPUT
GPIO 33: level=0 fsel=0 func=INPUT
GPIO 34: level=1 fsel=0 func=INPUT
GPIO 35: level=1 fsel=0 func=INPUT
GPIO 36: level=1 fsel=0 func=INPUT
GPIO 37: level=0 fsel=0 func=INPUT
GPIO 38: level=0 fsel=0 func=INPUT
GPIO 39: level=0 fsel=0 func=INPUT
GPIO 40: level=0 fsel=0 func=INPUT
GPIO 41: level=0 fsel=0 func=INPUT
GPIO 42: level=0 fsel=0 func=INPUT
GPIO 43: level=0 fsel=0 func=INPUT
GPIO 44: level=1 fsel=6 alt=2 func=SDA1
GPIO 45: level=1 fsel=6 alt=2 func=SCL1
BANK2 (GPIO 46 to 53):
GPIO 46: level=1 fsel=0 func=INPUT
GPIO 47: level=1 fsel=1 func=OUTPUT
GPIO 48: level=1 fsel=4 alt=0 func=SD0_CLK
GPIO 49: level=1 fsel=4 alt=0 func=SD0_CMD
GPIO 50: level=0 fsel=4 alt=0 func=SD0_DAT0
GPIO 51: level=0 fsel=4 alt=0 func=SD0_DAT1
GPIO 52: level=0 fsel=4 alt=0 func=SD0_DAT2
GPIO 53: level=0 fsel=4 alt=0 func=SD0_DAT3
```

forcemagic, 2019-10-12:
Same issue here! Didn't see this, so I posted on the Raspberry SO first. https://raspberrypi.stackexchange.com/questions/104545/raspberry-camera-v1-2-results-in-kernel-panic

rolandjitsu, 2020-12-03:
> Please add `start_debug=1` to config.txt, and then run `sudo vcdbg log assert` and `sudo vcdbg log msg` after you have had the lockup.

@6by9 getting the following:
```
010144.610: assert( edid_state[hdmi_core].i2c_driver ) failed; ../../../../../middleware/hdmi/edid.c::hdmi_edid_shutdown line 736 rev 0e6daa5
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
025986.762: assert( camera_subsystem_state.power_enable_ref_count[port] > 0 ) failed; ../../../../../vcfw/drivers/device/camera_subsystem/camera_subsystem.c::camera_disable_ctrl line 1141 rev 0e6daa5
vcdbg_ctx_get_dump_stack: dump_stack failed
```

```
002899.259: brfs: File read: /mfs/sd/config.txt
002900.583: brfs: File read: 2019 bytes
002901.613: brfs: File read: /mfs/sd/dt-blob.bin
003265.009: HDMI:EDID error reading EDID block 0 attempt 0
003627.028: HDMI:EDID error reading EDID block 0 attempt 1
003989.042: HDMI:EDID error reading EDID block 0 attempt 2
004351.060: HDMI:EDID error reading EDID block 0 attempt 3
004713.074: HDMI:EDID error reading EDID block 0 attempt 4
005075.091: HDMI:EDID error reading EDID block 0 attempt 5
005437.106: HDMI:EDID error reading EDID block 0 attempt 6
005799.123: HDMI:EDID error reading EDID block 0 attempt 7
006161.138: HDMI:EDID error reading EDID block 0 attempt 8
006523.154: HDMI:EDID error reading EDID block 0 attempt 9
006524.211: HDMI:EDID giving up on reading EDID block 0
006885.296: HDMI:EDID error reading EDID block 0 attempt 0
007247.317: HDMI:EDID error reading EDID block 0 attempt 1
007609.337: HDMI:EDID error reading EDID block 0 attempt 2
007971.357: HDMI:EDID error reading EDID block 0 attempt 3
008333.377: HDMI:EDID error reading EDID block 0 attempt 4
008695.397: HDMI:EDID error reading EDID block 0 attempt 5
009057.417: HDMI:EDID error reading EDID block 0 attempt 6
009419.437: HDMI:EDID error reading EDID block 0 attempt 7
009781.457: HDMI:EDID error reading EDID block 0 attempt 8
010143.477: HDMI:EDID error reading EDID block 0 attempt 9
010144.539: HDMI:EDID giving up on reading EDID block 0
010148.358: brfs: File read: 7150 bytes
010150.565: brfs: File read: /mfs/sd/config.txt
010151.873: gpioman: gpioman_get_pin_num: pin DISPLAY_SDA not defined
010151.943: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
010153.528: gpioman: gpioman_get_pin_num: pin BT_ON not defined
010153.587: gpioman: gpioman_get_pin_num: pin WL_ON not defined
010188.622: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
010188.718: *** Restart logging
010188.782: brfs: File read: 2019 bytes
010550.413: hdmi: HDMI:EDID error reading EDID block 0 attempt 0
010912.439: hdmi: HDMI:EDID error reading EDID block 0 attempt 1
011274.464: hdmi: HDMI:EDID error reading EDID block 0 attempt 2
011636.489: hdmi: HDMI:EDID error reading EDID block 0 attempt 3
011998.514: hdmi: HDMI:EDID error reading EDID block 0 attempt 4
012360.540: hdmi: HDMI:EDID error reading EDID block 0 attempt 5
012722.564: hdmi: HDMI:EDID error reading EDID block 0 attempt 6
013084.589: hdmi: HDMI:EDID error reading EDID block 0 attempt 7
013446.615: hdmi: HDMI:EDID error reading EDID block 0 attempt 8
013808.640: hdmi: HDMI:EDID error reading EDID block 0 attempt 9
013809.708: hdmi: HDMI:EDID giving up on reading EDID block 0
014170.811: hdmi: HDMI:EDID error reading EDID block 0 attempt 0
014532.834: hdmi: HDMI:EDID error reading EDID block 0 attempt 1
014894.855: hdmi: HDMI:EDID error reading EDID block 0 attempt 2
015256.877: hdmi: HDMI:EDID error reading EDID block 0 attempt 3
015618.899: hdmi: HDMI:EDID error reading EDID block 0 attempt 4
015980.920: hdmi: HDMI:EDID error reading EDID block 0 attempt 5
016342.943: hdmi: HDMI:EDID error reading EDID block 0 attempt 6
016704.965: hdmi: HDMI:EDID error reading EDID block 0 attempt 7
017066.987: hdmi: HDMI:EDID error reading EDID block 0 attempt 8
017429.008: hdmi: HDMI:EDID error reading EDID block 0 attempt 9
017430.071: hdmi: HDMI:EDID giving up on reading EDID block 0
017433.734: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
017434.962: brfs: File read: /mfs/sd/cmdline.txt
[... 20 lines cut ...]
018162.271: brfs: File read: 1283 bytes
018166.896: brfs: File read: /mfs/sd/overlays/pi3-disable-bt.dtbo
018181.782: dterror: can't find symbol 'uart0_pins'
018181.797: Failed to resolve overlay 'pi3-disable-bt'
018181.941: dtparam: watchdog=on
018191.900: brfs: File read: 954 bytes
018195.470: brfs: File read: /mfs/sd/overlays/i2c-rtc.dtbo
018218.098: Loaded overlay 'i2c-rtc'
018218.152: dtparam: pcf8523=true
020401.316: Device tree loaded to 0x2eff9800 (size 0x6742)
020403.301: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
021663.563: vchiq_core: vchiq_init_state: slot_zero = 0xef980000, is_master = 1
021667.946: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
021673.414: TV service:host side not connected, dropping notification 0x00000002, 0x00000002, 0x00000055
025380.883: camsubs: Looking for camera 0: i2c_port = 0, led gpio = 2, power enable gpio = 3
025681.667: camsubs: Camera not found
025681.784: camsubs: Looking for camera 0: i2c_port = 0, led gpio = 2, power enable gpio = 3
025982.551: camsubs: Camera not found
025982.658: camsubs: Looking for camera 0: i2c_port = 0, led gpio = 2, power enable gpio = 3
025984.340: camsubs: Camera found OK
025986.448: gpioman: gpioman_get_pin_num: pin FLASH_0_ENABLE not defined
025986.522: gpioman: gpioman_get_pin_num: pin FLASH_0_INDICATOR not defined
025986.619: gpioman: gpioman_get_pin_num: pin FLASH_0_ENABLE not defined
025986.688: gpioman: gpioman_get_pin_num: pin FLASH_0_INDICATOR not defined
```