raspberrypi/userland #565: Outdated headers

Tombana, 2019-07-25:
This might be related to issue #469 .

In [this commit](https://github.com/raspberrypi/firmware/commit/64b5649a41b69d09bfe0ed05448d28a66be3edfd) to the [firmware repository](https://github.com/raspberrypi/firmware) many new header files in `/opt/vc/include` were added, and they are newer than the ones in this repository (userland). An example is `opt/vc/include/interface/vmcs_host/vc_tvservice.h`

I'm trying to compile a project based on the source of raspicam (RaspiVid.c etc). It compiles fine when using the include files from this repo (userland), but it gives compilation errors when using the new include files from `/opt/vc/include` that came from my linux package manager (for example: `unknown typename TV_ATTACHED_DEVICES_T in file opt/vc/include/interface/vmcs_host/vc_tvservice.h` ).
This suggests there exists a newer version of `RaspiVid.c` and other files (that allows compiling with the newer headers) that has not been pushed to the userland repo yet.

6by9, 2019-07-25:
#469 is that the EGL headers have been updated by Khronos, but as we don't support any of the new features we haven't updated our copies.
If an application supports the new stuff  should it be available and tries to use the new defines/functions/typedefs, then it won't compile against our headers.

You are right that there are a number of changes that have been made to the userland code for Pi4 support that haven't been pushed to the public tree.
@popcornmix Did you have a magic script for taking stuff from firmware to userland, or is it manual still? There's at least tvservice and vcsm changes that need pulling in from the firmware tree, although I suspect the changes may not have been brought through from 2711 to master.

popcornmix, 2019-07-25:
I updated userland with about 40 commits yesterday. It is still a manual (and painful) process.
I believe userland and pi4 vc4 tree are very close.

Can you point out a file that is not up to date? 

vc_tvservice.h seems to match in userland, firmware and vc4 trees on a quick check.

Tombana, 2019-07-25:
My apologies, I actually didn't see the commit from yesterday yet. Indeed `vc_tvservice.h` now matches the one in firmware. Possibly all files are up to date now; if I find something I'll comment here but for now I guess the issue can be closed.

6by9, 2019-07-25:
I thought I'd checked too - sorry popcornmix.

Tombana, 2019-07-28:
I found another difference, but this time I think the version in the firmware repo is the older one.
`/opt/vc/include/interface/mmal/mmal_parameters_camera.h` does not have the line `MMAL_PARAM_AWBMODE_GREYWORLD` whereas  `userland/interface/mmal/mmal_parameters_camera.h` does, on line 231.
(This gives a compile error when trying to build `raspivid` with includes from `/opt/vc/include` instead of the userland repo.)