raspberrypi/userland #566: Tvservice does not detect modes via VGA666 on RPi4

alphanu1, 2019-07-25:
Tvservice shows 0 modes for DMT and CEA on RPi4. This also means it no longer has the ability to change to these modes.

6by9, 2019-07-25:
What commands specifically are you using via tvservice to change the DPI timings? I'm not aware of any way on previous Pis to do that.

VGA666 is seen as a DPI device, and currently we're not looking at dynamically changing resolutions on DPI displays.
WIth DRM/KMS you'd need to provide the list of modes supported by the display in order to change them. Changing mode via tvservice under the feet of DRM/KMS is likely to cause issues.

6by9, 2019-07-25:
https://github.com/raspberrypi/firmware/issues/1206 duplicated this.

alphanu1, 2019-07-26:
With other models of RPi the following options in the config.txt would add DMT Mode 87 and we would be able to switch to this using " tvservice -e "DMT 87" ". however this no longer works and tvservice -m DMT or tvservice -m CEA show 0 modes.

dtoverlay=vga666
enable_dpi_lcd=1
display_default_lcd=1
dpi_group=2
dpi_mode=87
hdmi_timings=320 1 16 30 34 240 1 2 3 22 0 0 0 60 0 6400000 1

JamesH65, 2019-07-26:
So, do you have two display devices attached? What does 'tvservice -l' report?

6by9, 2019-07-26:
Pi4 by default is using DRM/KMS for rendering and mode switching. Changing mode with tvservice is going to cause grief with that.

Setting ```dpi_timings``` was added to config.txt a while ago to as we now support DPI at the same time as HDMI (on all Pi variants, although only HDMI0+DPI on Pi4). That removes the ambiguity between whether hdmi_timings is for HDMI or DPI. ```hdmi_timings``` should still work as a fallback.

dpi_timings / hdmi_timings is passed up from the firmware to FKMS as one fixed mode. Dynamic resolution changing is not currently supported on DPI, nor high up on the priority list to add.

If you've added those lines to config.txt then the Pi should be booting with them and retaining them.
For me with config.txt containing
```
enable_dpi_lcd=1
dpi_timings=320 1 16 30 34 240 1 2 3 22 0 0 0 60 0 6400000 1
dpi_output_format=454661
dpi_group=2
dpi_mode=87
dtoverlay=vga666
```
I get ```xrandr --verbose``` reporting
```
pi@raspberrypi:~ $ DISPLAY=:0 xrandr --verbose
Screen 0: minimum 320 x 200, current 1920 x 1080, maximum 7680 x 7680
DSI-1 connected 320x240+0+0 (0x44) normal (normal left inverted right x axis y axis) 0mm x 0mm
	Identifier: 0x41
	Timestamp:  28681
	Subpixel:   unknown
	Gamma:      1.0:1.0:1.0
	Brightness: 1.0
	Clones:    
	CRTC:       1
	CRTCs:      1
	Transform:  1.000000 0.000000 0.000000
	            0.000000 1.000000 0.000000
	            0.000000 0.000000 1.000000
	           filter: 
	Broadcast RGB: Automatic 
		supported: Automatic, Full, Limited 16:235
	bottom margin: 0 
		range: (0, 100)
	top margin: 0 
		range: (0, 100)
	right margin: 0 
		range: (0, 100)
	left margin: 0 
		range: (0, 100)
	link-status: Good 
		supported: Good, Bad
	CONNECTOR_ID: 51 
		supported: 51
	non-desktop: 0 
		range: (0, 1)
  FIXED_MODE (0x44)  6.400MHz +HSync +VSync *current +preferred
        h: width   320 start  336 end  366 total  400 skew    0 clock  16.00KHz
        v: height  240 start  242 end  245 total  267           clock  59.93Hz
```
which looks correct to me (I'm a step or two ahead on the kernel and therefore you may not have the margins or RGB Broadcast parameters).

What exactly is your use case? Full details please.