raspberrypi/userland #568: apticron.conf missing

Tuffiglig, 2019-08-06:
Hey. I hope I'm on the right place.
If not please tell me, and i will close this and open it in the hopefully correct place.

I was trying to configurate 'apticron' on a freshly installes Rasperrby Pi 4 with the image:
2019-07-10-raspbian-buster-lite.img
Normaly (on stecht) after installing apticron and this command:
$ sudo dpkg-reconfigure apticron
There apperas a screen to type in the email address.
Under buster nothing happens.
Then i figuerd out, that '/etc/apticron/apticron.conf' was missing.
After copying it from '/usr/lib/apticron/apticron.conf' and adjusting some parameters, apticron seems to work. Didn't get a new updated package yet, but setting NOTIFY_NO_UPDATES="0" and a 'sudo apticron' gets me an email.

**Expected behaviour**
I woul expect the the file apticron.conf at in folder /etc/apticron/
and
'sudo dpkg-reconfigure apticron' shows something.

**Actual behaviour**
Se above.

**System**
- Raspberry Pi 4 Model B Rev 1.1
- Mem total: 1989 mebibytes
- Raspbian GNU/Linux 10 (buster)
- Linux version 4.19.58-v7l+
- Raspberry Pi reference 2019-07-10
- Jul  9 2019 14:38:27
   Copyright (c) 2012 Broadcom
   version d2b1b7fb01475cb3914b2086299e32d724e832f1 (clean) (release) (start_cd)

**Logs**
Nothing found.

**Additional context**
-

6by9, 2019-08-06:
apticron is a Debian package, not anything from the Raspberry Pi userland repo. https://github.com/RPi-Distro/repo/issues is probably more appropriate.
@XECDesign can you transfer the issue, or do you want it somewhere else again?

Tuffiglig, 2019-08-06:
Thanks. I will delet it here and open a new one in /repo/issues/

closed 2019-08-06 by Tuffiglig

XECDesign, 2019-08-06:
I don't think that's the right place either, since it's not an archive.raspberrypi.org package.

If it appear in Raspbian, but not in Debian Buster, you can report it here https://bugs.launchpad.net/raspbian/

If it appears in Debian, it should be reported here https://www.debian.org/Bugs/

However, it's not clear that there is a problem here at all. The change in behaviour may have been intentional by the maintainers.

XECDesign, 2019-08-06:
Yup, looks like it was intentional - https://salsa.debian.org/debian/apticron/commit/495e5b078021b6878e3235192aacfcce60d447a7

Tuffiglig, 2019-08-06:
Thanks. What is the best way to find those things? I googled a little bit around, but didn't find something helpfull.

XECDesign, 2019-08-06:
changelogs in /usr/share/doc. I saw it mention stateless configuration and went to check what the actual corresponding change was. salsa.debian.org is where most of the git repos for debian packages can be found to get the full history of changes.

Tuffiglig, 2019-08-06:
Thank you. I hope it will help me the next time and i don't need to bother you guys.