raspberrypi/userland #575: Raspivid could use a FPS value smaller as 1 (improvement request)

zanco, 2019-09-14:
Hi,
To record to a timelapse movie it would be nice to have Raspivid to handle -fps values smaller as 1.

.1 gives an error and 0.1 appears to be taken as zero.

If 0.1 was taken as one tenth and one image was taken every 10 seconds you could make a timelapse movie without the need of other software.

I am not shure if this is the right place for such a request however.

Thanks,
Ben

JamesH65, 2019-09-14:
THis is not currently possible, otherwise we would have already done it.

6by9, 2019-09-16:
With imx219 (v2.1 camera) that may just fall out on the camera side if you amend the sscanf line for ```CommandFramerate``` to read the fraction. It can run down to 0.1fps.
```state->framerate``` is currently stored as the integer numerator for the framerate fraction, so change the denominator to 1000 (instead of 1) and multiply the supplied value by 1000.
(Having to use floats is annoying though).

ov5647 (v1 sensor) can only go to 6 seconds per frame.

Whether the rate control algorithm in the codec does something sensible is a totally different question.

pelwell, 2019-09-16:
Whilst not quite the same, how hard would it be to drop frames before the encoder to reduce the frame rate?

JamesH65, 2019-09-16:
I thought it was the encoder framerate stuff that went wrong, not at the camera end? It's minimum value was 1 IIRC, but I may not be RC!

6by9, 2019-09-16:
Raspivid uses MMAL_CONNECTION_T between the camera and encoder, and that has no mechanism to drop frames. Pulling the frames back to the ARM and only passing one in N isn't that complex, but not something that really wants to get added to raspivid as it's already too complex.

If the encoder goes wrong then that's a different thing. I do recall that an integer framerate is passed in to the codec API for rate control in the absence of timestamps, but if that is set to 0 then it should just rely on timestamps. 

One to possibly play with should someone find some spare time, but it's not a high priority.

pelwell, 2019-09-16:
> it's not a high priority.

Yes, absolutely.

6by9, 2019-09-16:
The other thing to say is that it does not require any input from Raspberry Pi to do at least the initial investigations.

Requesting 0.1fps from the camera is relatively trivial (see my comments above on what is required). If that just works then create a Pull Request (PR).
If the codec complains or does weird things, then report that back as part of the PR and it can be investigated without us having to make the application changes first.

ccdanieldb, 2020-08-04:
Any progress on this?

JamesH65, 2020-08-04:
No, not something we at RPT are working on.