raspberrypi/userland #584: MMAL Stats is wrong after some hours work

99600036, 2019-10-14:
I create own application which use MMAL.  When I start my app I can get stats by /opt/vc/bin/mmal_vc_diag mmal-stats. After one-two hours continuous work I got next report: 
pi@raspberrypi:~ $ /opt/vc/bin/mmal_vc_diag mmal-stats

component               port            buffers         fps     delay
ril.image_encode        0 [in ]rx       0                0.0    0
ril.image_encode        0 [in ]tx       0                0.0    0
ril.image_encode        0 [out]rx       1                0.0    0
ril.image_encode        0 [out]tx       0                0.0    0
ril.image_encode        0 [in ]rx       81699           50.5    36658437
ril.image_encode        0 [in ]tx       81696           50.5    36873027
ril.image_encode        0 [out]rx       81697           50.5    36874843
ril.image_encode        0 [out]tx       81696           50.5    36873965
ril.isp                 0 [in ]rx       81702           50.5    36549306
ril.isp                 0 [in ]tx       81699           50.5    36595790
ril.isp                 0 [out]rx       81699           50.5    36873964
ril.isp                 0 [out]tx       81699           50.5    36608016
ril.isp                 1 [out]rx       0                0.0    0
ril.isp                 1 [out]tx       0                0.0    0
ril.video_encode        0 [in ]rx       157838          97.5    311954
ril.video_encode        0 [in ]tx       157836          97.5    314818
ril.video_encode        0 [out]rx       157840          97.5    910662
ril.video_encode        0 [out]tx       157837          97.5    684648
ril.isp                 0 [in ]rx       157841          97.5    210040
ril.isp                 0 [in ]tx       157839          97.5    294292
ril.isp                 0 [out]rx       157842          97.5    381098
ril.isp                 0 [out]tx       157840          97.5    311989
ril.isp                 1 [out]rx       0                0.0    0
ril.isp                 1 [out]tx       0                0.0    0
ril.video_encode        0 [in ]rx       187039          115.5   170192
ril.video_encode        0 [in ]tx       187039          115.5   318590
ril.video_encode        0 [out]rx       187042          115.5   894893
ril.video_encode        0 [out]tx       187039          115.5   745081
ril.video_render        0 [in ]rx       187071          115.5   173415
ril.video_render        0 [in ]tx       187070          115.5   183342
ril.video_splitter      0 [in ]rx       187071          115.5   221030
ril.video_splitter      0 [in ]tx       187071          115.5   221022
ril.video_splitter      0 [out]rx       187073          115.5   201926
ril.video_splitter      0 [out]tx       187071          115.5   173417
ril.video_splitter      1 [out]rx       187042          115.5   315896
ril.video_splitter      1 [out]tx       187039          115.5   170185
ril.video_splitter      2 [out]rx       157844          97.5    311986
ril.video_splitter      2 [out]tx       157843          97.5    210041
ril.video_splitter      3 [out]rx       81703           50.5    36608011
ril.video_splitter      3 [out]tx       81703           50.5    36549305
ril.camera              0 [out]rx       0                0.0    0
ril.camera              0 [out]tx       0                0.0    0
ril.camera              1 [out]rx       187081          115.5   249221
ril.camera              1 [out]tx       187071          115.5   164350
ril.camera              2 [out]rx       10              114583.3        11
ril.camera              2 [out]tx       0                0.0    0

* Which model of Raspberry Pi? 
 Pi3B+;
* Which OS and version (`cat /etc/rpi-issue`)?
Raspberry Pi reference 2019-04-08
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, e910bf55ac92e27c6e77043e8995bcb5e0a2d742, stage2

* Which firmware version (`vcgencmd version`)?
Sep 24 2019 17:38:33
Copyright (c) 2012 Broadcom
version 6820edeee4ef3891b95fc01cf02a7abd7ca52f17 (clean) (release) (start_db)
* Which kernel version (`uname -a`)?
Linux raspberrypi 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l GNU/Linux

pelwell, 2019-10-14:
It really helps if you can say not only the behaviour you see but also what you think it ought to be, i.e. make it clear what is wrong.

99600036, 2019-10-14:
Normaly mmal_vc_diag mmal-stats have to return real fps (in my case it was 17-19). When I start my app - all work correctly (please see log below). After on 1-2 hours  I'v got log form start message - with wrong information about fps ( I check fps in my app - application continues get 17-19FPS, in time when in log 70 -300 FPS)

pi@raspberrypi:~ $ /opt/vc/bin/mmal_vc_diag mmal-stats
component               port            buffers         fps     delay
ril.image_encode        0 [in ]rx       0                0.0    0
ril.image_encode        0 [in ]tx       0                0.0    0
ril.image_encode        0 [out]rx       1                0.0    0
ril.image_encode        0 [out]tx       0                0.0    0
ril.image_encode        0 [in ]rx       26               8.5    721222
ril.image_encode        0 [in ]tx       23              10.0    698395
ril.image_encode        0 [out]rx       24              10.4    691316
ril.image_encode        0 [out]tx       23              10.0    698304
ril.isp                 0 [in ]rx       29               8.0    721219
ril.isp                 0 [in ]tx       26               8.5    716986
ril.isp                 0 [out]rx       26               9.3    698302
ril.isp                 0 [out]tx       26               8.5    721219
ril.isp                 1 [out]rx       0                0.0    0
ril.isp                 1 [out]tx       0                0.0    0
ril.video_encode        0 [in ]rx       71              16.2    249002
ril.video_encode        0 [in ]tx       71              16.3    236543
ril.video_encode        0 [out]rx       74              17.0    245372
ril.video_encode        0 [out]tx       71              16.4    229074
ril.isp                 0 [in ]rx       74              16.2    127341
ril.isp                 0 [in ]tx       72              16.4    232667
ril.isp                 0 [out]rx       74              16.7    236528
ril.isp                 0 [out]tx       72              16.4    249065
ril.isp                 1 [out]rx       0                0.0    0
ril.isp                 1 [out]tx       0                0.0    0
ril.video_encode        0 [in ]rx       89              19.3    127225
ril.video_encode        0 [in ]tx       89              19.3    122265
ril.video_encode        0 [out]rx       92              19.6    223860
ril.video_encode        0 [out]tx       89              19.0    165766
ril.video_render        0 [in ]rx       89              19.3    125718
ril.video_render        0 [in ]tx       88              19.2    133343
ril.video_splitter      0 [in ]rx       89              19.3    154439
ril.video_splitter      0 [in ]tx       89              19.3    154443
ril.video_splitter      0 [out]rx       92              18.8    241583
ril.video_splitter      0 [out]tx       91              19.2    125769
ril.video_splitter      1 [out]rx       94              19.3    141470
ril.video_splitter      1 [out]tx       93              19.3    127191
ril.video_splitter      2 [out]rx       77              16.0    249064
ril.video_splitter      2 [out]tx       77              16.1    127335
ril.video_splitter      3 [out]rx       30               7.5    721220
ril.video_splitter      3 [out]tx       30               7.6    721221
ril.camera              0 [out]rx       0                0.0    0
ril.camera              0 [out]tx       0                0.0    0
ril.camera              1 [out]rx       103             20.3    249719
ril.camera              1 [out]tx       93              19.3    120665
ril.camera              2 [out]rx       10              112244.9        12
ril.camera              2 [out]tx       0                0.0    0

6by9, 2019-10-14:
```mmal_vc_diag mmal-stats``` was used for very basic level automated performance profiling.

The timestamps are stored in uint32_t in usecs - https://github.com/raspberrypi/userland/blob/master/interface/mmal/mmal_common.h#L64
2^32 usecs is 71 minutes. Any use for more than that time period will be erroneous.

As it was only rough debug information it's not something that is a high enough priority to even consider fixing.

99600036, 2019-10-14:
Thanks
If it not system error - Ok. I worried that it was more serious trouble.

closed 2019-10-14 by 99600036