raspberrypi/userland #587: Additional initializations for raspicamcontrol_set_defaults

Mr-DaveDev, 2019-10-29:
**Is this the right place for my bug report?**
This repository contains the Raspberry Pi userland code. If you believe that the issue you are seeing is in the userland libraries, this is the right place. If not, we have other repositories for the GPU firmware at [github.com/raspberrypi/firmware](https://github.com/raspberrypi/firmware) and linux kernel at [github.com/raspberrypi/linux](https://github.com/raspberrypi/linux). If you have problems with the Raspbian distribution packages, report them in the [github.com/RPi-Distro/repo](https://github.com/RPi-Distro/repo). If you simply have a question, then [the Raspberry Pi forums](https://www.raspberrypi.org/forums) are the best place to ask it.

**Describe the bug**

> In the function `raspicamcontrol_set_defaults` the newly added items for `analog_gain`, `digital_gain` et al need to be initialized.

**To reproduce**

> Randomly occurs on pi zero.

**Expected behaviour**

> Normal Image

**Actual behaviour**

> Completely white image due to gain being a random value.

**System**

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW: 

> PiZeroW

* Which OS and version (`cat /etc/rpi-issue`)? 

> Raspberry Pi reference 2019-07-10
> Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 175dfb027ffabd4b8d5080097af0e51ed9a4a56c, stage2
> 

* Which firmware version (`vcgencmd version`)?

> Sep 24 2019 17:39:25 
> Copyright (c) 2012 Broadcom
> version 6820edeee4ef3891b95fc01cf02a7abd7ca52f17 (clean) (release) (start_x)
> 

* Which kernel version (`uname -a`)?

> Linux raspberrypi 4.19.75+ #1270 Tue Sep 24 18:38:54 BST 2019 armv6l GNU/Linux

**Logs**

> none

**Additional context**
Additional default values need to be specified in `raspicamcontrol_set_defaults`.  Random memory values are sometimes assigned for `analog_gain` and `digital_gain` resulting in a completely white image.  The following are best guess defaults for the parameters observed to be missing.  These would need to be added after this [line](https://github.com/raspberrypi/userland/blob/5070cb7fc150fc98f1ed64a7739c3356970d9f76/host_applications/linux/apps/raspicam/RaspiCamControl.c#L976) 
```
   params->annotate_justify = 0;
   params->annotate_x = 0;
   params->annotate_y = 0;
   params->analog_gain = 0;
   params->digital_gain = 0;
   params->settings = 0;
```
Note that the parameter `settings` appears to be used as a bool but was assigned a value of 1 in the code so a numeric value was used above instead of `MMAL_FALSE` to be consistent.

6by9, 2019-10-31:
Which raspicam app are you using?
The calling app is normally expected to have memset the entire state before starting, eg https://github.com/raspberrypi/userland/blob/master/host_applications/linux/apps/raspicam/RaspiStill.c#L271 and https://github.com/raspberrypi/userland/blob/master/host_applications/linux/apps/raspicam/RaspiVid.c#L381

TBH It would be more sensible for ```raspicamcontrol_set_defaults``` to memset the entire structure anyway, so I may look at changing that.

Mr-DaveDev, 2019-11-03:
Closing since this is my error for not doing a memset

closed 2019-11-03 by Mr-DaveDev