raspberrypi/userland #588: vcgencmd measure_temp inaccurate under 64 bit?

sbromberger, 2019-10-29:
**Describe the bug**
`vcgencmd` built in 64-bit mode (running on Ubuntu 19.10) is showing temps 20-25C lower than are shown under Raspbian.

**To reproduce**
Install Raspbian. Run `vcgencmd measure_temp` under idle load (on my system, the temp fluctuates between 71C and 75C).

Install Ubuntu 19.10 and compile userland for 64-bit. Run `vcgencmd measure_temp`. On the same system with this new OS, temp is recorded at between 50C and 55C.

**Expected behaviour**
I'd expect similar temperature readings on idle systems.

**Actual behaviour**
Temperature difference is too pronounced to make sense.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? RPi4, 4GB
* Which OS and version (`cat /etc/rpi-issue`)?
Raspian:
```
Raspberry Pi reference 2019-06-20
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 150e25c4f8123a4c9c63e8dca1b4737fa6c1135c, stage2
```

Ubuntu: 19.10

* Which firmware version (`vcgencmd version`)?
```
Aug 15 2019 12:03:51 
Copyright (c) 2012 Broadcom
version 9f8431fb7839c7f00f52b81f5822ddab2b31d0db (clean) (release) (start)
```
(From the Ubuntu system)

* Which kernel version (`uname -a`)?
```
Linux red5.3.0-1008-raspi2 #9+newupdate SMP Mon Oct 21 18:09:52 CST 2019 aarch64 aarch64 aarch64 GNU/Linux
```
**Logs**
n/a

**Additional context**
n/a

JamesH65, 2019-10-29:
Does the same thing happen on the Raspbian 64bit test kernel? Otherwise, this need to be reported to Canonical - Ubuntu is their OS.

sbromberger, 2019-10-29:
I haven't used the raspbian test kernel (didn't know it existed).

It may be premature to call this an OS problem, but if you can help me describe why it might be OS-related as opposed to some issue with the userland code in a 64-bit environment, I'm happy to open up an issue there.

JamesH65, 2019-10-29:
Well, you'd need to show it goes wrong on our stuff - we don't tend to fix bugs in 3rd party distros - they can fix their own issues! Worth noting that on the whole we have not tested userland code in 64 bit  so there is absolutely no guarantee it will work. In fact there are quite big chunks of it that we know will simply not work - one of the reasons we have not yet released a full 64bit distro.

sbromberger, 2019-10-29:
I totally understand the beta-nature of the userland utils under 64-bit, and I'm really not expecting a confirmation and commitment to fix, but it's probably worth the developers keeping in the back of their mind that there's this (possible) issue with the utilities.

I'm installing the 64-bit raspbian test kernel now and will let you know how it goes. Thanks.

popcornmix, 2019-10-29:
My guess is the temperatures reported are accurate, and the difference is one system is more idle than the other.

Make sure you are using the same firmware on both (``vcgencmd version``). I assume the raspbian and ubuntu test were done on the same Pi4?

[bcmstat](https://github.com/MilhouseVH/bcmstat) is a useful tool for looking at cpu usage, temperature and clocks.

sbromberger, 2019-10-29:
OK, under the 64-bit raspbian kernel, I'm not seeing the higher temps either (currently at 62C idle.)

```
(61.0'C/0) seth@red:~$ vcgencmd version
Oct 28 2019 18:51:52 
Copyright (c) 2012 Broadcom
version ffd3d31b2652a3eb6deb221d21ba6a6bcf200411 (tainted) (release) (start_cd)
```

So maybe I need to rebuild the userland under my ubuntu system? I built it yesterday from source (master) so I don't know why the versions would be so different.

Anyway - will close this out and will reopen if something else comes up. Thanks, and sorry for the (now probable) false alarm.

closed 2019-10-29 by sbromberger