raspberrypi/userland #589: hello_pi

ShorTie8, 2019-11-01:
Is it posible to build userland without hello_pi ??
Like a -DHELLO_PI=OFF maybe or something ??

It's like 31.7 of the 34.4 which is built.

6by9, 2019-11-03:
hello_pi isn't built by default, you have to add ALL_APPS for it to build at all.
https://github.com/raspberrypi/userland/blob/master/host_applications/linux/CMakeLists.txt#L21

ShorTie8, 2019-11-05:
Sorry, Don't know what you mean by 'add ALL_APTS ??
grep -r 'ALL_APPS' sources/rpi-userland
sources/rpi-userland/userland-master/host_applications/linux/CMakeLists.txt:if(ALL_APPS)

And if I rem it out, it still gets installed.
if(ALL_APPS)
 add_subdirectory(apps/vcdbg)
 add_subdirectory(libs/elftoolchain)
 rem# add_subdirectory(apps/smct)
 add_subdirectory(apps/edid_parser)
 rem# add_subdirectory(apps/hello_pi)
endif()

It's just a whole bunch of examples that are not needed.
note: rem# is really just #

6by9, 2019-11-05:
Take a clean userland tree, and run ```./buildme```. It does NOT **build** hello_pi if you look within the build subdirectory.

To build hello_pi directly from userland you need to invoke cmake manually via
```
cd build/raspberry/release
cmake -DCMAKE_BUILD_TYPE=release -DARM64=OFF -DALL_APPS=ON ../../..
make
```
which will actually error out as a couple of other libraries and apps are not present in userland (vcdbg, elftoolchain, and edid_parser).
Alternatively you invoke [rebuld.sh](https://github.com/raspberrypi/userland/blob/master/host_applications/linux/apps/hello_pi/rebuild.sh)

It does **copy** the source files to /opt/vc/src, and that includes a 30MB sample H264 file.
If you felt like writing the relevant CMake option to omit that, then PRs are welcome. It's not a high priority to fix.