raspberrypi/userland #593: Using the -gps option with the raspistil command will fail to load libGPS.

matoken, 2019-11-14:
**Describe the bug**
Using the `-gps` option with the raspistil command will fail to load libGPS.

**To reproduce**
```
$ raspistill -gps -o test.jpg
libgps.so.22: cannot open shared object file: No such file or directory
Unable to load the libGPS library
```

**workaround**
```
$ sudo ln -s /usr/lib/arm-linux-gnueabihf/libgps.so.23.0.0 /usr/lib/arm-linux-gnueabihf/libgps.so.22
```

**System**

* raspinfo
https://gist.github.com/d7bf0d1d82b53c38ccd8c966ad25d6d8

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
```
$ cat /proc/device-tree/model
Raspberry Pi Model B Plus Rev 1.2
```
* Which OS and version (`cat /etc/rpi-issue`)?
```
$ cat /etc/rpi-issue
Raspberry Pi reference 2019-09-26
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 80d486687ea77d31fc3fc13cf3a2f8b464e129be, stage2
```

* Which firmware version (`vcgencmd version`)?
```
$ vcgencmd version
Sep 24 2019 17:39:25 
Copyright (c) 2012 Broadcom
version 6820edeee4ef3891b95fc01cf02a7abd7ca52f17 (clean) (release) (start_x)
```

* Which kernel version (`uname -a`)?
```
$ uname -a
Linux raspberrypi 4.19.75+ #1270 Tue Sep 24 18:38:54 BST 2019 armv6l GNU/Linux
```

* package version
```
$ dpkg-query -W gpsd libgps23 libraspberrypi-bin
gpsd 3.17-7
libgps23:armhf 3.17-7
libraspberrypi-bin 1.20190925+1-1
```

JamesH65, 2019-11-14:
PR to fix this is here https://github.com/raspberrypi/userland/pull/594

6by9, 2019-11-21:
PR merged. @matoken  please retest and close if happy.