raspberrypi/userland #595: tvservice -n does not working with regular bash statements

8street, 2019-11-17:
**Describe the bug**
tvservice -n does not working with regular bash statements.

**To reproduce**
#in terminal:
tvservice -n | grep "Monitor" #grep search
#or
tvservice -n > tvservice_log.txt #file output 

**Expected behaviour**
Grep should search and out only find text.
File "tvservice_log.txt" should contain "tvservice -n" output.

**Actual behaviour**
See screenshot.
tvservise -s is working correctly.
tvservise -n does not work correctly with bash statement. 

**System**.
* Which model of Raspberry Pi?

* Which firmware version (`vcgencmd version`)?

RPI 3B+
tvservice installed with package libraspberrypi-bin 1.20190819~stretch-1.

* Which OS and version (`cat /etc/rpi-issue`)?
Raspberry Pi reference 2018-03-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 00013d7972122d1304aacda8fff5098f073ceb43, stage2

* Which firmware version (`vcgencmd version`)?
Aug 15 2019 12:06:42
Copyright (c) 2012 Broadcom
version 0e6daa5106dd4164474616408e0dc24f997ffcf3 (clean) (release) (start)

* Which kernel version
Linux retropie 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
[image]

JamesH65, 2019-11-17:
I suspect its writing to stderr. Redirect that as well, and it should work fine.
http://tldp.org/HOWTO/Bash-Prog-Intro-HOWTO-3.html

6by9, 2019-11-17:
tvservice -n
https://github.com/raspberrypi/userland/blob/master/host_applications/linux/apps/tvservice/tvservice.c#L1235
So "No device present" goes to stderr, "device_name=..." goes to stdout.

tvservice -s
https://github.com/raspberrypi/userland/blob/master/host_applications/linux/apps/tvservice/tvservice.c#L471
So that goes to stdout, but the "Error getting current display state" failure would go to stderr.

As JamesH65 says, redirect stderr should you wish to grep the error`` messages
```tvservice -n 2>&1 | grep device```

pelwell, 2019-11-17:
Or in bash you can use `tvservice -n |& grep device`.

JamesH65, 2019-11-17:
@pelwell I did not know that. (|&)

8street, 2019-11-17:
Thanks. Redirection works correctly.

closed 2019-11-17 by 8street