raspberrypi/userland #597: aarch64 build failed on raspberry pi 4

ykla, 2019-11-20:
`gentoo ~/userland # ./buildme --aarch64
~/userland/build/raspberry/release ~/userland                     CMake Warning at makefiles/cmake/arm-linux.cmake:5 (message):         *********************************************************
    *   CMAKE_TOOLCHAIN_FILE not defined                    *         *   This is correct for compiling on the Raspberry Pi   *
    *                                                       *
    *   If you are cross-compiling on some other machine    *
    *   then DELETE the build directory and re-run with:    *
    *   -DCMAKE_TOOLCHAIN_FILE=toolchain_file.cmake         *
    *                                                       *
    *   Toolchain files are in makefiles/cmake/toolchains.  *
    *********************************************************
Call Stack (most recent call first):
  CMakeLists.txt:24 (include)

-- Configuring done
-- Generating done
-- Build files have been written to: /root/userland/build/raspberry/release
[ 10%] Built target debug_sym_static
[ 10%] Built target debug_sym
[ 18%] Built target fdt
[ 42%] Built target vcos
[ 45%] Built target dtovl
[ 50%] Built target vchiq_arm
[ 54%] Building C object host_applications/linux/apps/dtoverlay/CMakeFiles/dtoverlay.dir/dtoverlay_main.c.o
[ 54%] Built target dtmerge
[ 62%] Built target vcilcs
[ 65%] Built target vchiq_test
[ 81%] Built target vchostif
[ 87%] Built target vcgencmd
[ 87%] Built target tvservice
[ 93%] Built target bcm_host
[ 96%] Built target vcmailbox
In file included from /root/userland/opensrc/helpers/libfdt/libfdt.h:54,
                 from /root/userland/host_applications/linux/apps/dtoverlay/dtoverlay_main.c:37:
/root/userland/opensrc/helpers/libfdt/libfdt_env.h:64: error: "__bitwise" redefined [-Werror]
   64 | #define __bitwise
      |
In file included from /usr/include/linux/stat.h:5,
                 from /usr/include/bits/statx.h:30,
                 from /usr/include/sys/stat.h:446,
                 from /root/userland/host_applications/linux/apps/dtoverlay/dtoverlay_main.c:33:
/usr/include/linux/types.h:22: note: this is the location of the previous definition
   22 | #define __bitwise __bitwise__
      |
cc1: all warnings being treated as errors
make[2]: *** [host_applications/linux/apps/dtoverlay/CMakeFiles/dtoverlay.dir/build.make:63: host_applications/linux/apps/dtoverlay/CMakeFiles/dtoverlay.dir/dtoverlay_main.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:687: host_applications/linux/apps/dtoverlay/CMakeFiles/dtoverlay.dir/all] Error 2
make: *** [Makefile:152: all] Error 2
[ 23%] Built target vcos
[ 28%] Built target vchiq_arm
[ 35%] Built target vcilcs
[ 51%] Built target vchostif
[ 54%] Built target vchiq_test
[ 60%] Built target bcm_host
[ 64%] Built target vcgencmd
[ 67%] Built target tvservice
[ 70%] Built target vcmailbox
[ 73%] Built target debug_sym
[ 76%] Built target debug_sym_static
[ 89%] Built target fdt
[ 92%] Built target dtovl
[ 93%] Building C object host_applications/linux/apps/dtoverlay/CMakeFiles/dtoverlay.dir/dtoverlay_main.c.o
In file included from /root/userland/opensrc/helpers/libfdt/libfdt.h:54,
                 from /root/userland/host_applications/linux/apps/dtoverlay/dtoverlay_main.c:37:
/root/userland/opensrc/helpers/libfdt/libfdt_env.h:64: error: "__bitwise" redefined [-Werror]
   64 | #define __bitwise
      |
In file included from /usr/include/linux/stat.h:5,
                 from /usr/include/bits/statx.h:30,
                 from /usr/include/sys/stat.h:446,
                 from /root/userland/host_applications/linux/apps/dtoverlay/dtoverlay_main.c:33:
/usr/include/linux/types.h:22: note: this is the location of the previous definition
   22 | #define __bitwise __bitwise__
      |
cc1: all warnings being treated as errors
make[2]: *** [host_applications/linux/apps/dtoverlay/CMakeFiles/dtoverlay.dir/build.make:63: host_applications/linux/apps/dtoverlay/CMakeFiles/dtoverlay.dir/dtoverlay_main.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:687: host_applications/linux/apps/dtoverlay/CMakeFiles/dtoverlay.dir/all] Error 2
make: *** [Makefile:152: all] Error 2
~/userland
gentoo ~/userland #`

pelwell, 2019-11-20:
We don't support a 64-bit userland. This is likely to change in the future, but we aren't there yet.

If you have a patch that fixes the problem then open a Pull Request.

6by9, 2019-11-20:
Duplicate of #582.

Later versions of libfdt have fixed this up - https://github.com/dgibson/dtc/commit/881012e443860c7662a7c53ea38f398d623f820c

Bumping to a newer version of libfdt should be safe, but it's not something I've checked out. It obviously needs to be tested with the firmware's manipulation of DT as well.

ykla, 2019-11-20:
But 32bit also makes errors. I used 2 ways.

Way 1:
git clone https://github.com/raspberrypi/userland.git
cd userland
mkdir build
cd build
cmake -DCMAKE_BUILD_TYPE=Release ..
make
---------------------------------------------------------------
/root/userland/interface/khronos/common/khrn_client_check_types.h:45:1: note: in expansion of macro 'vcos_static_assert'
   45 | vcos_static_assert(sizeof(EGLContext) == 4);
      | ^~~~~~~~~~~~~~~~~~
/root/userland/build/inc/interface/vcos/vcos_assert.h:303:69: error: size of array 'vcos_static_assert' is negative
  303 | #define vcos_static_assert(cond) __attribute__((unused)) extern int vcos_static_assert[(cond)?1:-1]
      |                                                                     ^~~~~~~~~~~~~~~~~~
/root/userland/interface/khronos/common/khrn_client_check_types.h:46:1: note: in expansion of macro 'vcos_static_assert'
   46 | vcos_static_assert(sizeof(EGLDisplay) == 4);
      | ^~~~~~~~~~~~~~~~~~
/root/userland/build/inc/interface/vcos/vcos_assert.h:303:69: error: size of array 'vcos_static_assert' is negative
  303 | #define vcos_static_assert(cond) __attribute__((unused)) extern int vcos_static_assert[(cond)?1:-1]
      |                                                                     ^~~~~~~~~~~~~~~~~~
/root/userland/interface/khronos/common/khrn_client_check_types.h:47:1: note: in expansion of macro 'vcos_static_assert'
   47 | vcos_static_assert(sizeof(EGLSurface) == 4);
      | ^~~~~~~~~~~~~~~~~~
/root/userland/build/inc/interface/vcos/vcos_assert.h:303:69: error: size of array 'vcos_static_assert' is negative
  303 | #define vcos_static_assert(cond) __attribute__((unused)) extern int vcos_static_assert[(cond)?1:-1]
      |                                                                     ^~~~~~~~~~~~~~~~~~
/root/userland/interface/khronos/common/khrn_client_check_types.h:48:1: note: in expansion of macro 'vcos_static_assert'
   48 | vcos_static_assert(sizeof(EGLClientBuffer) == 4);
      | ^~~~~~~~~~~~~~~~~~
/root/userland/build/inc/interface/vcos/vcos_assert.h:303:69: error: size of array 'vcos_static_assert' is negative
  303 | #define vcos_static_assert(cond) __attribute__((unused)) extern int vcos_static_assert[(cond)?1:-1]
      |                                                                     ^~~~~~~~~~~~~~~~~~
/root/userland/interface/khronos/common/khrn_client_check_types.h:49:1: note: in expansion of macro 'vcos_static_assert'
   49 | vcos_static_assert(sizeof(NativeDisplayType) == 4);
      | ^~~~~~~~~~~~~~~~~~
/root/userland/build/inc/interface/vcos/vcos_assert.h:303:69: error: size of array 'vcos_static_assert' is negative
  303 | #define vcos_static_assert(cond) __attribute__((unused)) extern int vcos_static_assert[(cond)?1:-1]
      |                                                                     ^~~~~~~~~~~~~~~~~~
/root/userland/interface/khronos/common/khrn_client_check_types.h:50:1: note: in expansion of macro 'vcos_static_assert'
   50 | vcos_static_assert(sizeof(NativePixmapType) == 4);
      | ^~~~~~~~~~~~~~~~~~
/root/userland/build/inc/interface/vcos/vcos_assert.h:303:69: error: size of array 'vcos_static_assert' is negative
  303 | #define vcos_static_assert(cond) __attribute__((unused)) extern int vcos_static_assert[(cond)?1:-1]
      |                                                                     ^~~~~~~~~~~~~~~~~~
/root/userland/interface/khronos/common/khrn_client_check_types.h:51:1: note: in expansion of macro 'vcos_static_assert'
   51 | vcos_static_assert(sizeof(NativeWindowType) == 4);
      | ^~~~~~~~~~~~~~~~~~
/root/userland/build/inc/interface/vcos/vcos_assert.h:303:69: error: size of array 'vcos_static_assert' is negative
  303 | #define vcos_static_assert(cond) __attribute__((unused)) extern int vcos_static_assert[(cond)?1:-1]
      |                                                                     ^~~~~~~~~~~~~~~~~~
/root/userland/interface/khronos/common/khrn_client_check_types.h:71:1: note: in expansion of macro 'vcos_static_assert'
   71 | vcos_static_assert(sizeof(GLintptr) == 4);
      | ^~~~~~~~~~~~~~~~~~
/root/userland/build/inc/interface/vcos/vcos_assert.h:303:69: error: size of array 'vcos_static_assert' is negative
  303 | #define vcos_static_assert(cond) __attribute__((unused)) extern int vcos_static_assert[(cond)?1:-1]
      |                                                                     ^~~~~~~~~~~~~~~~~~
/root/userland/interface/khronos/common/khrn_client_check_types.h:72:1: note: in expansion of macro 'vcos_static_assert'
   72 | vcos_static_assert(sizeof(GLsizeiptr) == 4);
      | ^~~~~~~~~~~~~~~~~~
In file included from /root/userland/build/inc/interface/vcos/vcos.h:144,
                 from /root/userland/interface/khronos/common/khrn_int_util.h:37,
                 from /root/userland/interface/khronos/common/khrn_client_check_types.h:31,
                 from /root/userland/interface/khronos/common/khrn_client.c:30:
/root/userland/interface/khronos/common/khrn_client.c: In function 'client_send_make_current':
/root/userland/interface/khronos/common/khrn_client.c:392:18: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
  392 |                  (unsigned int)(char *)thread->opengl.context, (unsigned int)(char *)thread->opengl.draw);
      |                  ^
/root/userland/build/inc/interface/vcos/vcos_logging.h:234:113: note: in definition of macro '_VCOS_LOG_X'
[... 596 lines cut ...]
[... cut ...]
/root/userland/interface/khronos/common/khrn_int_hash_asm.s:221: Error: unknown register alias 'BB'
/root/userland/interface/khronos/common/khrn_int_hash_asm.s:224: Error: unknown register alias 'DAT1'
make[2]: *** [interface/khronos/CMakeFiles/khrn_client.dir/build.make:102: interface/khronos/CMakeFiles/khrn_client.dir/common/khrn_int_hash_asm.s.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:601: interface/khronos/CMakeFiles/khrn_client.dir/all] Error 2
make: *** [Makefile:152: all] Error 2
~/userland
gentoo ~/userland #

6by9, 2019-11-20:
Trying to build without the ```--aarch64``` flag on a 64bit environment (you appear to be using gentoo) is not the same as building on and for a 32bit environment.

ykla, 2019-11-20:
> Trying to build without the `--aarch64` flag on a 64bit environment (you appear to be using gentoo) is not the same as building on and for a 32bit environment.

Yes. I try it but errors also . And I am using 64 bit Gentoo.

ykla, 2019-11-20:
OK. 
[](https://github.com/raspberrypi/userland/issues/582)

> satmandu commented on 15 Oct
> This fixes it for me:
> 
> sed -i 's/__bitwise/FDT_BITWISE/' rpi-userland/opensrc/helpers/libfdt/libfdt_env.h
> sed -i 's/__force/FDT_FORCE/' rpi-userland/opensrc/helpers/libfdt/libfdt_env.h
> This is what has been done in libfdt in newer kernels as far as I can tell.

Make it well. It works. Thanks.

6by9, 2019-11-20:
Closing as duplicate of #582.
Slightly odd as I've just tried the Gentoo64 build that sakaki has done for the Pi, and I don't get errors when building userland.

We'll try and find time to update libfdt, or at least cherry-pick that one commit, at some point in the future.

closed 2019-11-20 by 6by9

ykla, 2019-11-20:
My way to install 64bit userland：

```
sed -i 's/__bitwise/FDT_BITWISE/'  /root/userland/opensrc/helpers/libfdt/libfdt_env.h
sed -i 's/__force/FDT_FORCE/'  /root/userland/opensrc/helpers/libfdt/libfdt_env.h
```
```
cd /root/userland
mkdir build && cd build
cmake -DCMAKE_BUILD_TYPE=Release -DARM64=ON ../
make -j4 && make install
```
`cp -rfp /opt/vc/* /usr`

`nano /etc/ld.so.conf` add
`/opt/vc/lib`

`ldconfig`

Test:
```
gentoo ~/userland/build # ldd /opt/vc/bin/vcgencmd 
	linux-vdso.so.1 (0x0000007fb8c50000)
	libvchiq_arm.so => /usr/lib/libvchiq_arm.so (0x0000007fb8bd5000)
	libvcos.so => /usr/lib/libvcos.so (0x0000007fb8bb7000)
	libpthread.so.0 => /lib64/libpthread.so.0 (0x0000007fb8b84000)
	libdl.so.2 => /lib64/libdl.so.2 (0x0000007fb8b70000)
	librt.so.1 => /lib64/librt.so.1 (0x0000007fb8b58000)
	libc.so.6 => /lib64/libc.so.6 (0x0000007fb89d5000)
	/lib/ld-linux-aarch64.so.1 => /lib64/ld-linux-aarch64.so.1 (0x0000007fb8c22000)
gentoo ~/userland/build # vcgencmd 
Usage: vcgencmd [-t] command
Send a command to the VideoCore and print the result.

  -t          Time how long the command takes to complete
  -h, --help  Show this information

Use the command 'vcgencmd commands' to get a list of available commands

Exit status:
   0    command completed successfully
  -1    problem with VCHI
  -2    VideoCore returned an error

For further documentation please see
https://www.raspberrypi.org/documentation/raspbian/applications/vcgencmd.md

gentoo ~/userland/build # 

```

sakaki-, 2019-11-21:
> Slightly odd as I've just tried the Gentoo64 build that sakaki has done for the Pi, and I don't get errors when building userland.

So, I'm in the process of putting together a v1.5.2 release of that image now, and out of interest:
* userland builds succesfully under an out-of-box, unupdated v1.5.**1** release of [Gentoo64](https://github.com/sakaki-/gentoo-on-rpi-64bit), but
* fails to build (with the `__bitwise` issue) if the image is updated first (to bring its packages in line with the current upstream Gentoo tree).

@satmandu's [fixup](https://github.com/raspberrypi/userland/issues/582#issuecomment-541991743) seems to work though and [I have applied it](https://github.com/sakaki-/genpi64-overlay/commit/bc0bf8bf6eccd9b5466829abe7ca70ff7f41d563) to the latest ebuild.

hth, sakaki

sbromberger, 2019-11-28:
> My way to install 64bit userland：

Just to report in: I used @ykla's process to update and install userland on my RPi4 running Ubuntu 19.10. Thank you!!

SuddenGunter, 2020-01-17:
> I used @ykla's process to update and install userland on my RPi4 running Ubuntu 19.10. Thank you!!
@sbromberger same
./builtme didn't make it out of the box for me, but @ykla's guide allowed me to fix it