raspberrypi/userland #599: raspivid raw capture hangs

astillich-igniti, 2019-11-27:
**Describe the bug**
The tool raspivid hangs when capturing raw frames to any output. Running with preview only does not hang. Also, writing compressed output does not result in a hang.

**To reproduce**
Run the following command:
```
raspivid -v -fl -w 640 -h 480 -ex fixedfps -fps 55 -t 0 -rf gray -r capture.raw
```
Or any other variation which outputs raw video frames to a file / socket OR /dev/null.

**Expected behaviour**
I expect raspivid to capture frames continuously without hanging.

**Actual behaviour**
Add a clear and concise description of what actually happened.

**System**
Output of raspinfo is attached.

**Logs**
Output of raspinfo is attached.

**Additional context**

I tracked this down to `splitter_buffer_callback()` not being called after receiving a certain amount of frames which seems to vary slightly.

Adding the following lines to the function
```
   static size_t total = 0;
   total += buffer->length;
   printf("buffer chunk: %d, total: %d\n", buffer->length, total);
   fflush(stdout);
```
results in the following output (+/- a few bytes difference in each run):
```
buffer chunk: 460800, total: 460800
buffer chunk: 460800, total: 921600
....
buffer chunk: 460800, total: 51609600
buffer chunk: 460800, total: 52070400
```

[raspinfo.txt](https://github.com/raspberrypi/userland/files/3897290/raspinfo.txt)

JamesH65, 2019-11-27:
Have you tried with a standard output as well (-o fred.h264)? With verbose enabled, what sequence of output do you get?

astillich-igniti, 2019-11-27:
Yes, I tried that:

```Also, writing compressed output does not result in a hang.```

Note that lowering the resolution to 320x240 "fixes" it (hides the real issue).

Attached is a full verbose log up until the hang.
[raspivid-out.txt](https://github.com/raspberrypi/userland/files/3897456/raspivid-out.txt)

JamesH65, 2019-11-27:
I suspect you are are exceeding the write speed of the SD card.  640x480x55 is just over 16MB/s at greyscale, that's 128MBits/s and well over class 10 speed. You get a few buffers cached before it has to start syncing to SD card, at which point it all slows to a crawl. It also explains why dropping to a lower resolution works.

astillich-igniti, 2019-11-27:
That was my initial impression too. However, neither writing to /dev/null nor writing to a network location improves the situation. As a test I wrote a small C program which reads from stdin and discards the data. Piping raspivid output to this simple program results in a hang too. This (and writing to /dev/null) should not involve any SD card I/O.

Attached the output of `sudo vcdbg log assert` and `sudo vcdbg log msg` after a fresh boot and one failed capture
[log-assert.txt](https://github.com/raspberrypi/userland/files/3897510/log-assert.txt)
[log-msg.txt](https://github.com/raspberrypi/userland/files/3897511/log-msg.txt)

.

astillich-igniti, 2019-11-27:
As I wrote above, the issue is that the MMAL stack ceases to call the callback function `splitter_buffer_callback()` registered with `mmal_port_enable()` after a while.

astillich-igniti, 2019-11-27:
Incrementing a counter at the beginning of the callback and decrementing it at the end always results in the counter being 0 at the end of the funciton. 

I suspected that the stack would queue up calls to the callback while older callback requests are still running because they cannot flush data fast enough. This does **not** seem to be the case.

JamesH65, 2019-11-27:
In that case I have no idea, although it really does look like a speed of output problem. Does dropping the frame rate make any difference?

astillich-igniti, 2019-11-27:
When writing to /dev/null it starts to work reliably at 42 FPS (640x480 grayscale). Any frame rate above results in a hang.

astillich-igniti, 2019-11-27:
Now I have a lot of these messages in `sudo vcdbg log msg`:

```
2481787.163: mmal: mmal_port_send_buffer: ril.video_encode:in:0(OPQV): send failed: EINVAL
2481787.248: mmal: mmal_port_connected_output_cb: ril.camera:out:1(OPQV) could not send buffer on port ril.video_encode:in:0(OPQV) (EINVAL)
```

astillich-igniti, 2019-11-28:
I have enabled more logging and got a better log, when the capture hangs, the following message is written to the log:
```cameraRIL: cp_error_callback (video) with errorcode -2```
[log-msg2.txt](https://github.com/raspberrypi/userland/files/3901817/log-msg2.txt)

6by9, 2019-11-28:
errorcode -2 is that it can't allocate an image buffer for the new camera frame.
By the looks of it the video_encode has been passed frames and isn't producing output either.

I think what has happened is that the video encoder has been enabled but is unused. It'll therefore be getting frames from the camera and encoding them into the output FIFO (4MB IIRC). Nothing is consuming that output, therefore once the FIFO is full the encoder has to hold on to the input buffers. Then we run out of input buffers and the splitter/preview/raw image path stalls too.

Looking at it, you've told raspvid (by omission) that it is to be capturing by default, and then not provided a valid file to pass it to. That is the cause of it not passing buffers for that encoded output. Add ```-i pause``` to tell it that the initial state is paused and the raw capture and preview run fine.

NB Raspivid is there to demo capturing encoded data. Raspividyuv is the app to demo capturing raw image data.

astillich-igniti, 2019-11-28:
Thank you for the clarifications. Turns out that the camera was at fault. I just received two other cameras and at least one of them is working fine. Issue can be closed.

6by9, 2019-11-28:
I don't believe this is totally down to a faulty camera. You are correct that it will stall under those conditions

@JamesH65 up to you whether you want to know investigate further. It'd be better to not set ```MMAL_PARAMETER_CAPTURE``` on camera->output[1] ever if we're not actually going to be encoding the output, although I think it's more the ```if (state.callback_data.file_handle)```at https://github.com/raspberrypi/userland/blob/master/host_applications/linux/apps/raspicam/RaspiVid.c#L2778 resulting in the output buffer never being sent to the active output port that is stalling things.
You could even back up further and not enable the camera->encoder tunnel or encoder output, or disable "stuff" (initial state may be sufficient? Or abort?) if no filename is provided or the file fails to open.

JamesH65, 2019-11-28:
@6by9 The comment above that it also fails when an output filename is used (albeit /dev/null) has thrown me a bit. Wouldn't you expect that to work? That when combined with it working at lower frame frames make the above explanation a bit weird?

astillich-igniti, 2019-11-28:
The camera being at fault was a false alert. After rebooting it started to hang again. Specifying `-i pause` solves the hanging, though.