raspberrypi/userland #600: Introducing latency to preview mode

PhilippeDuvivier, 2019-12-04:
Hi,
Great stuff this raspivid.
I am trying to use it together with an AUVIDEA B101 HDMI input board in order to create a simple parametric HDMI video delayer with a PI, ie to display the input HDMI on the output with Nx100ms delay.
One option would be to introduce this as a parameter to the preview mode. Does that make sense at all?
Thanks

JamesH65, 2019-12-04:
It would be possible. Probably best to continue any discussion on the forum - this tracker is for bug reports rather than tech help.

6by9, 2019-12-04:
As per the message that is displayed every time you run raspivid with a B101 or other TC358743 based HDMI capture board
> The driver for the TC358743 HDMI to CSI2 chip you are using is NOT supported. They were written for a demo purposes only, and are in the firmware on an as-is basis and therefore requests for support or changes will not be acted on.

The TC358743 is supported via V4L2 and works via things such as GStreamer. Use ```dtoverlay=tc358743``` should load up the appropriate drivers, although you will need to add the ```i2c_pins_28_29=1``` parameter if you're on Pi B+, 2, 0, and 0W.
Using the GStreamer ```queue``` component and associated ```min-threshold-time``` parameter should be able to deliver what you want. Further support on the use of that should be done on the forum, not here.

closed 2019-12-04 by 6by9

PhilippeDuvivier, 2019-12-05:
Thanks for this answer
I have created a Topic in the forum, please support me there :-)
https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=258529
Cheers
Philippe

Philippe Duvivier
Co-founder & CTO
+33 6 95 13 24 42
odiho - Sound On Demand 
philippe.duvivier@odiho.com
www.odiho.com