raspberrypi/userland #604: RPi4: Bluetooth crash while using BT keyboard

dfunkt, 2019-12-19:
**Describe the bug**
Bluetooth crashes while using BT keyboard (Anne Pro)

**To reproduce**
Normal usage, no specific trigger as far as I can tell

**Expected behaviour**
No crash expected

**Actual behaviour**
Bluetooth daemon crashes, having to manually restart it in order to use the keyboard

**System**
raspinfo: https://pastebin.com/guNUYb6W

**Logs**
```
systemctl status bluetoothd:
Dec 19 18:43:49 raspberrypi bluetoothd[6266]: Endpoint registered: sender=:1.36 path=/org/bluez/hci0/A2DP/SBC/Source/1
Dec 19 18:43:49 raspberrypi bluetoothd[6266]: Endpoint registered: sender=:1.36 path=/org/bluez/hci0/A2DP/SBC/Source/2
Dec 19 18:43:50 raspberrypi bluetoothd[6266]: Scan Refresh: notifications not enabled Encryption required before read/write
Dec 19 18:46:35 raspberrypi bluetoothd[6266]: Error reading PNP_ID value: Unexpected error code
Dec 19 18:46:39 raspberrypi bluetoothd[6266]: Error reading PNP_ID value: Unexpected error code
Dec 19 18:46:42 raspberrypi bluetoothd[6266]: Error storing included service - can't find it
Dec 19 18:46:42 raspberrypi bluetoothd[6266]: Error storing included service - can't find it
Dec 19 18:46:42 raspberrypi bluetoothd[6266]: Error storing included service - can't find it
Dec 19 18:46:43 raspberrypi systemd[1]: bluetooth.service: Main process exited, code=killed, status=11/SEGV
Dec 19 18:46:43 raspberrypi systemd[1]: bluetooth.service: Failed with result 'signal'.
```

**Additional context**
The "Bluetooth: hci0: security requested but not available" message went away after I manually added the keyboard (+trust) using bluetoothctl, since I was unable to pair it using the bluetooth menu in the task bar. On my other linux&windows machine I was usually asked for a PIN code in order to pair, but I suppose this might be a different issue.
Also using 5GHz WiFi if that matters.

pelwell, 2019-12-19:
A log taken after you fixed the security issue would be preferable. And can you confirm that:
1. the keyboard works for some time before a random bluetoothd crash, and
2. restarting bluetoothd is sufficient to bring back the keyboard (specifically, and Bluetooth generally)?

dfunkt, 2019-12-19:
The answer is yes to both questions.
I'll come back with a fresh log in a few mins.

dfunkt, 2019-12-19:
Logs are taken after a reboot

dmesg log: https://pastebin.com/LSDyphbu
```
systemctl status bluetoothd:
Dec 19 19:27:41 raspberrypi bluetoothd[556]: Failed to set privacy: Rejected (0x0b)
Dec 19 19:27:41 raspberrypi bluetoothd[556]: Endpoint registered: sender=:1.15 path=/org/bluez/hci0/A2DP/SBC/Source/2
Dec 19 19:27:41 raspberrypi bluetoothd[556]: No cache for FC:A3:08:00:61:61
Dec 19 19:27:41 raspberrypi bluetoothd[556]: BATT attribute not found
Dec 19 19:27:41 raspberrypi bluetoothd[556]: batt-profile profile accept failed for FC:A3:08:00:61:61
Dec 19 19:27:41 raspberrypi bluetoothd[556]: Scan Parameters attribute not found
Dec 19 19:27:41 raspberrypi bluetoothd[556]: Scan Parameters Client Driver profile accept failed for FC:A3:08:00:61:61
Dec 19 19:27:41 raspberrypi bluetoothd[556]: GAP attribute not found
Dec 19 19:27:41 raspberrypi bluetoothd[556]: gap-profile profile accept failed for FC:A3:08:00:61:61
Dec 19 19:27:41 raspberrypi bluetoothd[556]: input-hog profile accept failed for FC:A3:08:00:61:61
```

pelwell, 2019-12-19:
Was that kernel log taken after a keyboard failure? I just want to confirm that there are no hci0: messages generated.

dfunkt, 2019-12-19:
No, that was from a fresh boot, the keyboard is still working atm.

dfunkt, 2019-12-19:
And it's dead again.
dmesg: https://pastebin.com/QPXTJDnM

```
systemctl status bluetooth:
Dec 19 19:47:31 raspberrypi bluetoothd[556]: Error reading PNP_ID value: Unexpected error code
Dec 19 19:47:34 raspberrypi bluetoothd[556]: Error reading PNP_ID value: Unexpected error code
Dec 19 19:47:37 raspberrypi bluetoothd[556]: Error reading PNP_ID value: Unexpected error code
Dec 19 19:47:43 raspberrypi bluetoothd[556]: Error reading PNP_ID value: Unexpected error code
Dec 19 19:47:58 raspberrypi bluetoothd[556]: Error reading PNP_ID value: Unexpected error code
Dec 19 19:49:57 raspberrypi bluetoothd[556]: Error storing included service - can't find it
Dec 19 19:49:57 raspberrypi bluetoothd[556]: Error storing included service - can't find it
Dec 19 19:49:57 raspberrypi bluetoothd[556]: Error storing included service - can't find it
Dec 19 19:49:58 raspberrypi systemd[1]: bluetooth.service: Main process exited, code=killed, status=11/SEGV
Dec 19 19:49:58 raspberrypi systemd[1]: bluetooth.service: Failed with result 'signal'.
```

dfunkt, 2019-12-20:
Preliminary testing shows that disabling WiFi (switched to ethernet) fixes the crashing. Been running for a fair bit of time now without any crashes. Maybe there's some coexistence issue?

closed 2019-12-21 by dfunkt