raspberrypi/userland #607: error: expected ‘)’ before ‘sizeof’

RaspberryPiFan, 2019-12-30:
**Describe the bug**
I'm trying to build anbox, which uses this code, but I always get the error: `/opt/vc/include/interface/vcos/vcos_types.h:205:22: error: expected unqualified-id before ‘sizeof’`

**To reproduce**
Compile anbox on a Raspberry Pi

**Expected behaviour**
Builds as usually
**Actual behaviour**
Add a clear and concise description of what actually happened.

**System**
```
System Information
------------------

Raspberry Pi 4 Model B Rev 1.1
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2019-12-24
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 86a959eb5160c29587927aed3afe5eb1d0530522, stage5

Linux raspberrypi 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux
Revision	: c03111
Serial		: 10000000d327233d
Model		: Raspberry Pi 4 Model B Rev 1.1
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0

Videocore information
---------------------

Sep 24 2019 17:34:30 
Copyright (c) 2012 Broadcom
version cd3add54955f8fa065b414d8fc07c525e7ddffc8 (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Dateisystem    1K-Blöcke  Benutzt Verfügbar Verw% Eingehängt auf
/dev/root       60899220 20355236  37427440   36% /
devtmpfs         1867796        0   1867796    0% /dev
tmpfs            1999892   115740   1884152    6% /dev/shm
tmpfs            1999892     9136   1990756    1% /run
tmpfs               5120        4      5116    1% /run/lock
tmpfs            1999892        0   1999892    0% /sys/fs/cgroup
/dev/mmcblk0p6    258094    53033    205062   21% /boot
tmpfs             399976        8    399968    1% /run/user/1000
/dev/sda1        7610368  2525752   5084616   34% /media/pi/ROM
/dev/loop0         77824    77824         0  100% /snap/core/8270
/dev/loop1        247424   247424         0  100% /snap/anbox/91

Filename				Type		Size	Used	Priority
/var/swap                               file		102396	39424	-2

Package version information
---------------------------
raspberrypi-ui-mods:
  Installiert:           1.20191023
raspberrypi-sys-mods:
  Installiert:           20191105
openbox:
  Installiert:           3.6.1-8+rpt3
lxpanel:
[... 788 lines cut ...]
[40258.575340] usb 1-1.3: USB disconnect, device number 14
[40258.973787] usb 1-1.3: new low-speed USB device number 15 using xhci_hcd
[40259.111606] usb 1-1.3: New USB device found, idVendor=046d, idProduct=c016, bcdDevice= 3.40
[40259.111614] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[40259.111619] usb 1-1.3: Product: Optical USB Mouse
[40259.111624] usb 1-1.3: Manufacturer: Logitech
[40259.121756] input: Logitech Optical USB Mouse as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3:1.0/0003:046D:C016.000D/input/input14
[40259.121990] hid-generic 0003:046D:C016.000D: input,hidraw0: USB HID v1.10 Mouse [Logitech Optical USB Mouse] on usb-0000:01:00.0-1.3/input0
[41435.706917] bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if you need this.
[41944.519884] ashmem_linux: loading out-of-tree module taints kernel.
[43905.256060] new mount options do not match the existing superblock, will be ignored

EEPROM
------

BOOTLOADER: up-to-date
CURRENT: Di 10. Sep 10:41:50 UTC 2019 (1568112110)
 LATEST: Di 10. Sep 10:41:50 UTC 2019 (1568112110)
VL805: up-to-date
CURRENT: 000137ab
 LATEST: 000137ab

```

I think this is the right place to post this, or should I post this at anbox?

RaspberryPiFan, 2019-12-30:
```
/opt/vc/include/interface/vcos/vcos_types.h:205:22: error: expected unqualified-id before ‘sizeof’
 #define countof(x) ( sizeof((x)) / sizeof((x)[0]))
                      ^~~~~~
/home/pi/anbox2/external/glm/glm/detail/setup.hpp:994:25: note: in expansion of macro ‘countof’
   constexpr std::size_t countof(T const (&)[N])
                         ^~~~~~~
/opt/vc/include/interface/vcos/vcos_types.h:205:22: error: expected ‘)’ before ‘sizeof’
 #define countof(x) ( sizeof((x)) / sizeof((x)[0]))
                    ~ ^~~~~~
/home/pi/anbox2/external/glm/glm/detail/setup.hpp:994:25: note: in expansion of macro ‘countof’
   constexpr std::size_t countof(T const (&)[N])
                         ^~~~~~~

```
The error

pelwell, 2019-12-30:
The glm (OpenGL Mathematics) code declares a countof member functions, which is incompatible with using countof as a macro as VCOS does. countof works as expected in the userland repo, and we can't be held responsible for other uses that are less successful.

closed 2019-12-30 by pelwell