raspberrypi/userland #611: Build error on Manjaro

solskogen, 2020-01-14:
**Describe the bug**
Building userland on aarch64 / Manjaro ARM gives this error:

Scanning dependencies of target bcm_host
In file included from /home/solskogen/userland/opensrc/helpers/libfdt/libfdt.h:54,
                 from /home/solskogen/userland/host_applications/linux/apps/dtoverlay/dtoverlay_main.c:37:
/home/solskogen/userland/opensrc/helpers/libfdt/libfdt_env.h:64: error: "__bitwise" redefined [-Werror]
   64 | #define __bitwise
      |
In file included from /usr/include/linux/stat.h:5,
                 from /usr/include/bits/statx.h:30,
                 from /usr/include/sys/stat.h:446,
                 from /home/solskogen/userland/host_applications/linux/apps/dtoverlay/dtoverlay_main.c:33:
/usr/include/linux/types.h:22: note: this is the location of the previous definition
   22 | #define __bitwise __bitwise__
      |
Scanning dependencies of target vcgencmd
[ 31%] Building C object host_applications/linux/apps/gencmd/CMakeFiles/vcgencmd.dir/gencmd.c.o
cc1: all warnings being treated as errors
make[2]: *** [host_applications/linux/apps/dtoverlay/CMakeFiles/dtoverlay.dir/build.make:63: host_applications/linux/apps/dtoverlay/CMakeFiles/dtoverlay.dir/dtoverlay_main.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2957: host_applications/linux/apps/dtoverlay/CMakeFiles/dtoverlay.dir/all] Error 2

**To reproduce**
Run ./buildme --aarch64 on Manjaro ARM 

[solskogen@rpi4-desktop userland]$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/aarch64-unknown-linux-gnu/9.2.0/lto-wrapper
Target: aarch64-unknown-linux-gnu
Configured with: /build/gcc/src/gcc/configure --prefix=/usr --libdir=/usr/lib --libexecdir=/usr/lib --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=https://github.com/archlinuxarm/PKGBUILDs/issues --enable-languages=c,c++,fortran,go,lto,objc,obj-c++,d --enable-shared --enable-threads=posix --with-system-zlib --with-isl --enable-__cxa_atexit --disable-libunwind-exceptions --enable-clocale=gnu --disable-libstdcxx-pch --disable-libssp --enable-gnu-unique-object --enable-linker-build-id --enable-lto --enable-plugin --enable-install-libiberty --with-linker-hash-style=gnu --enable-gnu-indirect-function --disable-multilib --disable-werror --enable-checking=release --enable-default-pie --enable-default-ssp --host=aarch64-unknown-linux-gnu --build=aarch64-unknown-linux-gnu --with-arch=armv8-a --enable-fix-cortex-a53-835769 --enable-fix-cortex-a53-843419 gdc_include_dir=/usr/include/dlang/gdc
Thread model: posix
gcc version 9.2.0 (GCC)

pelwell, 2020-01-14:
Duplicate of #582.

closed 2020-01-14 by pelwell

pelwell, 2020-01-14:
#582 should be fixed now.

solskogen, 2020-01-14:
Great, thanks! :)