raspberrypi/userland #614: Possible issue related to 64-bit client updates

drone-guy, 2020-02-13:
**Describe the bug**
I believe there is a bug / race condition in allocation / release of user context in mmal vc client. It is possible that the issue has been introduced recently with 64-bit -related changes, since I noticed my custom application (somewhat similar to raspivid) started crashing when I built and installed mmal libs from latest master branch. Used GDB to find the cause of seg fault (function pointer is null) and added print statements to help debug. Added some flags to CMakefiles (-ggdb, -mcpu but those should not be related to the crash)

**To reproduce**
use the forked version of userland, as of the following commit : https://github.com/drone-guy/userland/commit/ab216b6a0e289094ad2e79511e9c1e09d66547b8 . build on raspberry pi using buildme and then run test_raspivid script in root directory of the repository. Then wait for some time.. Sometimes the issue happens in less than 5 minutes, sometimes it can run for at least an hour or more. There were several changes to use case that are different from original raspivid:
* encoder output buffer pool is of size 3, not 1
* release the buffer header very early in the callback, basically do no processing

The crash does *not occur* if :
* you do not request motion vectors (no pending back to back callbacks)
* release the buffer header AFTER replenishing the buffer pool (effectively delaying the buffer going back to the port by one callback) -- THIS IS MY WORKAROUND, which works if buffer pool is just one extra than you need.

**Expected behaviour**
Should not crash

**Actual behaviour**
Segmentation fault was reported upon crash. I used GDB and print statements to narrow down the problem. See  for crash log (link below).. Crash happens at the following line since callback function is a null pointer : https://github.com/drone-guy/userland/blob/vc_client_segfault/interface/mmal/vc/mmal_vc_client.c#L681 . Note (from crash log) that it seems that msg->drvbuf.client_context is being called twice with same value 0xfedc0004 (the last time it is called, it's actually released but not allocated !!!)

**System**
Raspberry Pi 4 Model B Rev 1.1
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
--
Jan 10 2020 16:53:17 
Copyright (c) 2012 Broadcom
version fdb5c37e330e7cb3027ac4fcc5b1cd5f244b351f (clean) (release) (start_x)
--
Linux raspberrypi 4.19.93-v7l+ #1290 SMP Fri Jan 10 16:45:11 GMT 2020 armv7l GNU/Linux

**Logs**
https://github.com/drone-guy/userland/blob/vc_client_segfault/crashlog.txt

6by9, 2020-02-14:
Apologies - the 64bit support has been reverted in the official releases due to seeing the same situation as you are reporting via Kodi, but that revert wasn't done here in userland. Generally Kodi was taking 45mins+ to reproduce. I've not had a chance to dig into it as yet.
If raspivid can trigger it within a few minutes then it'll be easier to track down.

The VPU shouldn't be returning the same client context handle on 2 buffers, it's a value that is assigned on the client when the buffer is sent to the port, and should be returned with the buffer.

It may be that there's been a bug lurking in there all along as we've always had a ```vcos_blockpool_alloc``` for the client_context at almost exactly the same point as the ```mmal_vc_allocate_client_context``` is (https://github.com/drone-guy/userland/blob/vc_client_segfault/interface/mmal/vc/mmal_vc_api.c#L610), and the releases are likewise at the same point https://github.com/drone-guy/userland/blob/vc_client_segfault/interface/mmal/vc/mmal_vc_api.c#L544 (Hmm, those should be in the opposite order, but nothing should block on them). In the old code there is only a dereference to client_context rather than a lookup, and I haven't checked what ```vcos_blockpool_free``` does for a double free.

drone-guy, 2020-02-15:
Thank you for the reply, 6by9. At least this confirms the issue.. As I mentioned, I have found a workaround, which is not to release the newly returned buffer from the pool right away, but delay that by one callback (simply by releasing available buffers from pool to port before freeing the received buffer in the callback). The time to reproduce is quite unpredictable. At some point I thought that I was insane because I could no longer reproduce the bug when running the same code, but luckily (???) the crash happened again. I cannot find any way to reproduce more reliably and quickly than the modified raspivid example I made for demonstrating the bug.. If I am able to find a quicker way, I will let you know..

6by9, 2024-02-07:
Closing as 64bit support was reverted, and MMAL is now deprecated.

closed 2024-02-07 by 6by9