raspberrypi/userland #618: ImageFx component - GPU memory leak

techyian, 2020-04-12:
**Is this the right place for my bug report?**
MMAL issue

**Describe the bug**
I think this may be an edge case when working with MMAL but I seem to have discovered a  memory leak when wiring up the ImageFx component within MMALSharp. What I have found is if a user doesn't set the native encoding type against the component's output port then memory assigned to the GPU's heap is eventually starved with continued use. To be clear, this is the `encoding` member of the `MMAL_ES_FORMAT_T` struct. I wasn't sure whether this component featured the format conversion block so during testing I decided not to set the encoding type in the hope that it would be configured on my behalf when connecting the component to the camera's still port. 

My pipeline so far looks as follows:

Camera (Still port) -> ImageFx -> Image Encoder

Eventually I receive a ENOSPC native error. Tested using `sudo vcgencmd get_mem reloc`.

Easy fix is just to set the encoding type, but I thought I'd mention it.

**Expected behaviour**
Unsure whether this is expected or not. If the component doesn't have the format conversion block I think it should handle not having the encoding type set and should use whatever format is passed to it from earlier in the pipeline.

**Actual behaviour**
GPU Memory leak.

**System**

Pi 3B+

Raspberry Pi reference 2020-02-13
Feb 12 2020 12:39:27

Copyright (c) 2012 Broadcom
version 53a54c770c493957d99bf49762dfabc4eee00e45 (clean) (release) (start_x)

Linux raspberrypi 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l GNU/Linux

6by9, 2020-04-12:
```encoding``` is being left at 0? That is erroneous, and shouldn't be permitted in mmal_port_format_commit. I'll have a look at correcting that after the bank holiday.

techyian, 2020-04-12:
Thanks, it's not something I'd usually do but I'm just trying to familiarise myself with the component and noticed it during testing.