raspberrypi/userland #620: vcgencmd - usability enhancements not yet in Raspbian binary

andrum99, 2020-05-02:
**Describe the bug**
This is not a bug as such, more an issue with a contribution which I made some time ago that has not yet made it into Raspbian, yet has been in this source repo for some time.

The issue is that some usability enhancements which I made to the vcgencmd tool have not yet made it into Raspbian, despite having been in this source repo since 1 May 2019. The usability enhancements provide some basic online help, a link to a web page with full help, and more useful feedback when certain commands are issued when they do not return any useful information from the firmware. I believe that users would find these enhancements useful, otherwise I would not have spent time implementing them. If they don't get used in Raspbian, then there wasn't really any point in me producing them. They have been tested and have been shown to work correctly.

6by9, 2020-05-02:
Apologies, this is likely to be that userland is a subset of the internal firmware source repo, and releases get built from that. PRs to this repo have to be manually cherry-picked across to that repo (normally by popcornmix), and I suspect that has been missed.

I'll do that on Monday unless @popcornmix picks up on it first.
It looks like it's https://github.com/raspberrypi/userland/commit/517cdc30da167d81a485e7a994e02cec2390a269 and possibly https://github.com/raspberrypi/userland/commit/91da39af21508f02bf734b135ba118a9fa022dad that you're referencing.

andrum99, 2020-05-02:
Not a problem - it was waiting for James to complete the documentation page it references in the official documentation repo, then obviously got forgotten about, by me as well. 😮😁

It may just be https://github.com/raspberrypi/userland/commit/517cdc30da167d81a485e7a994e02cec2390a269 - I've a feeling https://github.com/raspberrypi/userland/commit/91da39af21508f02bf734b135ba118a9fa022dad may have been integrated to your internal repo first before it appeared in this public repo, or I may have imagined it.

andrum99, 2020-05-20:
Hi @6by9 - has 517cdc3 made it to the internal repo?

pelwell, 2020-05-21:
Yes - this patch was merged yesterday, and will be in subsequent releases (except where a release candidate has already been selected).

popcornmix, 2020-05-21:
It's in rpi-update version of vcgencmd now.

andrum99, 2020-05-21:
> It's in rpi-update version of vcgencmd now.

Confirmed - many thanks!

closed 2020-06-05 by popcornmix