raspberrypi/userland #626: OpenMAX insufficient resources when getting handle for egl_render component

ardera, 2020-05-19:
## Description
```
OMX_GetHandle(..., "OMX.broadcom.egl_render", ..., {EventHandler, EmptyBufferDone, FillBufferDone});
```
fails with `OMX_ErrorInsufficientResources`

## To Reproduce
I'm afraid I can't give a reduced example for this one. I encountered this when I was trying to play the `/opt/vc/src/hello_pi/hello_video/test.h264` video (without Desktop) with GStreamer, using the following command:
```shell
gst-launch-1.0 --gst-debug omx*:5 filesrc location= /opt/vc/src/hello_pi/hello_video/test.h264 ! h264parse ! omxh264dec ! fakevideosink
```
_This needs `gstreamer1.0-tools` and `gstreamer1.0-omx` to be installed._

## Expected behaviour
Expected is a return value of `OMX_ErrorNone`, since the GPU should have enough memory. I tried allocating 16, 128, 256 and 512MB to the GPU, (I'm not sure the Pi 4 cares about how much mem is allocated to its GPU) all with the same error. I also tried using the legacy graphics driver, with exactly the same result.

## Actual behaviour
In reality, `OMX_ErrorInsufficientResources` is returned. I'm not sure where exactly the failed allocation ocurrs, but it looks like it's in [this](https://github.com/raspberrypi/userland/blob/291f9cb826d51ac30c1114cdc165836eacd8db52/interface/vmcs_host/vcilcs_out.c#L756) procedure.

Also, some `vcdbg` outputs before and after running `gstreamer`:

### `vcdbg reloc stats`
**Before executing gstreamer:**
<details>

```
Relocatable heap version 4 found at 0x30000000
total space allocated is 236M, with 236M relocatable, 0 legacy and 0 offline
0 legacy blocks of size 2359296

small_allocs                  : 25
allocs                        : 17
alloc_fails                   : 0
legacy_block_fails            : 0
compactions                   : 0
discard_compactions           : 0
aggressive_compactions        : 0
aggressive_compaction_waits   : 0
aggressive_compaction_timeouts: 0
locks                         : 0
small_locks                   : 0

free list at 0x3ea82d20
235M free memory in 1 free block(s)
largest free block is 235M bytes
```

</details>

**After:**
<details>

```
Relocatable heap version 4 found at 0x30000000
total space allocated is 236M, with 236M relocatable, 0 legacy and 0 offline
0 legacy blocks of size 2359296

small_allocs                  : 25
allocs                        : 18
alloc_fails                   : 0
legacy_block_fails            : 0
compactions                   : 0
discard_compactions           : 0
aggressive_compactions        : 0
aggressive_compaction_waits   : 0
aggressive_compaction_timeouts: 0
locks                         : 0
small_locks                   : 0

free list at 0x3ea82ae0
[... 1580 lines cut ...]

Some more logs from gstreamer, if needed:

<details>

```
omxvideodec gstomxvideodec.c:258:gst_omx_video_dec_open:<omxh264dec-omxh264dec0> Opening decoder
0:00:00.104451988  1390   0xbf0260 DEBUG                    omx gstomx.c:152:gst_omx_core_acquire: Successfully loaded core '/opt/vc/lib/libopenmaxil.so'
0:00:00.104812762  1390   0xbf0260 DEBUG                    omx gstomx.c:167:gst_omx_core_acquire: Successfully initialized core '/opt/vc/lib/libopenmaxil.so'
0:00:00.105490942  1390   0xbf0260 DEBUG                    omx gstomx.c:807:gst_omx_component_new:<omxh264dec-omxh264dec0> Successfully got component handle 0xc11800 (OMX.broadcom.video_decode) from core '/opt/vc/lib/libopenmaxil.so'
0:00:00.105580607  1390   0xbf0260 DEBUG                    omx gstomx.c:969:gst_omx_component_get_state:<omxh264dec-omxh264dec0> Getting state of video_decode
0:00:00.105605274  1390   0xbf0260 DEBUG                    omx gstomx.c:1019:gst_omx_component_get_state:<omxh264dec-omxh264dec0> video_decode returning state Loaded
0:00:00.105626533  1390   0xbf0260 DEBUG                    omx gstomx.c:1041:gst_omx_component_add_port:<omxh264dec-omxh264dec0> video_decode adding port 130
0:00:00.105648736  1390   0xbf0260 DEBUG                    omx gstomx.c:1132:gst_omx_component_get_parameter:<omxh264dec-omxh264dec0> Getting video_decode parameter at index 0x02000001
0:00:00.105740476  1390   0xbf0260 DEBUG                    omx gstomx.c:1135:gst_omx_component_get_parameter:<omxh264dec-omxh264dec0> Got video_decode parameter at index 0x02000001: None (0x00000000)
0:00:00.105785439  1390   0xbf0260 DEBUG                    omx gstomx.c:1041:gst_omx_component_add_port:<omxh264dec-omxh264dec0> video_decode adding port 131
0:00:00.105808605  1390   0xbf0260 DEBUG                    omx gstomx.c:1132:gst_omx_component_get_parameter:<omxh264dec-omxh264dec0> Getting video_decode parameter at index 0x02000001
0:00:00.105908808  1390   0xbf0260 DEBUG                    omx gstomx.c:1135:gst_omx_component_get_parameter:<omxh264dec-omxh264dec0> Got video_decode parameter at index 0x02000001: None (0x00000000)
0:00:00.105929827  1390   0xbf0260 DEBUG            omxvideodec gstomxvideodec.c:327:gst_omx_video_dec_open:<omxh264dec-omxh264dec0> Opened decoder
0:00:00.105947160  1390   0xbf0260 DEBUG            omxvideodec gstomxvideodec.c:330:gst_omx_video_dec_open:<omxh264dec-omxh264dec0> Opening EGL renderer
0:00:00.106112677  1390   0xbf0260 ERROR                    omx gstomx.c:799:gst_omx_component_new:<omxh264dec-omxh264dec0> Failed to get component handle 'OMX.broadcom.egl_render' from core '/opt/vc/lib/libopenmaxil.so': 0x80001000
0:00:00.106133917  1390   0xbf0260 DEBUG                    omx gstomx.c:209:gst_omx_core_release: Releasing core 0xc18e28
```

</details>

6by9, 2020-05-19:
```
System Information
------------------

Raspberry Pi 4 Model B Rev 1.1
```
The Pi4 doesn't support the firmware GLES/EGL interface, therefore the egl_render component is disabled. Otherwise it will fail on enabling the component, so better to fail early.

Someone needs to look again at disentangling the GStreamer use of egl_render within omxh264dec, but the other approach is to use v4l2h264dec instead.

ardera, 2020-05-19:
> The Pi4 doesn't support the firmware GLES/EGL interface, therefore the egl_render component is disabled. Otherwise it will fail on enabling the component, so better to fail early.

Okay that makes sense. Still, `OMX_ErrorInsufficientResources` seems like a kinda weird error to return in that case. I think `OMX_ErrorComponentNotFound` would be more fitting, even though it's not that precise as well.

`v4l2h264dec` works as expected! Thanks!

6by9, 2020-05-19:
> Okay that makes sense. Still, OMX_ErrorInsufficientResources seems like a kinda weird error to return in that case.

The component module is there, but deliberately fails the create as it checks on the platform. The return value from create within the Broadcom framework is a pointer to the component, or NULL. NULL is translated to ```OMX_ErrorInsufficientResources```.

The module choice is done through a dll loading mechanism, therefore there is no central list that we can nobble to drop egl_render from on Pi4. I also don't like messing with core code unless really necessary, particularly when it comes to IL!

closed 2020-05-19 by ardera

ardera, 2020-05-26:
@6by9 What is the replacement for the `egl_render` component on a Pi 4? The `egl_render` component also handled YUV -> RGB conversion before, it seems. There must a faster way for rendering into a GL texture (with no access to shaders) other than manual conversion and then `glTexImage2D`?

Btw it seems gstreamer forgets to do that YUV -> RGB conversion as well, since the colors are really really weird and not at all what they should be when using glImageSink.

6by9, 2020-05-26:
Import the buffers into EGL using dmabufs and eglCreateImageKHR.
The Pi4 has a texture formatter hardware block that can take YUV images in and create a texture from it.

It may have bitrotted, but https://github.com/6by9/drm_mmal/blob/x11_export/drm_mmal.c or https://github.com/6by9/drm_mmal/blob/x11/drm_mmal.c should work as very basic examples for merging MMAL and Mesa EGL (written by Eric originally).
The source of the dmabuf is largely irrelevant as to whether it is a dma-heap, V4L2 buffer, vcsm-cma, DRM, or something else. EGL shouldn't even require it to be contiguous as V3D has an IOMMU, but I wouldn't put money on that.

ardera, 2020-06-02:
Thank you very much!

That helped me resolve my issue.

Though I didn't exactly do it the way you suggested. Instead of rendering the video to a texture, I implemented hardware plane support so I can render "around" the video output of omxplayer. HW plane support is basically impossible to achieve with the framework I am using, but thanks to some hacks involving EGLImages, I was able to work around that. Your comment pointed me in the right direction.

Btw, is it normal that I have to reflect the y-axis of the DRM overlay plane to get it to draw correctly? It's somehow scanned out upside down with the default settings. Also, is there some reason the number of available overlay planes is so limited?

6by9, 2020-06-02:
What is it that you're rendering? GL output? In which case, yes, as GL works with bottom left as the origin, whilst DRM uses top left.

If you look at the master branch of drm_mmal, it takes MMAL buffers from the video_decode component and pokes them straight into DRM. No y-reflect required there.

The number of overlays can  be increased, but there are some limits.
DRM has a limit of 32 planes total because it uses a u32 bitmask to handle them. You have to have a primary and cursor plane per crtc. Other planes can be left as available for any crtc and this is what the beta full KMS driver does (it has 6 crtcs, so otherwise hits that limit very fast). 

FKMS does take it to the minimum and only adds one overlay, but that would only require a quick loop at https://github.com/raspberrypi/linux/blob/rpi-5.4.y/drivers/gpu/drm/vc4/vc4_firmware_kms.c#L1742 to create multiple overlay planes pepr crtc. You'd need to pass in a zpos or overlay number to vc4_fkms_plane_init so that it can set up default_zpos sensibly. I'm open to a PR for that, otherwise it'll get added to my long list of jobs.
The firmware side of FKMS allows for up to 16 planes spread over both displays, although that could be increased if really needed (it's only a define).

ardera, 2020-06-03:
> What is it that you're rendering? GL output? In which case, yes, as GL works with bottom left as the origin, whilst DRM uses top left.

Okay that makes sense. Yeah, it's GL output

> FKMS does take it to the minimum and only adds one overlay, but that would only require a quick loop at [...]

Okay. I think multiple overlay planes would be nice to have, though they're not strictly necessary for me right now either. If I find a use for them at some point I'll write a PR.

> The firmware side of FKMS allows for up to 16 planes spread over both displays, although that could be increased if really needed (it's only a define).

Is it also the same with the DSI display? (I assume you meant the HDMI displays) Does this 16 plane firmware-side limitation also affect the planes used by OpenMAX or MMAL? In other words, could I make use of 16 DRM planes simultaneously and at the same time use omxplayer to display some video?

6by9, 2020-06-03:
16 planes over whichever 2 displays have been initialised. DSI+HDMI0, DPI+HDMI0, or HDMI0+HDMI1.

IL, MMAL, and DispmanX APIs all create their own client instance of DispmanX, so you can add as many layers as you can get the hardware to handle via that route. This limit of 16 planes is only through the mailbox API that FKMS uses.

We will be looking at shifting the main focus towards the pure ARM KMS driver (full KMS) at some point, at which point IL, MMAL, and DispmanX stop working, and DRM is the only route to putting images on the screen.

ardera, 2020-06-04:
> 16 planes over whichever 2 displays have been initialised. DSI+HDMI0, DPI+HDMI0, or HDMI0+HDMI1.
>
> IL, MMAL, and DispmanX APIs all create their own client instance of DispmanX, so you can add as many [...]

Okay, that makes sense. Thanks for explaining it

> We will be looking at shifting the main focus towards the pure ARM KMS driver (full KMS) at some point, at which point IL, MMAL, and DispmanX stop working, and DRM is the only route to putting images on the screen.

I'm looking forward to full KMS!