raspberrypi/userland #630: 64 bit build failures

Noltari, 2020-05-28:
64 bit build fails:
```
[ 22%] Building C object interface/mmal/vc/CMakeFiles/mmal_vc_client.dir/mmal_vc_client.c.o
In file included from /home/noltari/workspace/openwrt/build_dir/target-aarch64_cortex-a72_musl/bcm27xx-userland-ad8ef597bdcf23032064c5036a9154b6b8beb649/build/inc/interface/vcos/vcos.h:114,
                 from /home/noltari/workspace/openwrt/build_dir/target-aarch64_cortex-a72_musl/bcm27xx-userland-ad8ef597bdcf23032064c5036a9154b6b8beb649/interface/mmal/mmal_common.h:40,
                 from /home/noltari/workspace/openwrt/build_dir/target-aarch64_cortex-a72_musl/bcm27xx-userland-ad8ef597bdcf23032064c5036a9154b6b8beb649/interface/mmal/mmal.h:355,
                 from /home/noltari/workspace/openwrt/build_dir/target-aarch64_cortex-a72_musl/bcm27xx-userland-ad8ef597bdcf23032064c5036a9154b6b8beb649/interface/mmal/vc/mmal_vc_client.c:29:
/home/noltari/workspace/openwrt/build_dir/target-aarch64_cortex-a72_musl/bcm27xx-userland-ad8ef597bdcf23032064c5036a9154b6b8beb649/build/inc/interface/vcos/vcos_assert.h:303:69: error: size of array 'vcos_static_assert' is negative
 #define vcos_static_assert(cond) __attribute__((unused)) extern int vcos_static_assert[(cond)?1:-1]
                                                                     ^~~~~~~~~~~~~~~~~~
/home/noltari/workspace/openwrt/build_dir/target-aarch64_cortex-a72_musl/bcm27xx-userland-ad8ef597bdcf23032064c5036a9154b6b8beb649/interface/mmal/vc/mmal_vc_msgs.h:139:1: note: in expansion of macro 'vcos_static_assert'
 vcos_static_assert(sizeof(mmal_worker_service_closed) <= MMAL_WORKER_MSG_LEN);
 ^~~~~~~~~~~~~~~~~~
```

If both BUILD_MMAL and BUILD_MMAL_APPS are manually set to FALSE:
```
[ 51%] Linking C executable ../../../../build/bin/raspistill
/home/noltari/workspace/openwrt/staging_dir/toolchain-aarch64_cortex-a72_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/../../../../aarch64-openwrt-linux-musl/bin/ld: cannot find -lmmal_core
/home/noltari/workspace/openwrt/staging_dir/toolchain-aarch64_cortex-a72_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/../../../../aarch64-openwrt-linux-musl/bin/ld: cannot find -lmmal_util
/home/noltari/workspace/openwrt/staging_dir/toolchain-aarch64_cortex-a72_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/../../../../aarch64-openwrt-linux-musl/bin/ld: cannot find -lmmal_vc_client
```

6by9, 2020-05-28:
https://github.com/raspberrypi/userland/commit/7d3c6b9f4c3ddeecefdeb2b882bada74a235249b needs reverting too.

Noltari, 2020-05-28:
> [7d3c6b9](https://github.com/raspberrypi/userland/commit/7d3c6b9f4c3ddeecefdeb2b882bada74a235249b) needs reverting too.

Yes, I confirm that reverting 7d3c6b9 works perfectly.

popcornmix, 2020-05-28:
Added revert.

pwuertz, 2020-06-01:
Am I reading this correctly? The MMAL component of this library currently cannot be built for 64bit, thus FFMPEG hardware encoding (which requires `h264_mmal`) is not possible on such systems?

6by9, 2020-06-01:
Use h264_v4l2m2m.

We are still investigating why there are problems within MMAL with 64bit. We've had an official 64bit beta distro for under a week.

pwuertz, 2020-06-01:
Thanks @6by9 for pointing in the right direction! Until today I didn't know what `v4l2m2m` was (associated `v4l2` with webcam-capturing up to now). Hit another wall though, as the Ubuntu 20.04 64bit distribution for Raspi4 doesn't have the `bcm2835_codec` kernel module that seems to be required for `h264_v4l2m2m`. I'll try moving to the recently released Raspberry OS you mentioned.

lead-free, 2020-07-06:
> Am I reading this correctly? The MMAL component of this library currently cannot be built for 64bit, thus FFMPEG hardware encoding (which requires `h264_mmal`) is not possible on such systems?

To my understanding without this library picamera will not work on 64bit Ubuntu?

6by9, 2020-07-06:
> To my understanding without MMAL library picamera will not work on 64bit Ubuntu?

Correct.

mjbogusz, 2020-08-27:
Reversion of [7d3c6b9](https://github.com/raspberrypi/userland/commit/7d3c6b9f4c3ddeecefdeb2b882bada74a235249b) had the side-effect of removing the `raspistill`, `raspivid` and other binaries from the `libraspberrypi-bin` package in RaspiOS 64 (since the `20200813` version)
There is a workaround of downgrading the `libraspberrypi*` packages to the `20200520` versions, but if I understood correctly, those versions had other issues.

Upon further testing I've found out that grabbing image using OpenCV still works (that's all I really need), but `raspistill` was a very useful utility for testing whether the camera works at all - when playing around with device trees (especially custom ones) it's not a given.

I've searched and scratched my head for quite a long time before finding this thread and diving into the code of the commits mentioned above to finally find out why the until-this-point perfectly working `raspistill` on my Pi suddenly disappeared after a simple `apt update && apt upgrade`.
If this side-effect was not an unforeseen one, maybe it would be possible to publish a changelog somewhere? Preferably with version tags tied to the published DEBs? It would be much easier to find such information in the future 👍

altineller, 2020-11-06:
> > Am I reading this correctly? The MMAL component of this library currently cannot be built for 64bit, thus FFMPEG hardware encoding (which requires `h264_mmal`) is not possible on such systems?
> 
> To my understanding without MMAL library picamera will not work on 64bit Ubuntu?

Hello, I managed to compile it all and make it work, using the unofficial ubuntu 18 distribution by TheRemote. 

It will not compile in official ubuntu 20.04.

hope this helps,
can

omartijn, 2021-07-19:
I'm unsure what the status is of this issue. I just tried compiling on bullseye and I got many warnings about casting pointer to integers (ough) and also the assert failure because assertions are being made that pointers are the same size as ints.

Given that some people report having compiled this successfully, are they working with local modifications? What am I missing?

JamesH65, 2021-07-19:
We are moving away from MMAL and replacing it and other firmware stuff with standard linux libraries. So for example, the camera API's are being replaced with the new libcamera library. picamera should follow at some point, although note that this was not an official Raspberry Pi developed library, its was from a third party. So the amount of time spent trying to make the legacy system work on 64bit is small, if any. It seems unlikely that anyone has compiled the libraries AND been able to make them work 100%, as that requires firmware side changes as well, which can only be done by us.

mjbogusz, 2021-07-19:
Disclaimer: the following applies mostly to the camera handling, that's my main point of interest regarding MMAL.

The problem is that using `MMAL` was easy - I could get an image from a camera module using C++/OpenCV in literally 4 lines. Even if some intricate parts of it didn't work properly, the basic stuff worked. But not anymore :/ and I don't even know which package to downgrade (if any), because my post from almost a year ago doesn't apply anymore - the 'recovered' `raspistill` is not able to capture images.

The `libcamera`... I have quite a few problems with it, mainly revolving around no examples/guides combined with quite complicated design and API (maybe rightfully so! I don't know much about handling ), being very immature (not a single tagged release so far) and IMO _questionable_ implementation decisions. But that's a rant for another time and place ;)

I think that if there is an approach or an API that allows similarly simple code as the 'old' ways (even with limited capabilities), it should be recommended instead of `libcamera` until it's more mature and documented.

JamesH65, 2021-07-20:
The issue is that there is a mismatch between the 32bit software running on the VC4/6 and the 64bit. some data structures, and more importantly, memory pointers, don't translate from the 64 to the 32 bit arena. And fixing that is not trivial (Imagine having a 64bit memory pointer in ARM space, and having to deal with is, somehow, in 32bit VC4 space...)

As for libcamera, its still under development - there is more stuff added to it everyday, so worth ensuring your libraries are up to date. But libcamera is the direction we are moving and where all development work is being done.