raspberrypi/userland #631: error: ‘%s’ directive argument is null [-Werror=format-overflow=]

error10, 2020-05-29:
**Describe the bug**
Compilation fails on RPi4 running Fedora 32 aarch64 with the error shown below.

**To reproduce**
 * Boot RPi4 from Fedora 32 aarch64 image, SD card created with Fedora arm-image-installer tool as `sudo arm-image-installer --target=rpi4 --image=/home/error/Downloads/Fedora-Minimal-32-1.6.aarch64.raw.xz --media=/dev/sdb`
 * `sudo dnf -y install @development-tools gcc-c++ cmake`
 * `./buildme --aarch64`

**Expected behaviour**
Compilation should succeed.

**Actual behaviour**
```
[ 91%] Building C object host_applications/linux/libs/bcm_host/CMakeFiles/bcm_host.dir/__/__/__/__/interface/vmcs_host/linux/vcfilesys.c.o
In file included from /home/error/userland/build/inc/interface/vcos/vcos.h:144,
                 from /home/error/userland/interface/vmcs_host/linux/vcfilesys.c:56:
/home/error/userland/interface/vmcs_host/linux/vcfilesys.c: In function ‘vc_hostfs_totalspace64’:
/home/error/userland/build/inc/interface/vcos/vcos_logging.h:234:88: error: ‘%s’ directive argument is null [-Werror=format-overflow=]
  234 | #  define _VCOS_LOG_X(cat, _level, fmt...)   do { if (vcos_is_log_enabled(cat,_level)) vcos_log_impl(cat,_level,fmt); } while (0)
      |                                                                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/error/userland/build/inc/interface/vcos/vcos_logging.h:245:32: note: in expansion of macro ‘_VCOS_LOG_X’
  245 | # define vcos_log_info(...)    _VCOS_LOG_X(VCOS_LOG_CATEGORY, VCOS_LOG_INFO, __VA_ARGS__)
      |                                ^~~~~~~~~~~
/home/error/userland/interface/vmcs_host/linux/vcfilesys.c:79:26: note: in expansion of macro ‘vcos_log_info’
   79 | #define DEBUG_MINOR(...) vcos_log_info(__VA_ARGS__)
      |                          ^~~~~~~~~~~~~
/home/error/userland/interface/vmcs_host/linux/vcfilesys.c:1019:4: note: in expansion of macro ‘DEBUG_MINOR’
 1019 |    DEBUG_MINOR( "vc_hostfs_totalspace for '%s' returning %" PRId64 "", path, ret );
      |    ^~~~~~~~~~~
/home/error/userland/interface/vmcs_host/linux/vcfilesys.c:1019:44: note: format string is defined here
 1019 |    DEBUG_MINOR( "vc_hostfs_totalspace for '%s' returning %" PRId64 "", path, ret );
      |                                            ^~
cc1: all warnings being treated as errors
make[2]: *** [host_applications/linux/libs/bcm_host/CMakeFiles/bcm_host.dir/build.make:96: host_applications/linux/libs/bcm_host/CMakeFiles/bcm_host.dir/__/__/__/__/interface/vmcs_host/linux/vcfilesys.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:517: host_applications/linux/libs/bcm_host/CMakeFiles/bcm_host.dir/all] Error 2
make: *** [Makefile:172: all] Error 2
```

**System**
* Raspberry Pi 4B 4GiB
* Fedora 32 aarch64
* gcc (GCC) 10.1.1 20200507 (Red Hat 10.1.1-1)
* Apr  1 2020 16:30:12 
Copyright (c) 2012 Broadcom
version 8e16a2ff5d1cda6c1c6cc0ac04a4db7b4d77e3b6 (clean) (release) (start_x)
* Linux mandori 5.6.14-300.fc32.aarch64 #1 SMP Wed May 20 20:23:09 UTC 2020 aarch64 aarch64 aarch64 GNU/Linux

6by9, 2020-05-29:
Joys, gcc-9 and above are being fussier about printf("%s", NULL), or by the looks of it any potential route for the parameter being NULL.

I don't believe vcfilesys.c is ever used by the system?
@popcornmix Am I right in this being the host side for the VPU accessing the host filesystem? If so then the whole thing can be culled.

tom--pollard, 2020-06-02:
Yes this is broken with gcc10, and is manually patched in the yocto bsp layer http://git.yoctoproject.org/cgit/cgit.cgi/meta-raspberrypi/tree/recipes-graphics/userland/files/0021-cmake-Disable-format-overflow-warning-as-error.patch