raspberrypi/userland #632: Camera Still Images - Issues and API-Docs

k-a-z-u, 2020-05-31:
First of all, thank you very much for the great piece of hardware and software! Many other embedded platforms are dead-on-arrival..
Probably the mentioned aspects are no issues, but my false assumptions due to not being able to find
a documentation how the camera API (mmal/vcos) is intended to be used (calling/setup order, constraints/requirements, and similar)

**System**
Revision        : c03111
Model           : Raspberry Pi 4 Model B Rev 1.1

Raspberry Pi reference 2020-02-13
vcgencmd version May 11 2020 19:00:52 
Linux dia 4.19.118-v7l+ #1311 SMP Mon Apr 27 14:26:42 BST 2020 armv7l GNU/Linux

raspberrypi-ui-mods:  Installed: 1.20200218
raspberrypi-sys-mods:  Installed: 20200514
openbox:  Installed: 3.6.1-8+rpt4
lxpanel:  Installed: 0.10.0-2+rpt9
pcmanfm:  Installed: 1.3.1-1+rpt21
rpd-plym-splash:  Installed: 0.23

Camera: v2.1

**Background**
I am working on something using only the still-image part of the raspberry camera.
Based on raspistill I played around with the API and tried to use only the bare minimum required.
Thus: Working without preview and encoder, just caputring RGB stills directly via the still image port into the buffer-callback.

Here, I noticed several "issues", that seem to work with the original raspistill, but not with my reduced code base.
I guess that they are related to not using a connected preview, but I could not find any documentation how all the aspects relate.

**Issues**
- Auto Exposure does not work
  Maybe this makes sense without an enabled preview, but AWB works without a preview.
  So I guess one would expect auto exposure to work as well, or adjust itself when capturing multiple consecutive stills.

- When using manual AWB, the values for Red and Blue seem to be ignored, and a constant 1.0.
  NOTE: This morning I updated the Raspberry and it suddenly started to work, so I guess this was fixed?
  
- One can not change back to Auto-AWB after having switched to Manual-AWB without disabling and re-enabling the Camera

- When not using a preview, control-callbacks are received, but all values (exposure, analog/digital gain, AWB gain) are 0

When enabling the preview port (connection or buffer-callback), some of above issues were solved.
However, my still images then all turned out to be black.
I guess I'm unable to figure out how port-coexistence should look like, and how exposure/AWB/gains are handled by the hardware when capturing only stills.

JamesH65, 2020-05-31:
Some of the statistics used to work out things like AWB etc are calculated from the preview frames. There is an option in raspistill to work them out from the capture, think its the --stats option. See how that works and use the same technique in your code. Should fix a few issues you are seeing.

k-a-z-u, 2020-05-31:
Thank you for the quick reply. Unfortunately, setting MMAL_PARAMETER_CAPTURE_STATS_PASS did not seem to change anything. I occasionally get a control callback with all parameters showing 0 [except for the denominators, which are 256]. Does this option require other specific options/configurations in order to work?

6by9, 2020-05-31:
The basis of the camera component is the OpenMax IL defined standard camera component. That had only two output ports - preview and capture. When you then considered stills during video capture you had an ambiguity as to what that meant and how you were meant to configure the pipeline.

There was then an extension to the spec (annoyingly only accessible to Khronos members) to create a 3 port camera variant - preview, video capture, and stills capture. That is what is implemented.
Stills capture without preview was not an use case required there.

The preview port is used to set up exposure and gain as these can't be done after the fact. White balance and some others can be computed from the captured frame (which is what MMAL_PARAMETER_CAPTURE_STATS_PASS enables). 

You should be able to set manual exposure time and gains if just doing stills captures, but AE is locked during stills captures.