raspberrypi/userland #633: glGetString returns error on RPi without vc4

limbouser, 2020-06-04:
I'm trying a Yocto build for RPi, when I disable vc4, and try to run an openGL application, I get an error saying it can't resolve glGetString(GL_VERSION). When I looked for the cause of that error, it said I doesn't have a proper openGL implementation. Any ideas on how I can get past it? (Enabling vc4 is not an option for me)

JamesH65, 2020-06-04:
Do you mean you have disabled FKMS on the Pi4? If so, you have no 3D support/driver, so gl calls will not work correctly.

On the Pi3 you could use the older userland code talking to the VC4 3D HW, the FKMS driver is disabled by default,

limbouser, 2020-06-04:
Actually I'm using a Raspberry Pi 3 and is working on Yocto images, not Raspbian. When I turn on the fkms the glGetString error doesn't occur, but when it's disable, I am having the glGetString error.