raspberrypi/userland #634: host_applications/linux/apps/vcdbg missing

hvenev, 2020-06-04:
host_applications/linux/CMakeLists.txt refers to the apps/vcdbg subdirectory, which does not appear to exist in this repository. Where is it?

timg236, 2020-06-04:
https://github.com/raspberrypi/userland
""
Note that this repository does not contain the source for the edidparser and vcdbg binaries due to licensing restrictions.
"""

The binary is release as part of the firmware repo
https://github.com/raspberrypi/firmware/tree/master/opt/vc/bin

hvenev, 2020-06-04:
So how do I read the videocore logs?

timg236, 2020-06-04:
sudo vcdbg log msg

hvenev, 2020-06-04:
As you said, vcdbg is not available.

6by9, 2020-06-04:
Binaries are included with the Pi firmware - https://github.com/raspberrypi/firmware/tree/master/opt/vc/bin

timg236, 2020-06-04:
> As you said, vcdbg is not available.

See first reply or on Raspbian /opt/vc/bin

hvenev, 2020-06-04:
I'm not interested in binaries of questionable origin.

timg236, 2020-06-04:
Well given that the firmware comes from the same repo good luck booting!

hvenev, 2020-06-04:
OK, I'm not interested in running binaries on the ARM core which I cannot reproducibly build.

hvenev, 2020-06-04:
It looks like the MMAL log is accessed through the "mmal_dbg_log" symbol. Does `vcdbg log msg` do something similar?

6by9, 2020-06-04:
Sorry, as stated, it is closed source and not ours to release (it's Broadcom's).
If you wish to read the VideoCore logs then you will have to trust the binaries.

closed 2020-06-04 by 6by9

hvenev, 2020-06-04:
vcfw/logging/logging.h is under a more permissive license, right?

pelwell, 2020-06-04:
We negotiated to release the source we have in a time when it was easier, but we I doubt we'll be releasing any more. Whether you choose to run an untrusted binary on your Pi I leave to you and your principles, but as somebody who wrote some of vcdbg and read the rest I can tell you there is nothing to worry about.

hvenev, 2020-06-04:
Is it OK if I use vcfw/logging/logging.h to aid with reading the logs?

hvenev, 2020-06-04:
Anyway, here is what I did based on the information I gathered from the code in this repository:

https://git.venev.name/hristo/rpi-vc-log/tree/vc-log.c

It seems to be able to read the message log reasonably well.

pelwell, 2020-06-04:
It's in the public domain with a BSD 3-clause licence - I don't think we can stop you. We're not trying to protect any secrets here, we just can't give you the source code.

hvenev, 2020-06-04:
Should I try to integrate https://git.venev.name/hristo/rpi-vc-log/tree/vc-log.c into this repository? Or do you think there's a better place for it?

pelwell, 2020-06-04:
If you had a BSD 2/3-clause licence (or a dual GPL2/BSD 2/3-clause licence) then it would be an easier decision.

hvenev, 2020-06-04:
Would that be `SPDX-License-Identifier:  BSD-3-Clause OR GPL-2.0-or-later`?

pelwell, 2020-06-04:
libfdt (which is in userland) is `SPDX-License-Identifier: (GPL-2.0-or-later OR BSD-2-Clause)`, and we've used the 3-Clause version, so `SPDX-License-Identifier: BSD-3-Clause OR GPL-2.0-or-later` is likely to be acceptable. Note that this isn't a promise or undertaking to adopt your source code - that decision isn't mine to make - but changing the licence is lowering the barrier to doing so.

hvenev, 2020-06-04:
OK, I've released the file under `SPDX-License-Identifier: (GPL-2.0-or-later OR BSD-3-Clause)`. I chose BSD 3-clause because it appears that it is the only BSD-like license that according to SPDX is approved both by the FSF and the OSI.

pelwell, 2020-06-04:
Even if we don't adopt it, I'm sure somebody will.

hvenev, 2020-06-05:
Yeah, I didn't use the debug_sym library because its header file is not installed. Maybe you should also fix that.

Given the tricky nature of arbitrarily accesing VideoCore memory, I feel like this functionality is better implemented in the kernel. I could try to write a driver that exposes `/sys/firmware/videocore/symbols` and `/sys/firmware/videocore/messages`.

Also, DMA from userspace through the framebuffer device? That surprised me. Does `/dev/vc-mem` do `read()` and `write()` correctly? I could try to add code that implements those using DMA instead, and switch to `read()` instead of `mmap()`.

pelwell, 2020-06-06:
> Yeah, I didn't use the debug_sym library because its header file is not installed.

Do you mean it isn't under /usr/include or /opt/vc/include etc.? That's true, but at least it is in the userland repo - I think it's just an oversight.

The reason for the framebuffer hack is that the FB driver was a user of memory-to-memory DMA, and this was convenient.

vc_mem only implements `mmap()` as an interface, no `read()` or `write()` currently. I'm not keen on improving access to VC memory at a time when we're trying to move more services to the ARM - it seems like a backwards step.

What I did want to mention is that we are open to the idea of adopting your vclog  code. The timestamps need to be decoded correctly (they are microsecond ticks), although for some weird reason vcdbg itself chooses to display them in milliseconds rather than seconds, and the wrap-around cases considered, but you're pretty close already.

hvenev, 2020-06-06:
> The reason for the framebuffer hack is that the FB driver was a user of memory-to-memory DMA, and this was convenient.

Sadly that hack is of no use to me because I have disabled the framebuffer driver. So libdebug_sym falls back to `memcpy()`. On a Pi 4 `OpenVideoCoreMemory` crashes with "Bus error" when accessing memory at 0x3edc9232 (containing a symbol name, I think). are there any alignment restrictions? It seems like glibc's aarch64 `memcpy()` performs unaligned accesses.

> What I did want to mention is that we are open to the idea of adopting your vclog code. The timestamps need to be decoded correctly (they are microsecond ticks), although for some weird reason vcdbg itself chooses to display them in milliseconds rather than seconds,

Fixed. Timestamps are displayed as seconds, the same way as by dmesg.

> and the wrap-around cases considered, but you're pretty close already.

Done, I think. It's just a byte-oriented circular buffer, right?

pelwell, 2020-06-08:
For 1GB Pis which aren't Pi 4, the workaround to avoid the need for DMA is to sacrifice the top 16MB with `total_mem=1008`.

> Done, I think. It's just a byte-oriented circular buffer, right?

Yes, that's right.

hvenev, 2020-06-08:
> For 1GB Pis which aren't Pi 4, the workaround to avoid the need for DMA is to sacrifice the top 16MB with `total_mem=1008`.

Don't get me wrong, I agree that DMA is the right thing to do. However, given that I am not getting paid to do this, I am not willing to write code that would be of no use to me. As I don't use the framebuffer interface, libdebug_sym's `vc_mem_copy` does not work for me. In my mind, that function is equivalent to `return -ENOENT`. I explained why `memcpy()` does not work either.

The two options I have are the following:

 - Implement DMA using `/dev/vc-mem`'s `read()` and `write()`, and deprecate `mmap` (`WARN_ON_ONCE` maybe?) because it is very broken.
 - Hack a custom `memcpy()` that works with VC memory on some devices.

The first approach works for all users, even the ones who have made the objectively correct choice `CONFIG_FB_BCM2708=n`.

pelwell, 2020-06-08:
I was trying to suggest a way to avoid writing code, but if you are prepared to do so then clearly the first option - adding restricted DMA support to the log area - is the better one.