raspberrypi/userland #636: Can't save non-corrupted image files with imx477 camera

ramiromagno, 2020-06-23:
```
raspistill -o output_filename.jpg -v
```

```
"raspistill" Camera App (commit f97b1af1b3e6 Tainted)

Camera Name imx477
Width 4056, Height 3040, filename output_filename.jpg
Using camera 0, sensor mode 0

GPS output Disabled

Quality 85, Raw no
Thumbnail enabled Yes, width 64, height 48, quality 35
Time delay 5000, Timelapse 0
Link to latest frame enabled  no
Full resolution preview No
Capture method : Single capture

Preview Yes, Full screen Yes
Preview window 0,0,1024,768
Opacity 255
Sharpness 0, Contrast 0, Brightness 50
Saturation 0, ISO 0, Video Stabilisation No, Exposure compensation 0
Exposure Mode 'auto', AWB Mode 'auto', Image Effect 'none'
Flicker Avoid Mode 'off'
Metering Mode 'average', Colour Effect Enabled No with U = 128, V = 128
Rotation 0, hflip No, vflip No
ROI x 0.000000, y 0.000000, w 1.000000 h 1.000000
Camera component done
Encoder component done
Starting component connection stage
Connecting camera preview port to video render.
Connecting camera stills port to encoder input port
Opening output file output_filename.jpg
Enabling encoder output port
Starting capture -1
Finished capture -1
Closing down
Close down completed, all components disconnected, disabled and destroyed
```

No matter what I try, I can't for the life of me to get image files that are not corrupted. I tried different encodings: jpg, png, gif and bmp, and the saved files are always corrupted.

Note that I can see the image preview which looks fine, and I can also see live streaming just fine, but every single export results in a corrupted image, no matter the file format. I tried to open the image files with programs like `sxiv`, `gpicview`, `chromium`, etc.. but to no avail.

JamesH65, 2020-06-23:
Please update your system, and retest. 

```
sudo apt update
sudo apt full-upgrade
```

6by9, 2020-06-23:
At present the fixes for imx477 are only in apt for Pi4. For Pi0-3 you need to use ```sudo rpi-update```.

Please also confirm that you are NOT using the 64bit kernel or OS. 64bit kernel with 32bit userspace should be OK, but MMAL is not currently supported with 64bit userspace.

ramiromagno, 2020-06-23:
I'm using Archlinux ARM actually. So I am guessing I need to change userspace to 32bit...?

6by9, 2020-06-23:
If you clone userland, revert f97b1af1b3e653f9da2c1a3643479bfd469e3b74 and e31da99739927e87707b2e1bc978e75653706b9c, and rebuild using ```buildme --aarch64```, then you may have a chance of things working, but particularly with video playback we saw some odd behaviour (hence the reverts).

ramiromagno, 2020-06-23:
okay... thanks!

ramiromagno, 2020-06-23:
I just followed your instructions: (1) I cloned userland, (2) reverted those two commits, (3) and ran `./buildme --aarch64`. Everything ran smoothly, and it installed all sorts of files into `/opt/vc/`. Yet, the generated files with:

```
raspistill -o output_filename.jpg -v
```

still generates a corrupted file...

JamesH65, 2020-06-23:
Since you are not using a Raspberry PI OS, I doubt there is much else we can suggest. We know our official OS (32bit) works. I suggest moving to that.

ramiromagno, 2020-06-23:
Okay, thanks! But does Raspberry PI OS work with the Pi 4B 8GB RAM version?

6by9, 2020-06-23:
> Okay, thanks! But does Raspberry PI OS (32bit) work with the Pi 4B 8GB RAM version?

Yes, it works fine. A single process can only use 3GB of RAM, but that is rarely a restriction. Multiple processes can use the full 8GB.

ramiromagno, 2020-06-23:
I see. Thank you so much. I am going to try Raspberry PI OS (32bit).

pelwell, 2020-06-23:
Can you upload a few of the corrupted images so we can inspect them?

ramiromagno, 2020-06-23:
I was about to dd the old image... Hang on a sec...

ramiromagno, 2020-06-23:
[image]

[image]

pelwell, 2020-06-23:
Thanks - that looks like 640 bytes of rubbish, followed by a JFIF header, then mostly 5MB of random memory content (Quake I save file, `file` utility, etc.).

ramiromagno, 2020-06-23:
I just tried running `raspicam_test` from the package `raspicam`:

```
Name            : raspicam
Version         : 0.1.8-1
Description     : C++ API for using Raspberry camera (with OpenCV)
Architecture    : armv7h
URL             : http://www.uco.es/investiga/grupos/ava/node/40
Licenses        : BSD
Groups          : None
Provides        : raspicam
Depends On      : vtk
Optional Deps   : None
Required By     : None
Optional For    : None
Conflicts With  : raspicam
Replaces        : raspicam
Installed Size  : 102.38 KiB
Packager        : Unknown Packager
Build Date      : Tue 23 Jun 2020 01:48:00 AM WEST
Install Date    : Tue 23 Jun 2020 01:48:39 AM WEST
Install Reason  : Explicitly installed
Install Script  : No
Validated By    : None
```
and voila:

```
Usage (-help for help)
Connecting to camera
Connected to camera =10000000a9f0e518 bufs=3686400
Capturing....
 capturing ...30/100Saving image30.ppm
 capturing ...60/100Saving image60.ppm
 capturing ...90/100Saving image90.ppm
 capturing ...95/100
Images saved in imagexx.ppm
4.04819 seconds for 100  frames : FPS 24.7024
```

The image below is after conversion from ppm to png with `convert` from imagemagick:

[image]

6by9, 2020-06-30:
> Architecture    : armv7h

Presumably that raspicam package has been built against a 32bit version of the MMAL library, which will work. If raspivid etc have been built as 64bit, and therefore linking against a 64bit MMAL, it won't work.

xlla, 2020-11-10:
I have build userland base on 4a57ea410
test on Pi 4 4GB, aarch64,

<pre>
raspistill -o output_filename.jpg -v
</pre>
the image was captured ok ,and I can view it use fim output_filename.jpg.