raspberrypi/userland #640: Raspiyuv always taking two pictures

Ismael2878, 2020-07-24:
It seems to be the raspiyuv command is always taking 2 pictures. Therefore the needed time is twice as big as expected.

E.g. when executing next command:
 raspiyuv -drc off -mm matrix -rgb -set -st -ex off -n -ss 50000000 -md 3 -t -1 -dg 2.0 -ag 2.0 -awb off -awbg 2.0,2.0 -o ~/output.raw

There are 2 camera control callbacks:
Camera control callback  cmd=0x48435045mmal: Exposure now 49999693, analog gain 512/256, digital gain 512/256
mmal: AWB R=512/256, B=512/256
Camera control callback  cmd=0x48435045mmal: Exposure now 49999693, analog gain 512/256, digital gain 512/256

Time needed is almost 2 minutes what is actually more than twice the defined shutter time.

System Information
------------------

Raspberry Pi Zero W Rev 1.1
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Used camera:  IMX477

JamesH65, 2020-07-24:
Why have you set -t to -1? What are you trying to achieve with that?

Ismael2878, 2020-07-24:
My excuses, this is a paste error. The used value was -t 1.

JamesH65, 2020-07-24:
What about camera *buffer* callbacks - they are the ones that save the image, and you may get more than one of those depending on how much data needs to be saved.  But it all ends up in the same file.

Ismael2878, 2020-07-24:
In the end we get indeed one file, but the time to make the file takes twice as long as we would expect (based on the shutter time). When we put the preview on we can also see that the previews before both callbacks are different. Therefore we get the impression there are 2 images captured instead of one.

JamesH65, 2020-07-24:
It seems to still be doing an AGC/AWB pass on the first frame, which sets it up to take the actual image. Perhaps.

6by9, 2020-07-24:
All the raspicam apps start preview by default. If as frame has been requested, then it will be completed, therefore you get one preview frame and one capture frame. The preview frame is also used to collect some stats for AE and AWB to work with.

If you have fixed all parameters, then raspistill gained a mode to drop immediately into burst mode (so no preview).
See https://www.raspberrypi.org/forums/viewtopic.php?p=1663899#p1663899 / https://github.com/raspberrypi/userland/commit/2fe4ca33dc29dc3ede0044d53b0c0abe0ea6536b

It looks like we missed out the equivalent change to RaspiStillYUV.c, but exactly the same change should be applicable there.

Ismael2878, 2020-07-27:
I can indeed confirm it is working well with raspistill in combination with the burst mode.
Do you have any suggestions where we can adapt this in the RaspiStillYUV.c so the behavior will be the same ?

6by9, 2020-07-27:
I needed to create a PR for userland, so have tacked this patch in as well. #645

Ruffio, 2020-09-14:
Has this issue been resolved and it can be closed?