raspberrypi/userland #642: Raspiyuv and raspividyuv docs

dherrada, 2020-07-24:
It seems there really isn't much documentation on raspiyuv and raspividyuv. One thing that is really missing from the docs is how to actually open the images. It took me forever, but I did find [this forum thread](https://www.raspberrypi.org/forums/viewtopic.php?t=54268) where that was explained. I was also not able to find a single piece of official documentation about raspividyuv. I would definitely be willing to make a pr with some more docs, although I don't think I'd be much help with the docs for raspividyuv.

JamesH65, 2020-07-24:
I'm not sure specifically what the docs would contain. If you have a need for these two applications, you would almost certainly understand what the output is. 

So what were you wanting to do with them that was not covered by raspistill and raspivid?

dherrada, 2020-07-24:
> I'm not sure specifically what the docs would contain. 

@JamesH65 Pretty much what the raspiyuv docs have, which are actually great except for the one thing I mentioned above, just for taking videos instead of pictures.

> If you have a need for these two applications, you would almost certainly understand what the output is.

Not necessarily. Every camera I've ever worked with that can output raw images output them in formats that can be opened by pretty much any image editor and most image viewers. Photographers have a very real need for raw images, but I seriously doubt very many of them would know much more about the specific raw format their camera uses than its file extension.

> So what were you wanting to do with them that was not covered by raspistill and raspivid?

@JamesH65 Really anything that you want to post-process. For example, I'm doing some astrophotography with the HQ camera, and JPGs are really bad for that. As far as raspivid goes, I just wanted to be able to color correct my videos in something like davinci resolve.

dherrada, 2020-07-24:
Also, thanks for the quick reply

JamesH65, 2020-07-24:
These are not raw images though. These are fully processed images, just using a different output format i.e. YUV instead of JPEG or H264. Better terminology would be uncompressed rather than raw. 

For really raw (er data off the sensor) stills you need to use the --raw option on raspstill. There's no such thing as raw bayer video .

dherrada, 2020-07-24:
> For really raw (ie data off the sensor) stills you need to use the --raw option on raspstill.

@JamesH65 Looks like I really need to work on my reading comprehension skills. You're totally right. Thanks for clearing all that up for me (although I do still think there should be _some_ documentation for raspividyuv)

closed 2020-07-24 by dherrada