raspberrypi/userland #643: Shutter Speed is multiplied by 7 on HQ Camera

alkemir, 2020-07-27:
**Describe the bug**
When passing the shutter speed option to raspistill or raspiyuv, the actual exposure time is 7 * whatever I passed.
For example, raspistill -ss 1000000 will actually take 7 seconds of exposure. I am using the HQ camera.

**To reproduce**
Use an HQ camera and run
time raspistill -t 100 -ss 1000000 -w 328 -h 246  -v -set -o test.jpg

It will take about 8 seconds. Run it with -ss 1000000 and it will take about 71 seconds. I added some printfs to time just the capture and it is always 7 * the parameter passed. 

**Expected behaviour**
I would expect that passing a million microseconds as shutter speed would generate a 1-second exposure shot.

**Actual behaviour**
It actually generates a 7-seconds exposure time.

**System**
PI4B
Raspberry Pi reference 2020-05-27
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 825107f04027269db77426046f5085475b1ea22f, stage5
Jun  1 2020 13:24:51 
Copyright (c) 2012 Broadcom
version 6379679d1ec6a8c746d7e77e015f5b56b939976f (clean) (release) (start_x)
Linux raspberrypi 4.19.118-v7l+ #1311 SMP Mon Apr 27 14:26:42 BST 2020 armv7l GNU/Linux

JamesH65, 2020-07-27:
You are confusing the overall time to take a picture with the exposure time. To take a picture will take multiple frames-  first frame is discarded, then a few frames are used to run the AGC and the AWB, then the picture is taken. So if each frame takes 1s the overall time is 7 times the number of frames actually used. To get round this turn off the AWB, the AGC, and set manual values, and also set burst mode.

6by9, 2020-07-27:
See https://www.raspberrypi.org/forums/viewtopic.php?p=1663899#p1663899

alkemir, 2020-07-28:
Makes sense. Running

`raspistill -t 10 -md 3 -bm -ex off -ag 1 -ss 6000000 -st -awb off`

give the expected results. May I suggest a -debug mode in which each frame taken is printed along with the reason why it was taken?

I noticed using -e png does increase the time considerably. Is this expected due to compression? Or is this causing capture of additional frames? (Also, as my original concerned has been addressed, feel free to close this issue)

Thank you for such a quick response.

6by9, 2020-07-28:
> May I suggest a -debug mode in which each frame taken is printed along with the reason why it was taken?

The choice is made by the firmware, not the application, therefore it's not possible for the app to state the reason.

2 reasons for PNG compression to be slower. Firstly it is an RGB format, so the YUV frames generated by the camera stack have to be converted. Secondly it is a software encode on the GPU rather than the hardware accelerated JPEG encode.

closed 2020-07-30 by alkemir