raspberrypi/userland #648: /etc/passwd lists incorrectly /usr/bin/bash for root shell, instead of /bin/bash

dimdin, 2020-08-28:
**Describe the bug**
/etc/passwd lists /usr/bin/bash for root shell.
Bash moved from /usr/bin/bash to /bin/bash.

**To reproduce**
```
$ sudo su
su: failed to execute /usr/bin/bash: No such file or directory
$ sudo vi /path/to/file
:!ls
Cannot execute shell /usr/bin/bash
```
**Expected behaviour**
/etc/passwd root entry must be:
```
root:x:0:0:root:/root:/bin/bash
```
**Actual behaviour**
/etc/passwd root entry is:
```
root:x:0:0:root:/root:/usr/bin/bash
```
**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:
raspinfo: command not found

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Pi3B

* Which OS and version (`cat /etc/rpi-issue`)?
Raspberry Pi reference 2019-07-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 175dfb027ffabd4b8d5080097af0e51ed9a4a56c, stage2

* Which firmware version (`vcgencmd version`)?
Aug  6 2020 16:24:30
Copyright (c) 2012 Broadcom
version af3edc2de473197cdfe1ff5a8ff2d34095d5b336 (clean) (release) (start_cd)

* Which kernel version (`uname -a`)?
Linux @@@@ 5.4.51-v7+ #1333 SMP Mon Aug 10 16:45:19 BST 2020 armv7l GNU/Linux

pelwell, 2020-08-28:
If that was ever true for an unmodified image from us, it isn't true now:
```
pi@raspberrypi:~$ cat /etc/rpi-issue
Raspberry Pi reference 2020-08-20
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 9a3a10bf1019ebb2d59053564dc6b90068bad27d, stage5
pi@raspberrypi:~$ grep bash /etc/passwd
root:x:0:0:root:/root:/bin/bash
pi:x:1000:1000:,,,:/home/pi:/bin/bash
pi@raspberrypi:~$
```

dimdin, 2020-08-28:
Nice, to have the problem fixed on new images.
But what happened? I am sure I did not changed the shell myself.

I have raspbian installed on two Pi. One of them have the problem.
The case is really strange, please note the dates.

The first one, without the problem:
```
> ls -l /etc/rpi-issue
-rw-r--r-- 1 root root 145 Jun 20  2019 /etc/rpi-issue
> cat /etc/rpi-issue
Raspberry Pi reference 2019-06-20
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 150e25c4f8123a4c9c63e8dca1b4737fa6c1135c, stage2
> grep bash /etc/passwd
root:x:0:0:root:/root:/bin/bash
```

The second one, with the problem:
```
> ls -al /etc/rpi-issue
-rw-r--r-- 1 root root 145 Jul 10  2019 /etc/rpi-issue
> cat /etc/rpi-issue
Raspberry Pi reference 2019-07-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 175dfb027ffabd4b8d5080097af0e51ed9a4a56c, stage2
> grep bash /etc/passwd
root:x:0:0:root:/root:/usr/bin/bash
```
The pi installed on June 2019 is fine. The but the next one installed one month later on July 2019 have this problem.

I am closing the issue, because there is no problem now, and it requires detective skills to find out what happened.
If someone sees a message "failed to" or "cannot execute /usr/bin/bash"; the fix is to change /usr/bin/bash to /bin/bash in /etc/passwd.

closed 2020-08-28 by dimdin