raspberrypi/userland #649: interface/khronos/common/khrn_int_hash_asm.s deleted after "make clean"

HinTak, 2020-09-04:
After build and install on aarch64 (ubuntu 20.04), this file is deleted. It seems to be specific to only some specific platform/build-config anyway. 

Anyhow, files being version-checked should either stay on the system, or should not be version-controlled. Having the build system destroying a version-controlled file seems wrong.

HinTak, 2020-09-04:
This appears to be a recent regression for the 3 commits leading up to https://github.com/raspberrypi/userland/commit/f73fca015d421b763936667a0b58fe5024d59921 as I had an earlier clone up to 3e59217 where it did not happen. Possibly the debug_sym change?

pelwell, 2020-09-04:
Really? List the steps from a clean clone to demonstrate the problem.

HinTak, 2020-09-04:
It was a scripted build as part of something else. The script is really just

```
git clone ...
pushd userland
./buildme --aarch64
popd
```

I noticed the missing file only because I was checking with `git status --ignored` (which shows the expected and untracked build directory) before throwing it away. I had and earlier clone / build elsewhere and that one does not shows any abnormality. Anyway, I am not too bothered either way, just thought it should be filed.

pelwell, 2020-09-04:
I can't reproduce it on a Pi 4 with arm_64bit=1. The userland is 32-bit, but that should make no difference to basic operations like copying and deleting files.

Are you sure it wasn't the rest of the script that did the deletion? Try running just those 4 commands again in a clean directory - it doesn't take long.

HinTak, 2020-09-04:
Argh, apologies. I realized what happened - there is nothing wrong with userland nor the build script themselves. But I was doing the clone inside an out-of-tree kernel module directory. Doing `make -C /lib/modules/$(uname -r)/build M=$(pwd) clean` will recursively delete any assumed generated assembly file, as is customary in kernel dev... Sorry for the noise.

It is a bit odd to have a single assembly file in userland though!

HinTak, 2020-09-04:
As I mentioned earlier, the script is part of something else. That "something else" is a kernel module, and it requires dtoverlay etc in userland to load...

closed 2020-09-04 by HinTak