raspberrypi/userland #652: "vcgencmd get_config int" omits settings with value "0"

MichaIng, 2020-10-14:
**Describe the bug**
`vcgencmd get_config int` consequently omits all settings which are set to a value of `0`.

**To reproduce**
1. With empty/default `config.txt`, run `vcgencmd get_config int`
2. Take any shown setting that can be set to `0`, like `enable_tvout` on non-RPi4 (I know has no effect) or `enable_uart` on models with no onboard WiFi.
3. Reboot and rerun `vcgencmd get_config int` to watch those settings missing, while running `vcgencmd get_config <setting>` shows the setting with applied value `0` just fine.

**Expected behaviour**
`vcgencmd get_config int` is expected to give a full overview of all available settings and their applied/active values, including those which are set to `0`.

**Actual behaviour**
```
2020-10-14 13:12:57 root@micha:/var/log# vcgencmd get_config int
aphy_params_current=819
arm_freq=900
arm_freq_min=300
audio_pwm_mode=514
config_hdmi_boost=5
disable_auto_turbo=1
disable_commandline_tags=2
disable_l2cache=1
disable_overscan=1
disable_splash=1
display_hdmi_rotate=-1
display_lcd_rotate=-1
dphy_params_current=547
force_pwm_open=1
framebuffer_depth=8
framebuffer_height=16
framebuffer_ignore_alpha=1
framebuffer_swap=1
framebuffer_width=16
init_uart_clock=0x2dc6c00
initial_turbo=20
lcd_framerate=60
max_framebuffer_height=16
max_framebuffer_width=16
over_voltage_avs=0x249f0
over_voltage_avs_boost=37500
pause_burst_frames=1
program_serial_random=1
sdram_freq=450
temp_limit=65
temp_soft_limit=50
total_mem=1024
hdmi_force_cec_address:0=65535
hdmi_force_cec_address:1=65535
hdmi_ignore_hotplug:0=1
hdmi_pixel_freq_limit:0=0x9a7ec80
2020-10-14 13:13:37 root@micha:/var/log# vcgencmd get_config enable_tvout
enable_tvout=0
2020-10-14 13:13:48 root@micha:/var/log# vcgencmd get_config enable_uart
enable_uart=0
2020-10-14 13:13:53 root@micha:/var/log# vcgencmd get_config sdtv_mode
sdtv_mode=0
2020-10-14 13:14:42 root@micha:/var/log# cat /proc/device-tree/model
Raspberry Pi 2 Model B Rev 1.1
2020-10-14 13:14:55 root@micha:/var/log# vcgencmd get_config force_eeprom_read
force_eeprom_read=0
2020-10-14 13:15:10 root@micha:/var/log# grep '=0$' /boot/config.txt
enable_tvout=0
enable_uart=0
force_eeprom_read=0
```

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? `RPi 2 v1.1 BCM2709`
* Which OS and version (`cat /etc/rpi-issue`)?
```
Raspberry Pi reference 2020-08-20
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 9a3a10bf1019ebb2d59053564dc6b90068bad27d, stage2
```
* Which firmware version (`vcgencmd version`)?
```
Sep 30 2020 12:28:58
Copyright (c) 2012 Broadcom
version e8f24e9043de5c5299e015b9f0e82f45bd66348b (clean) (release) (start_cd)
```
* Which kernel version (`uname -a`)?
```
Linux micha 5.4.68-v7+ #1343 SMP Mon Sep 28 12:38:29 BST 2020 armv7l GNU/Linux
```

popcornmix, 2020-10-14:
It is behaving as intended. It shows the non-default settings which is useful for collecting debugging information.

closed 2020-10-14 by pelwell

MichaIng, 2020-10-14:
It does not only show non-default settings as most of them are not set in my `config.txt` and it shows all of them even with empty `config.txt`. Examples are `force_eeprom_read` which defaults to `1` while I have it set to `0` and `enable_uart` where it is the same (on RPi2) and it is not shown. Here my `config.txt`:
```
2020-10-14 13:33:56 root@micha:/var/log# grep '^[[:blank:]]*[^#]' /boot/config.txt
hdmi_ignore_hotplug=1
framebuffer_width=16
framebuffer_height=16
max_framebuffer_width=16
max_framebuffer_height=16
framebuffer_depth=8
enable_tvout=0
disable_overscan=1
gpu_mem_256=16
gpu_mem_512=16
gpu_mem_1024=16
disable_splash=1
enable_uart=0
temp_soft_limit=50
temp_limit=65
initial_turbo=20
dtparam=sd_overclock=100
arm_freq_min=300
force_eeprom_read=0
```
I can

popcornmix, 2020-10-14:
Not all settings are purely user defined. Some are set by the firmware where they vary based on platform.

MichaIng, 2020-10-14:
Okay, then it is not clear what is meant by "default" and what not. Regardless, I see all settings which do not report `0` when asking explicitly via `vcgencmd get_config <setting>` and I see not a single setting which is set to `0`, even when it defaults to a different value (I omit it in `config.txt`) and I set it explicitly to `0`. I can show you any example, if needed, so if definitely does not behave as expected.

pelwell, 2020-10-14:
It shows all non-zero settings. In most cases that means non-default settings, but there are exceptions.

This behaviour is not going to change, which is why the issue is closed.

MichaIng, 2020-10-14:
Okay, thanks for clarification. Since it is consistent, it can be worked with:
> Assume `0` as long as `vcgencmd get_config int` does not tell different.