raspberrypi/userland #653: "vcgencmd get_config core_freq" reports "0"

MichaIng, 2020-10-14:
**Describe the bug**
```
2020-10-14 18:07:44 root@micha:/var/log# vcgencmd get_config arm_freq
arm_freq=900
2020-10-14 18:07:50 root@micha:/var/log# vcgencmd get_config core_freq
core_freq=0
2020-10-14 18:08:57 root@micha:/var/log# vcgencmd get_config gpu_freq
gpu_freq=0
2020-10-14 18:09:02 root@micha:/var/log# vcgencmd measure_clock core
frequency(1)=250000000
2020-10-14 18:09:07 root@micha:/var/log# vcgencmd measure_clock v3d
frequency(46)=250000000
2020-10-14 18:09:47 root@micha:/var/log# grep -E '^[[:blank:]]*(core|gpu)_freq=' /boot/config.txt
```

**To reproduce**
Run the above command when booting (RPi 2) without settings those settings in `config.txt` but booting with 16 MiB GPU memory, hence cut-down mode.

**Expected behaviour**
`vcgencmd get_config (core|gpu)_freq` should report the actual 250 MHz default values which can be verified by measuring them via `vcgencmd measure_clock (core|gpu)`.

**Actual behaviour**
`vcgencmd get_config core_freq` and `vcgencmd get_config gpu_freq` report `0`.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? `Raspberry Pi 2 Model B Rev 1.1`
* Which OS and version (`cat /etc/rpi-issue`)?
```
Raspberry Pi reference 2020-08-20
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 9a3a10bf1019ebb2d59053564dc6b90068bad27d, stage2
```
* Which firmware version (`vcgencmd version`)?
```
Sep 30 2020 12:28:58
Copyright (c) 2012 Broadcom
version e8f24e9043de5c5299e015b9f0e82f45bd66348b (clean) (release) (start_cd)
```
* Which kernel version (`uname -a`)?
```
Linux micha 5.4.68-v7+ #1343 SMP Mon Sep 28 12:38:29 BST 2020 armv7l GNU/Linux
```

popcornmix, 2020-10-14:
I thought we'd discussed this.
get_config queries a structure that contains non-default settings obtained from config.txt and platform specific initialisation code.
If the value is zero you end up with the default build time setting, which is 250MHz for core_freq.
But that value isn't contained in the list of non-default settings, so can't be queried through this interface.

MichaIng, 2020-10-14:
Okay let me see if I did understand:
- `vcgencmd get_config int` prints only settings with values which do not equal zero, regardless if zero is the default value on that particular model, i.e. when the setting is not set in `config.txt`.
- But the other way round it is true as well, when checking a specific setting like `vcgencmd get_config <setting>` and it reports zero, this can mean it is the default value, regardless if that default is zero or not?

How can one know then, when `vcgencmd get_config <setting>` reports zero, if zero is the actually active/set value or if a different default is active? Case of binary settings it is even worse:
```
2020-10-14 18:22:13 root@micha:/var/log# vcgencmd get_config sdtv_mode
sdtv_mode=0
```
How do I know now whether the `sdtv_mode` is indeed set to zero or whether the build-time default is active which might be `1`, without having a look into the documentation?
__________
But I guess I understood know: Regardless if used with or without argument, `get_config` prints values from a sort of table with is empty (or contains zeros) unless/until firmware/`config.txt`/something sets/changes a specific value. When a "field" is empty (or zero), it is omitted in the `get_config` overview (no argument) and zero is printed when asking for that specific field/setting via argument.

Just one question: When setting `v3d_freq=0` in `config.txt`, does it try to apply 0 MHz for the 3D block or does this explicitly mean to apply the default value and hence equals `v3d_freq=250` (on RPi 2)?

closed 2020-10-14 by pelwell

photron, 2020-11-23:
Any change this behavior along with the default values could be documented somewhere? For example here:

https://www.raspberrypi.org/documentation/raspbian/applications/vcgencmd.md